arduino/arduino-builder #2: Use os.PathListSeparator for list of folders in cmd-line args

wayoda, 2015-09-15:
Being able to specify arbitrary folders for libraries is a nice feature, but please don't use the comma to separate the items in a list of folders.

Use the platform independent [os.PathListSeparator](http://golang.org/pkg/os/#pkg-constants) (`:` on unix, `;` on windows).

Why?
- every commadline tool  I know of does it that way. Users expect to seperate different paths by their local path separator character 
- Go has a [function](http://golang.org/pkg/path/filepath/#example_SplitList) to split a list of folders based on the os.PathListSeparator constant 

Eberhard

ffissore, 2015-09-15:
Comma is the default way for the "flag" package. However, flag also supports multiple args of the same type. So, writing `-hardware folder1,folder2` is the same thing as `-hardware folder1 -hardware folder2`

This is the preferred way of providing params as it allows to use terminal autocompletion when available

ffissore, 2015-09-15:
Uhm I was wrong about the "default way". I'll fix it. Still, don't use that syntax. Specify multiple `-hardware` params (or `-tools`, or `-libraries` or `-prefs`)

commit 8c877a11 (2015-09-15): Values of params supporting multiple values can be separated by both a comma or
a platform specific path separator (":" on unixes, ";" on windows)

Specifying each param multiple times (-param value -param value) instead of
separating values with commas is preferred

Fixes #2
closed 2015-09-15 by ffissore