arduino/arduino-builder #3: error running the builder  : `recipe.preproc.includes pattern is missing`

wayoda, 2015-09-15:
Compile an empty sketch from the commandline :
```
void setup() {}
void loop() {}
```

Compile with
```
wayoda@shredder:~/sketchbook/empty$ arduino-builder -build-path . -fqbn arduino:avr:nano -hardware /home/wayoda/opt/arduino/arduino-1.6.6/hardware -tools /home/wayoda/opt/arduino/arduino-1.6.6/tools/ -verbose -debug-level 10 empty.ino 
```

yields
```
Running: SetupHumanLoggerIfMissing
Running: CreateBuildPathIfMissing
Running: ContainerSetupHardwareToolsLibsSketchAndProps
Running: AddAdditionalEntriesToContext
Running: RecipeByPrefixSuffixRunner
Looking for recipes like hooks.prebuild*.pattern
Running: HardwareLoader
Running: PlatformKeysRewriteLoader
Running: RewriteHardwareKeys
Running: ToolsLoader
Running: TargetBoardResolver
Running: AddBuildBoardPropertyIfMissing
Running: LibrariesLoader
WARNING: Category '' in library EEPROM is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library SPI is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library SoftwareSerial is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library Wire is not valid. Setting to 'Uncategorized'
Running: SketchLoader
Running: SetupBuildProperties
Running: LoadVIDPIDSpecificProperties
Running: SetCustomBuildProperties
Running: ContainerBuildOptions
Running: CreateBuildOptionsMap
Running: LoadPreviousBuildOptionsMap
Running: WipeoutBuildPathIfBuildOptionsChanged
Build options changed, rebuilding all
Running: StoreBuildOptionsMap
Running: ContainerMergeCopySketchFiles
Running: SketchSourceMerger
Running: SketchSaver
Running: AdditionalSketchFilesCopier
Running: ContainerFindIncludes
Running: IncludesFinderWithGCC
recipe.preproc.includes pattern is missing
*errors.errorString recipe.preproc.includes pattern is missing
/home/wayoda/dev/arduino/arduino-builder/src/arduino.cc/builder/utils/utils.go:251 (0x451fba)
        ErrorfWithLogger: return errors.Errorf(i18n.Format(format, a...))
/home/wayoda/dev/arduino/arduino-builder/src/arduino.cc/builder/builder_utils/utils.go:164 (0x4d69a1)
        ExecRecipe: return nil, utils.ErrorfWithLogger(logger, constants.MSG_PATTERN_MISSING, recipe)
/home/wayoda/dev/arduino/arduino-builder/src/arduino.cc/builder/includes_finder_with_gcc.go:53 (0x438189)
        (*IncludesFinderWithGCC).Run: output, err := builder_utils.ExecRecipe(properties, constants.RECIPE_PREPROC_INCLUDES, true, verbose, false, logger)
/home/wayoda/dev/arduino/arduino-builder/src/arduino.cc/builder/container_find_includes.go:48 (0x42fd9c)
        (*ContainerFindIncludes).Run: err := command.Run(context)
/home/wayoda/dev/arduino/arduino-builder/src/arduino.cc/builder/builder.go:133 (0x42e64f)
        runCommands: err := command.Run(context)
/home/wayoda/dev/arduino/arduino-builder/src/arduino.cc/builder/builder.go:111 (0x42ddf9)
        (*Builder).Run: return runCommands(context, commands)
/home/wayoda/dev/arduino/arduino-builder/src/arduino.cc/builder/builder.go:160 (0x42ebb0)
        RunBuilder: return command.Run(context)
/home/wayoda/dev/arduino/arduino-builder/main.go:247 (0x402c31)
        main: err = builder.RunBuilder(context)
/usr/lib/go/src/pkg/runtime/proc.c:247 (0x4132aa)
        main: main·main();
/usr/lib/go/src/pkg/runtime/proc.c:1445 (0x415a70)
        goexit: runtime·goexit(void)
```

`grep` can't find a rule `recipe.preproc.includes` in an arduino-1.6.6 (or older ) distribution

Eberhard

ffissore, 2015-09-15:
Every core out there misses that recipe. It's listed in a partial `platform.txt` file. Try downloading the current beta release from http://downloads.arduino.cc/tools/arduino-builder-linux64-1.0.0-beta3.tar.bz2 (or similar url) and be sure your first `-hardware` param is `-hardware ./hardware`

ffissore, 2015-09-15:
For the record, that `hardware` folder is [here](https://github.com/arduino/arduino-builder/tree/master/src/arduino.cc/builder/hardware)

ffissore, 2015-09-15:
And that recipe is not listed in platform.txt, but in `platform.keys.rewrite.txt`. Something I have to fix

wayoda, 2015-09-15:
That beta relase you linked to contains some code that looks for non portable settings.
(see next comment what happens when I switching back to the binary I compiled from the repo)
```
wayoda@shredder:~/sketchbook/empty$ arduino-builder -build-path . -fqbn arduino:avr:nano -hardware /home/wayoda/opt/arduino/hardware -tools /home/wayoda/opt/arduino/tools/ -verbose -debug-level 10 empty.ino 
Running: SetupHumanLoggerIfMissing
Running: CreateBuildPathIfMissing
Running: ContainerSetupHardwareToolsLibsSketchAndProps
Running: AddAdditionalEntriesToContext
Running: RecipeByPrefixSuffixRunner
Looking for recipes like hooks.prebuild*.pattern
Running: HardwareLoader
Running: PlatformKeysRewriteLoader
Running: RewriteHardwareKeys
Running: ToolsLoader
Running: TargetBoardResolver
arduino: Unknown package
*errors.errorString arduino: Unknown package
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/builder/utils/utils.go:251 (0x463115)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/builder/utils/utils.go:243 (0x462eff)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/builder/target_board_resolver.go:54 (0x457915)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/builder/container_setup.go:59 (0x43fff2)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/builder/builder.go:133 (0x43dbad)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/builder/builder.go:111 (0x43d302)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/builder/builder.go:160 (0x43e149)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/main.go:247 (0x402ff8)
/opt/go/src/runtime/proc.go:63 (0x414e83)
/opt/go/src/runtime/asm_amd64.s:2232 (0x43a261)

```

wayoda, 2015-09-15:
Compiled the empty sketch with :
```
wayoda@shredder:~/sketchbook/empty$ arduino-builder -build-path . -fqbn arduino:avr:nano -hardware /home/wayoda/opt/arduino/hardware -hardware /home/wayoda/opt/arduino/arduino-1.6.6 -tools /home/wayoda/opt/arduino/tools/ -tools /home/wayoda/opt/arduino/arduino-1.6.6 -verbose -debug-level 10 empty.ino 
```

Result (The  warning message `misses propertey ....` repeated many times; skip to end for final error:
```
Running: SetupHumanLoggerIfMissing
Running: CreateBuildPathIfMissing
Running: ContainerSetupHardwareToolsLibsSketchAndProps
Running: AddAdditionalEntriesToContext
Running: RecipeByPrefixSuffixRunner
Looking for recipes like hooks.prebuild*.pattern
Running: HardwareLoader
Running: PlatformKeysRewriteLoader
Running: RewriteHardwareKeys
Warning: platform.txt from core '' misses property 'preproc.includes.flags', using default value '-w -x c++ -M -MG -MP'. Consider upgrading this core.
Warning: platform.txt from core '' misses property 'preproc.macros.compatibility_flags', using default value '{build.mbed_api_include} {build.nRF51822_api_include} {build.ble_api_include} {compiler.libsam.c.flags} {compiler.arm.cmsis.path} {build.variant_system_include}'. Consider upgrading this core.
Warning: platform.txt from core '' misses property 'preproc.macros.flags', using default value '-w -x c++ -E -CC'. Consider upgrading this core.
Warning: platform.txt from core '' misses property 'recipe.preproc.includes', using default value '"{compiler.path}{compiler.c.cmd}" {preproc.includes.flags} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.c.extra_flags} {build.extra_flags} "{source_file}"'. Consider upgrading this core.
Warning: platform.txt from core '' misses property 'recipe.preproc.macros', using default value '"{compiler.path}{compiler.cpp.cmd}" {compiler.cpreprocessor.flags} {compiler.cpp.flags} {preproc.macros.flags} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.cpp.extra_flags} {build.extra_flags} {preproc.macros.compatibility_flags} {includes} "{source_file}"'. Consider upgrading this core.
Warning: platform.txt from core '' misses property 'tools.avrdude.path', using default value '{runtime.tools.avrdude.path}'. Consider upgrading this core.
Warning: platform.txt from core '' misses property 'preproc.includes.flags', using default value '-w -x c++ -M -MG -MP'. Consider upgrading this core.
Warning: platform.txt from core '' misses property 'preproc.macros.compatibility_flags', using default value '{build.mbed_api_include} {build.nRF51822_api_include} {build.ble_api_include} {compiler.libsam.c.flags} {compiler.arm.cmsis.path} {build.variant_system_include}'. Consider upgrading this core.
Warning: platform.txt from core '' misses property 'preproc.macros.flags', using default value '-w -x c++ -E -CC'. Consider upgrading this core.
Warning: platform.txt from core '' misses property 'recipe.preproc.includes', using default value '"{compiler.path}{compiler.c.cmd}" {preproc.includes.flags} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.c.extra_flags} {build.extra_flags} "{source_file}"'. Consider upgrading this core.
Warning: platform.txt from core '' misses property 'recipe.preproc.macros', using default value '"{compiler.path}{compiler.cpp.cmd}" {compiler.cpreprocessor.flags} {compiler.cpp.flags} {preproc.macros.flags} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.cpp.extra_flags} {build.extra_flags} {preproc.macros.compatibility_flags} {includes} "{source_file}"'. Consider upgrading this core.
Warning: platform.txt from core '' misses property 'tools.avrdude.path', using default value '{runtime.tools.avrdude.path}'. Consider upgrading this core.
Warning: platform.txt from core '' misses property 'preproc.includes.flags', using default value '-w -x c++ -M -MG -MP'. Consider upgrading this core.
Warning: platform.txt from core '' misses property 'preproc.macros.compatibility_flags', using default value '{build.mbed_api_include} {build.nRF51822_api_include} {build.ble_api_include} {compiler.libsam.c.flags} {compiler.arm.cmsis.path} {build.variant_system_include}'. Consider upgrading this core.
Warning: platform.txt from core '' misses property 'preproc.macros.flags', using default value '-w -x c++ -E -CC'. Consider upgrading this core.
Warning: platform.txt from core '' misses property 'recipe.preproc.includes', using default value '"{compiler.path}{compiler.c.cmd}" {preproc.includes.flags} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.c.extra_flags} {build.extra_flags} "{source_file}"'. Consider upgrading this core.
Warning: platform.txt from core '' misses property 'recipe.preproc.macros', using default value '"{compiler.path}{compiler.cpp.cmd}" {compiler.cpreprocessor.flags} {compiler.cpp.flags} {preproc.macros.flags} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.cpp.extra_flags} {build.extra_flags} {preproc.macros.compatibility_flags} {includes} "{source_file}"'. Consider upgrading this core.
Warning: platform.txt from core '' misses property 'tools.avrdude.path', using default value '{runtime.tools.avrdude.path}'. Consider upgrading this core.
[... cut ...]
/home/wayoda/dev/arduino/arduino-builder/src/arduino.cc/builder/builder.go:111 (0x42ddf9)
        (*Builder).Run: return runCommands(context, commands)
/home/wayoda/dev/arduino/arduino-builder/src/arduino.cc/builder/builder.go:160 (0x42ebb0)
        RunBuilder: return command.Run(context)
/home/wayoda/dev/arduino/arduino-builder/main.go:247 (0x402c31)
        main: err = builder.RunBuilder(context)
/usr/lib/go/src/pkg/runtime/proc.c:247 (0x4132aa)
        main: main·main();
/usr/lib/go/src/pkg/runtime/proc.c:1445 (0x415a70)
        goexit: runtime·goexit(void)

```

ffissore, 2015-09-15:
Given your IDE folder is `/home/wayoda/opt/arduino/arduino-1.6.6`, the correct command line is

```
wayoda@shredder:~/sketchbook/empty$ arduino-builder -build-path . -fqbn arduino:avr:nano:cpu=atmega328 -hardware /home/wayoda/opt/arduino/hardware -hardware /home/wayoda/opt/arduino/arduino-1.6.6/hardware -tools /home/wayoda/opt/arduino/tools/ -tools /home/wayoda/opt/arduino/arduino-1.6.6/hardware/tools -verbose -debug-level 10 empty.ino 
```
Pay attention to the differences between hardware and tools paths and to the additional attribute cpu which, for the nano, is mandatory

wayoda, 2015-09-15:
Your are right, that line worked, thanks

> Pay attention to the ... additional attribute cpu which, for the nano, is mandatory
Would be nice to have a list of easy to remember alias names for this (nano328)

ffissore, 2015-09-15:
Yes, but that would require duplicate almost entirely those board definition, which was what we wanted to avoid when introducing additional properties such as cpu. On the Arduino IDE they result in easy to use menus, which help select whatever additional compilation flag

wayoda, 2015-09-15:
@ffissore 
I don't want to waste your time discussing this minor issue ...

> Yes, but that would require duplicating almost entirely those board definitions

but in the current avr `boards.txt` it would be enough to add a cpu specific property to the board definitions and manage a list of the alias names for looking up the boards

Example
 `nano.menu.cpu.atmega168.alias=nano168` vs `nano.menu.cpu.atmega328.alias=nano328` 

But I can live with the current `-fqbn` too.

ffissore, 2015-09-15:
You should open an issue on https://github.com/arduino/Arduino/, as it's where avr core is developed. It's an improvement and it needs consensus and approval from @cmaglie and @facchinm

closed 2015-09-15 by ffissore

AMKOSHESH, 2017-02-04:
even i had the same problem

Arduino: 1.8.1 (Windows 10), Board: "ATtiny13 (internal 9.6 MHz clock)"

The current selected board needs the core 'arduino:arduino' that is not installed.
Detecting libraries used...
recipe.preproc.macros pattern is missing

Error compiling for board ATtiny13 (internal 9.6 MHz clock).
The current selected board needs the core 'arduino:arduino' that is not installed.

Do you have a comment?

per1234, 2017-02-05:
We're handling @AMKOSHESH's issue on the forum:
http://forum.arduino.cc/index.php?topic=453093