arduino/arduino-builder #5: New builder doesn't like this previously working code

pfeerick, 2015-09-15:
I just tried the Arduino 1.6.6 nightly that I downloaded a few hours ago with some random code, and the new builder didn't like it. I've mutilated the blink example sketch to include the offending code, and the error message is shown below. The release versions of the IDE haven't had any problems with this code... so is this a builder error or a user error :) ?

```
struct sensorData {
  sensorData(int iStatus, float iTemp, float iMinTemp) : status(iStatus), temp(iTemp), minTemp(iMinTemp) {}
  sensorData() : status(-1), temp(1023.0), minTemp(1023.0) {}
  int status;
  float temp;
  float minTemp;
} ;

sensorData sensors[2];

sensorData sensor1; //(-1,1023.0,1023.0);
sensorData sensor2; //(-1,1023.0,1023.0);

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin 13 as an output.
  pinMode(13, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(13, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);              // wait for a second
  digitalWrite(13, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);              // wait for a second
}
```
Compile Error Log
```
Arduino: 1.6.6 Hourly Build 2015/09/14 05:43 (Windows 10), Board: "Arduino/Genuino Uno"

sketch_sep15a:3: error: 'sensorData::sensorData(int, float, float)' cannot be overloaded

sketch_sep15a:3: error: with 'sensorData::sensorData(int, float, float)'

sketch_sep15a:4: error: 'sensorData::sensorData()' cannot be overloaded

sketch_sep15a:4: error: with 'sensorData::sensorData()'

exit status 1
'sensorData::sensorData(int, float, float)' cannot be overloaded
```

ffissore, 2015-09-15:
Confirm. It's a bug. Preprocessed code shows that I'm putting prototypes INSIDE the struct!!!
```c++
#include <Arduino.h>
#line 1
#line 1 "sketch_sep15a.ino"
struct sensorData {
 sensorData(int iStatus, float iTemp, float iMinTemp);
 sensorData();
void setup();
void loop();
#line 2
  sensorData(int iStatus, float iTemp, float iMinTemp) : status(iStatus), temp(iTemp), minTemp(iMinTemp) {}
  sensorData() : status(-1), temp(1023.0), minTemp(1023.0) {}
  int status;
  float temp;
  float minTemp;
} ;
```
I'll ignore prototypes generation for function inside structs

closed 2015-09-15 by ffissore

ffissore, 2015-09-15:
A nightly of the Arduino IDE is on the way. Check it out in 20 mins

pfeerick, 2015-09-16:
I've tried the "2015/0915 04:54" nightly build, can confirm this code works perfectly now. Thank you. btw, have thrown some of my more eclectic Arduino sketches at the latest nightly... and haven't managed to find anything else that breaks it yet ;)

ffissore, 2015-09-16:
:+1: :8ball:

commit 0379be02 (2015-09-15): Making CTagsParser aware of functions in struct, and filtering them out.
Also moving all ctags test outputs in separate files, for easier
maintenance
Fixes #5