arduino/arduino-builder #10: .a linkage for libraries via library.properties

NicoHood, 2015-09-22:
It would be really nice if you could include this fix to the arduino-builder:
https://github.com/arduino/Arduino/pull/3697

Otherwise the widely used HID-Project will not work with newer Arduino versions.
I tried to look at the source code, but there is not a single comment. I have got a bare overview but I have no idea where to start.
I downloaded the source and started the testfiles etc. It compiled but I still have no Idea how to add a library for my testcase, or how to move the arduino builder into the arduino IDE itself once its compiled. The arduino IDE script just downloads the betas of this builder and not a self compiled file.

Sorry but I am lost. I'd really like to help and fix this issue myself but I have to wave my white flag here. Please add this mostly simple feature so I can continue working on the library and test newer IDE versions as well. Right now I am forced to use an outdated self compiled IDE with my .alinkage patch applied.

ffissore, 2015-09-23:
@NicoHood I've started working on it. To test it, I downloaded your latest commit from https://github.com/NicoHood/HID/tree/dev_2_4 but when I try to compile example NKROKeyboard it fails with https://gist.github.com/ffissore/aee6fbaf1a6b9599fdf8

Also, can you please rename `alinkage` key to `dot_a_linkage` ? Otherwise it's harder to read and the "a" may be read as an article

NicoHood, 2015-09-23:
I havent recompiled the examples again after the raw HID testing.
The raw HID example should work with this branch:
https://github.com/NicoHood/HID/tree/6c9e2fd99ce4dc7430ff420d622b5bf9a947230c

The nkro example should compile with this branch:
https://github.com/NicoHood/HID/tree/b617fb0b3d3cb3a3cb55c0262b24f295abdd58a2

If not try to compile another example.
The original PR contains a version of the IDE with my patched version which works. I use a custom IDE with the same patch. you might want to compare this build, and if this works your arduino-builder has a problem, not my branch:
https://github.com/arduino/Arduino/pull/3697#issuecomment-136800598

I can test this the next day again.

ffissore, 2015-09-23:
Compiling another examples leads to the same error message. I'm not c++ expert but I guess you're redefining an already defined type HIDDescDescriptor (found in avr core USBCore.h file)
I'm trying to get to the point where your examples don't compile until I switch from linking object files to linking archive file

ffissore, 2015-09-23:
Done. Tomorrow I'll update the IDE

closed 2015-09-23 by ffissore

NicoHood, 2015-09-24:
Thanks so much!!! I'll test it when I am back home.

I cannot find `HIDDescDescriptor` here: https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/USBCore.h
I guess you found the bug related to this?

ffissore, 2015-09-24:
Yeah, sorry for missing to update you. @facchinm told me that thing is fixed in AVR code on Arduino master branch, but arduino-builder uses avr core version 1.6.7 which is old. I pocked @cmaglie for releasing it. Is it @cmaglie ? :)

NicoHood, 2015-09-24:
Alright thanks. :)
May you please trigger the bot for the hourly build? Its still a build of yesterday.

commit cd9236d3 (2015-09-23): Implemented library linking from archive files rather then object files.
Feature exposed ONLY to libraries with "dot_a_linkage=true" in their
library.properties file
Fixes #10