arduino/arduino-builder #11: Incomplete error message

matthijskooijman, 2015-09-22:
When compiling something, I get:
```
/home/matthijs/docs/Electronics/Arduino/Arduino-1.5/build/linux/work/arduino-builder -dump-prefs -logger=machine -hardware "/home/matthijs/docs/Electronics/Arduino/Arduino-1.5/build/linux/work/hardware" -hardware "/home/matthijs/.arduino15/packages" -hardware "/home/matthijs/docs/Electronics/Arduino/Sketches/hardware" -tools "/home/matthijs/docs/Electronics/Arduino/Arduino-1.5/build/linux/work/tools-builder" -tools "/home/matthijs/docs/Electronics/Arduino/Arduino-1.5/build/linux/work/hardware/tools/avr" -tools "/home/matthijs/.arduino15/packages" -libraries "/home/matthijs/docs/Electronics/Arduino/Sketches/libraries" -libraries "/home/matthijs/docs/Electronics/Arduino/Arduino-1.5/build/linux/work/libraries" -fqbn=arduino:avr:uno -ide-version=10606 -build-path "/tmp/buildd495060d520ba8e8919c653ff08dbe5f.tmp" -warnings=all -prefs=build.warn_data_percentage=75 -verbose "/tmp/arduino_d495060d520ba8e8919c653ff08dbe5f/sketch_sep22a.ino"
Third-party platform.txt does not define compiler.path. Please report this to the third-party hardware maintainer.

Error compiling.
```

(I removed some warnings about libraries that seem unrelated)

This warning is next to useless, since it doesn't specify what platform.txt / package is incorrect. I was compiling for the arduino Uno, so the error is weird in itself, but that's a separate issue (I'll look more closely at that issue tomorrow, let's keep this issue about the error message). I'm not sure what the best way to fix this is in Go, so I'll leave the fix to @ffissore (or whoever). There might be similarly unspecific error messages as well, that also need fixing (didn't check).

matthijskooijman, 2015-09-23:
I wonder if the full path to the platform.txt might be more useful, but I guess you don't have access to that information right there.

ffissore, 2015-09-23:
I do but it would be quite a change in how error messages are reported. That one as well as others came straight from the IDE. Let's keep them this way for some time and see how much feedback comes

matthijskooijman, 2015-09-23:
Looking more closely at the check that causes this error message makes me wonder if it needs to be present at all? Why would arduino-builder enforce having a `compiler.path` variable? It seems this is an implementation detail of the specific platform. If the platform does not want a `compiler.path`, but wants to code the path directly int the recipes, or perhaps have `c-compiler.path` and `cpp-compiler.path` that should also be possible and this check can be entirely removed?

The reason the check is present, I think, is that the compiler used to define `compiler.path`, but now platforms should define it themselves. Instead of checking for compiler.path explicitely, wouldn't it be more useful to check for any undefined placeholders inside recipes? That might catch even more problems, and lead to better error messages (I've seen gcc complain with "invalid mcu specified: {board.mcu}" or something similar when for some reason board.mcu is not defined).

If the check stays, I think it should allow an empty string as a valid value, which it currently does not. This is what caused the error here: I have overridden `compiler.path` to the empty value, so the system-wide avr-gcc is used instead of the bundled one.

ffissore, 2015-09-23:
Uhm yeah, I think it's a legacy from old IDE versions. @cmaglie what do you think? Can you get rid of that check?

cmaglie, 2015-09-23:
For me it can be removed.

This commit contains a bit of history:

cmaglie, 2015-09-23:
(edited the previous comment)

Previously the compiler.path was set by the IDE to the shipped avr-gcc or, if not found, to the system wide avr-gcc. This has been removed with the latest IDE from long time, so I guess the variable is no longer mandatory.

ffissore, 2015-09-23:
Thanks @cmaglie, warning removed

closed 2015-09-23 by ffissore

matthijskooijman, 2015-09-23:
Right, so it was originally a warning and set a default compiler.path value, but it turned into an error with arduino-builder. Seems like a good thing to remove it now indeed. Also, yay for detailed commit messages ;-)

commit 00364c01 (2015-09-23): Better error message when platform.txt misses compiler.path. Fixes #11