arduino/arduino-builder #12: Better library dependencies detection

ffissore, 2015-09-23:
Current implementation of library dependencies detection works only if dependent libraries are included in header files.
If a .cpp file includes a library, that library won't be detected. An example case is SD library version 1.0.5. See https://github.com/arduino-libraries/SD/blob/1.0.5/src/utility/Sd2Card.cpp#L26

A better implementation, suggested in https://github.com/arduino/Arduino/pull/2792, run `gcc -M` against every single library file, not just headers

matthijskooijman, 2015-09-23:
> run gcc -M against every single library file, not just headers

I guess you mean "against every source file (in sketch and libraries), not just the .ino files" ? In general, I think that the library detection should _never_ be run on header files at all, just source files (which might include header files).

The current situation, requiring that a library includes any libraries it needs from its own public header file(s) and not just its source files, seems inelegant (it discourages proper separation of concerns), but it could be considered acceptable if the proper solution runs into performance problems. We'd certainly have to try it before drawing this conclusion, of course.

ffissore, 2015-09-23:
Yes I meant library source files, not its examples .ino

As for running gcc -M against .h or .cpp, I'm not a c++ expert, but my knowledge is that every c++ dev out there uses a different "code philosophy", which may lead to including stuff only in headers, only in cpps, or (more linkely) in both of them. So I think running gcc -M against every source file is fair enough

matthijskooijman, 2015-09-23:
> As for running gcc -M against .h or .cpp, I'm not a c++ expert, but my knowledge is that every c++ dev out there uses a different "code philosophy", which may lead to including stuff only in [...]

If you run gcc -M against source files (.cpp / .c / etc.) then any includes in header files included *by those source files* will be automatically processed as well. Any includes happening in .h files that are never actually included in a source file themselves will be ignored, as I think they should be (since these includes and .h files aren't actually used in the real compilation either). But I think you might be already saying this :-)

commit 0d6e3789 (2015-10-02): Implemented library to library dependency full discovery. Fixes #12
closed 2015-10-02 by ffissore