arduino/arduino-builder #14: Object files in variant folder are not linked

cmaglie, 2015-09-23:
The source files in the variant folder are compiled as usual but not linked in the final step.

For example the Arduino Due core has a variant.cpp file that is compiled but its not included in the link. Compiling the empty sketch gives the following error:

```
/tmp/build000836b13d58999ef1ebfbc27ce3b3be.tmp/core.a(main.cpp.o): In function `main':
/home/megabug/.arduino15/packages/arduino/hardware/sam/1.6.4/cores/arduino/main.cpp:44: warning: undefined reference to `init'
```

because the init() function is defined inside variant.cpp.

ffissore, 2015-09-24:
Argh thank you! I didn't noticed because gcc was issuing a warning and was not failing. Good catch! Thank you!

closed 2015-09-24 by ffissore

cmaglie, 2015-09-24:
Pushed a fix also to the SAM core, now it should emit an error: https://github.com/arduino/Arduino/commit/7d0355c8f15692e4b7d1ac58c96af0059c3325bb

commit aee32ff3 (2015-09-24): core builder: variant files were compiled but not linked. Fixes #14