arduino/arduino-builder #17: Compilation should leverage my quad core laptop power

ffissore, 2015-09-24:
ATM every gcc call is done sequentially, but both library and core compilation can run in parallel

ffissore, 2015-12-01:
A first implementation is available in [parallel](https://github.com/arduino/arduino-builder/tree/parallel) branch. It lacks proper output management. At the moment, everything is thrown at stdout, meaning the output is likely to get clutter and logger messages will eventually contain syntax errors

ffissore, 2015-12-02:
IDE build available at https://github.com/arduino/Arduino/pull/4240

xcvista, 2016-07-10:
I wonder if you can modify arduino-build into emittng a Makefile and call GNU Make with `-j`? Or since it is license compatible cannibalize their code? They already have a working parallelizing mechanism.

PaulStoffregen, 2016-07-10:
arduino-builder does many things make does not.  The first part is an extensive discovery step to determine which libraries are needed... similar in concept to autoconfig and libtool used to make "configure" shell scripts.  Typically stand-alone makefiles (without such systems) are edited to specify library locations, dependencies, etc.

andig, 2016-07-10:
I hope its not inappropriate to say this, bit platformio- while I'm not sure it does parallel builds- is at least incremental and gives me huge productivity plus.

xcvista, 2016-07-10:
@PaulStoffregen `autoconf` and `automake` (heck, even `cmake` does this) emits a Makefile. So why can't arduino-build do the same, emitting a makefile as an intermediary step?

matthijskooijman, 2016-07-10:
> autoconf and automake (heck, even cmake does this) emits a Makefile. So why can't arduino-build do the same, emitting a makefile as an intermediary step?

I think that solving the actual issue (adding parallelism) is a good idea. However, I believe this is reasonably easy to solve within arduino-builder itself, so there really is no need to pull in a dependency on `make`, which would hurt the portability of arduino-builder, as well as make the build  more complex.

So I believe the question is mostly: when will someone find time to implement this? :-)

PaulStoffregen, 2016-07-10:
> So why can't arduino-build do the same, emitting a makefile as an intermediary step?

In an alternate reality where programmers are willing to pour a *lot* of time into implementing this, and where Arduino.cc would be willing to merge large complicated patches that don't really serve their interests yet result in greater long-term maintenance effort, there's really no reason this couldn't be done.  All it would take is a lot of programming work.

Anyone capable of actually doing that work probably wouldn't bother, in the real world where time is precious and where developers capable of such work have many competing demands for their limited hours.  They'd understand this would have much less practical benefit than spending *much* less work to update and integrate the effort already done on the parallel branch.

This is open source, so anyone can fork the code and try to implement whatever they want.  There's really no reason you couldn't emit a makefile mid way through arduino-builder's work, after all libraries are resolved and include paths are known, and the preprocessing has added the extra headers and function prototypes.  Maybe it could even be done fairly cleanly, mostly as code in its own files without much patching in other parts of the code, other than the obvious addition of a command line switch to enable it and quit after the makefile is written.  Whether the Arduino devs would merge such a pull request is a good question.

But that's the sort of question which really can't even be asked in a meaningful way until at least some work has been started in earnest, demonstrating someone is really putting in the necessary effort and giving some realistic view of what the pull request might ultimately look like.

xcvista, 2016-07-10:
@PaulStoffregen Just get rid of the actual invoking of the compiler and emit a few lines into the Makefile. And if the "emit makefile only" flag is missing call `make` after the makefile is generated. You don't need to emit a portable makefile at all.

PaulStoffregen, 2016-07-10:
> Just get rid of the actual invoking of the compiler and emit a few lines into the Makefile.

Ok then, if that's all, why don't *you* go forth with this.  Fork this repository and add the code.  Then invite me and others to give a try.

xcvista, 2016-07-10:
@PaulStoffregen It is not this code that need to be forked, it is the stock arduino-build. I have a language barrier here as I don't write Go code.

PaulStoffregen, 2016-07-10:
Maybe you should come up to speed on Go language?  That's probably a shorter & more successful path than trying to convince other people to spend their time implementing a feature they've said they do not intend to create because it doesn't make practical sense.

xcvista, 2016-07-10:
@PaulStoffregen Meanwhile would you guys offer some internal documentations?

PaulStoffregen, 2016-07-10:
I'm pretty sure the only available documentation, aside from the source code, is this wiki:  (notice this is 1 of the 14 pages)
https://github.com/arduino/Arduino/wiki/Build-Process

xcvista, 2016-07-10:
@PaulStoffregen So I, being a newbie to the Go language, have to reverse engineer an existing codebase. Yeah like that would help a lot.

PaulStoffregen, 2016-07-10:
That's a really creative usage of the term "reverse engineer", which usually means figuring out how something works **without** access to its source code.

xcvista, 2016-07-10:
@PaulStoffregen Backtracking a poorly-documented code base is no less difficult than black box analysis.

PaulStoffregen, 2016-07-10:
This arduino-builder repository contains only 7816 lines in all its source files, not including the automated tests, and *many* of those lines are redundant license headers, comments or white space.  It's simply not a huge code base.  Almost file and function has very descriptive names, like "generate_buildpath_if_missing.go".  There *is* a 14 page wiki which does document the data formats and general ideas of the build process.

Calling this project poorly-documented sounds to me like whining.

xcvista, 2016-07-10:
@PaulStoffregen All while I personally have zero days of Go knowledge.

NicoHood, 2016-07-10:
I miss @ffissore :(

andig, 2016-07-10:
@xcvista imho the discussion is a little inappropriate. As Paul has said: its open source, go fix it or leave it if the developers have other priorities. If thats beyond your skills too bad. You may also have chances offering a bounty to make your contribution to the project.

Or: give platformio a try, it has parallel builds and loads of other things that Arduino doesn't offer.

xcvista, 2016-09-21:
Coming to think of it, if moving to using Makefiles as an intermediary step, we need and only need the discovery step to find all the libraries. For my personal preference I would first assemble a pre-build source tree that have all core and library code files linked or copied over, throw a generic portable makefile in there and exec `make`.

commit a581d44a (2015-11-27): Parallel compiling: arduino-builder will leverage your multi core pc. If
-jobs X param is specified and X is greater than 0, arduino-builder won't spawn
more than X processes. Fixes #17
commit e29cba25 (2015-11-30): Parallel compiling: arduino-builder will leverage your multi core pc. If
-jobs X param is specified and X is greater than 0, arduino-builder won't spawn
more than X processes. Fixes #17
commit a9fd0501 (2015-12-01): Parallel compiling: arduino-builder will leverage your multi core pc. If
-jobs X param is specified and X is greater than 0, arduino-builder won't spawn
more than X processes. Fixes #17
commit 0e4410d9 (2015-12-02): External processes output is printed on stdout/stderr in a synchronized way.
Thus we are sure that those outputs do not interfere with our own output,
otherwise leading to wrongly formatted log messages. Fixes #17
closed 2018-05-31 by cmaglie