arduino/arduino-builder #18: USBCON not definied with a different boards.txt

NicoHood, 2015-09-24:
Using the board definitions for another board does not define USBCON and you get a compile error for the HID-Project:
https://github.com/NicoHood/HID/tree/dev_2_4
https://github.com/NicoHood/HoodLoader2

This also applies for 32u4 (mostly a copy of leonardo, just different upload speed).

ffissore, 2015-09-24:
Please give a second read to https://github.com/arduino/arduino-builder/blob/master/CONTRIBUTING.md#submitting-an-issue and provide proper debug output

NicoHood, 2015-09-24:
Oh sorry I missed that first important line!

```
/Arduino/Arduino/build/linux/work/arduino-builder -dump-prefs -logger=machine -hardware "/Arduino/Arduino/build/linux/work/hardware" -hardware "/home/linuxuser/.arduino15/packages" -hardware "/Arduino/sketchbook/hardware" -tools "/Arduino/Arduino/build/linux/work/tools-builder" -tools "/Arduino/Arduino/build/linux/work/hardware/tools/avr" -tools "/home/linuxuser/.arduino15/packages" -libraries "/Arduino/sketchbook/libraries" -libraries "/Arduino/Arduino/build/linux/work/libraries" -fqbn=arduino:avr:leonardo -ide-version=10606 -build-path "/tmp/build7b899e09f7eaf8de1453078e67d99ff6.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose -debug-level=10 "/tmp/arduino_7b899e09f7eaf8de1453078e67d99ff6/sketch_sep24a.ino"
===Running: {0} ||| [SetupHumanLoggerIfMissing]
===Progress {0} ||| [0.00]
===Running: {0} ||| [GenerateBuildPathIfMissing]
===Progress {0} ||| [25.00]
===Running: {0} ||| [ContainerSetupHardwareToolsLibsSketchAndProps]
===Progress {0} ||| [50.00]
===Running: {0} ||| [AddAdditionalEntriesToContext]
===Running: {0} ||| [FailIfBuildPathEqualsSketchPath]
===Running: {0} ||| [HardwareLoader]
===Running: {0} ||| [PlatformKeysRewriteLoader]
===Running: {0} ||| [RewriteHardwareKeys]
===Running: {0} ||| [ToolsLoader]
===Running: {0} ||| [TargetBoardResolver]
===Running: {0} ||| [AddBuildBoardPropertyIfMissing]
===Running: {0} ||| [LibrariesLoader]
===WARNING: Spurious {0} folder in '{1}' library ||| [.git AnalogTouch]
===WARNING: Spurious {0} folder in '{1}' library ||| [.git HID-Project]
===WARNING: Spurious {0} folder in '{1}' library ||| [.git IRLremote]
===WARNING: Spurious {0} folder in '{1}' library ||| [.git MSGEQ7]
===WARNING: Spurious {0} folder in '{1}' library ||| [.git Nintendo]
===WARNING: Spurious {0} folder in '{1}' library ||| [.git PinChangeInterrupt]
===WARNING: Spurious {0} folder in '{1}' library ||| [.git RCLSwitch]
===Running: {0} ||| [SketchLoader]
===Running: {0} ||| [SetupBuildProperties]
===Running: {0} ||| [LoadVIDPIDSpecificProperties]
===Running: {0} ||| [SetCustomBuildProperties]
===Running: {0} ||| [DumpBuildProperties]
===Progress {0} ||| [75.00]
_id=leonardo
bootloader.extended_fuses=0xcb
bootloader.file=caterina/Caterina-Leonardo.hex
bootloader.high_fuses=0xd8
bootloader.lock_bits=0x2F
bootloader.low_fuses=0xff
bootloader.tool=avrdude
bootloader.unlock_bits=0x3F
build.arch=AVR
build.board=AVR_LEONARDO
build.core=arduino
build.core.path=/Arduino/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino
build.extra_flags={build.usb_flags}
build.f_cpu=16000000L
build.mcu=atmega32u4
build.path=/tmp/build7b899e09f7eaf8de1453078e67d99ff6.tmp
build.pid=0x8036
build.project_name=sketch_sep24a.ino
build.source.path=/tmp/arduino_7b899e09f7eaf8de1453078e67d99ff6/sketch_sep24a.ino
build.system.path=/Arduino/Arduino/build/linux/work/hardware/arduino/avr/system
build.usb_flags=-DUSB_VID={build.vid} -DUSB_PID={build.pid} '-DUSB_MANUFACTURER={build.usb_manufacturer}' '-DUSB_PRODUCT={build.usb_product}'
build.usb_manufacturer="Unknown"
build.usb_product="Arduino Leonardo"
build.variant=leonardo
build.variant.path=/Arduino/Arduino/build/linux/work/hardware/arduino/avr/variants/leonardo
build.vid=0x2341
build.warn_data_percentage=75
compiler.S.extra_flags=
compiler.S.flags=-c -g -x assembler-with-cpp
compiler.ar.cmd=avr-ar
compiler.ar.extra_flags=
compiler.ar.flags=rcs
compiler.c.cmd=avr-gcc
compiler.c.elf.cmd=avr-gcc
compiler.c.elf.extra_flags=
compiler.c.elf.flags={compiler.warning_flags} -Os -Wl,--gc-sections
compiler.c.extra_flags=
compiler.c.flags=-c -g -Os {compiler.warning_flags} -std=gnu11 -ffunction-sections -fdata-sections -MMD
[... 541 lines cut ...]
 #error "This is not an USB AVR or you use an old version of the IDE."
  ^
In file included from /Arduino/sketchbook/libraries/HID-Project/src/HID-Project.h:138:0,
                 from sketch_sep24a.ino:1:
/Arduino/sketchbook/libraries/HID-Project/src/Gamepad.h:30:2: error: #error "This is not an USB AVR or you use an old version of the IDE."
 #error "This is not an USB AVR or you use an old version of the IDE."
  ^
In file included from /Arduino/sketchbook/libraries/HID-Project/src/HID-Project.h:139:0,
                 from sketch_sep24a.ino:1:
/Arduino/sketchbook/libraries/HID-Project/src/System.h:30:2: error: #error "This is not an USB AVR or you use an old version of the IDE."
 #error "This is not an USB AVR or you use an old version of the IDE."
  ^
exit status 1
*exec.ExitError exit status 1
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/builder/utils/utils.go:248 (0x464992)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/builder/builder_utils/utils.go:221 (0x4f9853)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/builder/includes_finder_with_gcc.go:63 (0x449e20)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/builder/container_find_includes.go:51 (0x4405d5)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/builder/builder.go:142 (0x43ebdd)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/builder/builder.go:115 (0x43e1b7)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/builder/builder.go:169 (0x43f179)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/main.go:297 (0x4032ac)
/opt/go/src/runtime/proc.go:63 (0x415933)
/opt/go/src/runtime/asm_amd64.s:2232 (0x43ad11)
```

ffissore, 2015-09-25:
Uhmm I can't reproduce: branch dev_2_4 contains a library, boards from Hoodloader2 say they require core HID which is not installed

NicoHood, 2015-09-25:
You need the dev version of HoodLoader2:
https://github.com/NicoHood/HoodLoader2/tree/fast-usbserial
Or the most bleeding edge (which will be leonardo compatible again)
https://github.com/NicoHood/HoodLoader2/tree/dev_205

ffissore, 2015-09-25:
I've cloned hooloader2 and switched to branch dev_205. Selected  32u2 and 32u4 and compiled the default empty sketch. If succeeded. See https://gist.github.com/ffissore/58761ba357afe05fb634

NicoHood, 2015-09-25:
You need to add `#include "HID-Project.h"` to the sketch. Then the error will pop up. The dev version (2.4) is required for HID Project.

I will check this out again when I am back home.

NicoHood, 2015-09-25:
```cpp
#include "HID-Project.h"
void setup() {}
void loop() {}
```
Happens again with the latest git. As written above ` -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics` is missing in the comand line if you look at the output above.

closed 2015-09-28 by ffissore

NicoHood, 2015-09-28:
Looking forward to test beta 16 :)

ffissore, 2015-09-28:
out

NicoHood, 2015-09-28:
Seems to work.

commit c90dedff (2015-09-28): Adding recipes from parent platforms only when build properties are set up.
Fixes #18