arduino/arduino-builder #19: Arduino-Builder not reloading .h files correct

NicoHood, 2015-09-26:
I guess this is related to the builder now. I've noticed this before as well, but now I have a testcase.

See this commit:
https://github.com/NicoHood/Arduino/commit/ed03da415cdb5415a1bca2c3c11789602505a3c2

First compilation:
```
/Arduino/build/linux/work/arduino-builder -dump-prefs -logger=machine -hardware "/Arduino/build/linux/work/hardware" -hardware "/home/linuxuser/.arduino15/packages" -hardware "/sketchbook/hardware" -tools "/Arduino/build/linux/work/tools-builder" -tools "/Arduino/build/linux/work/hardware/tools/avr" -tools "/home/linuxuser/.arduino15/packages" -libraries "/sketchbook/libraries" -libraries "/Arduino/build/linux/work/libraries" -fqbn=arduino:avr:leonardo -ide-version=10606 -build-path "/tmp/buildfe5726cdd8d5583bec28b2c8a5616d9b.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/tmp/untitled439604977.tmp/sketch_sep26a/sketch_sep26a.ino"
/Arduino/build/linux/work/arduino-builder -compile -logger=machine -hardware "/Arduino/build/linux/work/hardware" -hardware "/home/linuxuser/.arduino15/packages" -hardware "/sketchbook/hardware" -tools "/Arduino/build/linux/work/tools-builder" -tools "/Arduino/build/linux/work/hardware/tools/avr" -tools "/home/linuxuser/.arduino15/packages" -libraries "/sketchbook/libraries" -libraries "/Arduino/build/linux/work/libraries" -fqbn=arduino:avr:leonardo -ide-version=10606 -build-path "/tmp/buildfe5726cdd8d5583bec28b2c8a5616d9b.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/tmp/untitled439604977.tmp/sketch_sep26a/sketch_sep26a.ino"
WARNING: Spurious .git folder in 'AnalogTouch' library
WARNING: Spurious .git folder in 'FastLED' library
WARNING: Spurious .git folder in 'HID-Project' library
WARNING: Spurious .git folder in 'IRLremote' library
WARNING: Spurious .git folder in 'MSGEQ7' library
WARNING: Spurious .git folder in 'Nintendo' library
WARNING: Spurious .git folder in 'PinChangeInterrupt' library
WARNING: Spurious .git folder in 'RCLSwitch' library
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"'  "/tmp/buildfe5726cdd8d5583bec28b2c8a5616d9b.tmp/sketch/sketch_sep26a.ino.cpp"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/Arduino/build/linux/work/hardware/arduino/avr/variants/leonardo" "/tmp/buildfe5726cdd8d5583bec28b2c8a5616d9b.tmp/sketch/sketch_sep26a.ino.cpp"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/Arduino/build/linux/work/hardware/arduino/avr/variants/leonardo" "/tmp/buildfe5726cdd8d5583bec28b2c8a5616d9b.tmp/sketch/sketch_sep26a.ino.cpp"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/Arduino/build/linux/work/hardware/arduino/avr/variants/leonardo" "/tmp/buildfe5726cdd8d5583bec28b2c8a5616d9b.tmp/sketch/sketch_sep26a.ino.cpp"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/Arduino/build/linux/work/hardware/arduino/avr/variants/leonardo" "/tmp/buildfe5726cdd8d5583bec28b2c8a5616d9b.tmp/sketch/sketch_sep26a.ino.cpp"
"/Arduino/build/linux/work/tools-builder/ctags/5.8-patched/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns "/tmp/buildfe5726cdd8d5583bec28b2c8a5616d9b.tmp/preproc/ctags_target.cpp"
"/Arduino/build/linux/work/tools-builder/ctags/5.8-patched/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns "/tmp/buildfe5726cdd8d5583bec28b2c8a5616d9b.tmp/preproc/ctags_target.cpp"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/Arduino/build/linux/work/hardware/arduino/avr/variants/leonardo" "/tmp/buildfe5726cdd8d5583bec28b2c8a5616d9b.tmp/sketch/sketch_sep26a.ino.cpp" -o "/tmp/buildfe5726cdd8d5583bec28b2c8a5616d9b.tmp/sketch/sketch_sep26a.ino.cpp.o"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-gcc" -c -g -x assembler-with-cpp -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/Arduino/build/linux/work/hardware/arduino/avr/variants/leonardo" "/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino/wiring_pulse.S" -o "/tmp/buildfe5726cdd8d5583bec28b2c8a5616d9b.tmp/core/wiring_pulse.S.o"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/Arduino/build/linux/work/hardware/arduino/avr/variants/leonardo" "/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino/WInterrupts.c" -o "/tmp/buildfe5726cdd8d5583bec28b2c8a5616d9b.tmp/core/WInterrupts.c.o"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/Arduino/build/linux/work/hardware/arduino/avr/variants/leonardo" "/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino/hooks.c" -o "/tmp/buildfe5726cdd8d5583bec28b2c8a5616d9b.tmp/core/hooks.c.o"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/Arduino/build/linux/work/hardware/arduino/avr/variants/leonardo" "/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino/wiring.c" -o "/tmp/buildfe5726cdd8d5583bec28b2c8a5616d9b.tmp/core/wiring.c.o"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Ardu [... cut ...] ketch_sep26a.ino.eep"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-objcopy" -O ihex -R .eeprom  "/tmp/buildfe5726cdd8d5583bec28b2c8a5616d9b.tmp/sketch_sep26a.ino.elf" "/tmp/buildfe5726cdd8d5583bec28b2c8a5616d9b.tmp/sketch_sep26a.ino.hex"

Sketch uses 3,746 bytes (13%) of program storage space. Maximum is 28,672 bytes.
Global variables use 139 bytes (5%) of dynamic memory, leaving 2,421 bytes for local variables. Maximum is 2,560 bytes.

```

Note that the last recompile happens instant.

NicoHood, 2015-09-26:
If you compare those 3 commands, they are all the same:
```
/Arduino/build/linux/work/arduino-builder -dump-prefs -logger=machine -hardware "/Arduino/build/linux/work/hardware" -hardware "/home/linuxuser/.arduino15/packages" -hardware "/sketchbook/hardware" -tools "/Arduino/build/linux/work/tools-builder" -tools "/Arduino/build/linux/work/hardware/tools/avr" -tools "/home/linuxuser/.arduino15/packages" -libraries "/sketchbook/libraries" -libraries "/Arduino/build/linux/work/libraries" -fqbn=arduino:avr:leonardo -ide-version=10606 -build-path "/tmp/buildfe5726cdd8d5583bec28b2c8a5616d9b.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/tmp/untitled439604977.tmp/sketch_sep26a/sketch_sep26a.ino"
/Arduino/build/linux/work/arduino-builder -compile -logger=machine -hardware "/Arduino/build/linux/work/hardware" -hardware "/home/linuxuser/.arduino15/packages" -hardware "/sketchbook/hardware" -tools "/Arduino/build/linux/work/tools-builder" -tools "/Arduino/build/linux/work/hardware/tools/avr" -tools "/home/linuxuser/.arduino15/packages" -libraries "/sketchbook/libraries" -libraries "/Arduino/build/linux/work/libraries" -fqbn=arduino:avr:leonardo -ide-version=10606 -build-path "/tmp/buildfe5726cdd8d5583bec28b2c8a5616d9b.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/tmp/untitled439604977.tmp/sketch_sep26a/sketch_sep26a.ino"

/Arduino/build/linux/work/arduino-builder -dump-prefs -logger=machine -hardware "/Arduino/build/linux/work/hardware" -hardware "/home/linuxuser/.arduino15/packages" -hardware "/sketchbook/hardware" -tools "/Arduino/build/linux/work/tools-builder" -tools "/Arduino/build/linux/work/hardware/tools/avr" -tools "/home/linuxuser/.arduino15/packages" -libraries "/sketchbook/libraries" -libraries "/Arduino/build/linux/work/libraries" -fqbn=arduino:avr:leonardo -ide-version=10606 -build-path "/tmp/buildfe5726cdd8d5583bec28b2c8a5616d9b.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/tmp/untitled439604977.tmp/sketch_sep26a/sketch_sep26a.ino"
/Arduino/build/linux/work/arduino-builder -compile -logger=machine -hardware "/Arduino/build/linux/work/hardware" -hardware "/home/linuxuser/.arduino15/packages" -hardware "/sketchbook/hardware" -tools "/Arduino/build/linux/work/tools-builder" -tools "/Arduino/build/linux/work/hardware/tools/avr" -tools "/home/linuxuser/.arduino15/packages" -libraries "/sketchbook/libraries" -libraries "/Arduino/build/linux/work/libraries" -fqbn=arduino:avr:leonardo -ide-version=10606 -build-path "/tmp/buildfe5726cdd8d5583bec28b2c8a5616d9b.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/tmp/untitled439604977.tmp/sketch_sep26a/sketch_sep26a.ino"

/Arduino/build/linux/work/arduino-builder -dump-prefs -logger=machine -hardware "/Arduino/build/linux/work/hardware" -hardware "/home/linuxuser/.arduino15/packages" -hardware "/sketchbook/hardware" -tools "/Arduino/build/linux/work/tools-builder" -tools "/Arduino/build/linux/work/hardware/tools/avr" -tools "/home/linuxuser/.arduino15/packages" -libraries "/sketchbook/libraries" -libraries "/Arduino/build/linux/work/libraries" -fqbn=arduino:avr:leonardo -ide-version=10606 -build-path "/tmp/buildfe5726cdd8d5583bec28b2c8a5616d9b.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/tmp/untitled439604977.tmp/sketch_sep26a/sketch_sep26a.ino"
/Arduino/build/linux/work/arduino-builder -compile -logger=machine -hardware "/Arduino/build/linux/work/hardware" -hardware "/home/linuxuser/.arduino15/packages" -hardware "/sketchbook/hardware" -tools "/Arduino/build/linux/work/tools-builder" -tools "/Arduino/build/linux/work/hardware/tools/avr" -tools "/home/linuxuser/.arduino15/packages" -libraries "/sketchbook/libraries" -libraries "/Arduino/build/linux/work/libraries" -fqbn=arduino:avr:leonardo -ide-version=10606 -build-path "/tmp/buildfe5726cdd8d5583bec28b2c8a5616d9b.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/tmp/untitled439604977.tmp/sketch_sep26a/sketch_sep26a.ino"
```

NicoHood, 2015-09-26:
And now the detailed output in the same order as above:
```
/Arduino/build/linux/work/arduino-builder -dump-prefs -logger=machine -hardware "/Arduino/build/linux/work/hardware" -hardware "/home/linuxuser/.arduino15/packages" -hardware "/sketchbook/hardware" -tools "/Arduino/build/linux/work/tools-builder" -tools "/Arduino/build/linux/work/hardware/tools/avr" -tools "/home/linuxuser/.arduino15/packages" -libraries "/sketchbook/libraries" -libraries "/Arduino/build/linux/work/libraries" -fqbn=arduino:avr:leonardo -ide-version=10606 -build-path "/tmp/build69e5c36745d922c4f42be303e81a4359.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose -debug-level=10 "/tmp/untitled1835095756.tmp/sketch_sep26a/sketch_sep26a.ino"
===Running: {0} ||| [SetupHumanLoggerIfMissing]
===Progress {0} ||| [0.00]
===Running: {0} ||| [GenerateBuildPathIfMissing]
===Progress {0} ||| [25.00]
===Running: {0} ||| [ContainerSetupHardwareToolsLibsSketchAndProps]
===Progress {0} ||| [50.00]
===Running: {0} ||| [AddAdditionalEntriesToContext]
===Running: {0} ||| [FailIfBuildPathEqualsSketchPath]
===Running: {0} ||| [HardwareLoader]
===Running: {0} ||| [PlatformKeysRewriteLoader]
===Running: {0} ||| [RewriteHardwareKeys]
===Running: {0} ||| [ToolsLoader]
===Running: {0} ||| [TargetBoardResolver]
===Running: {0} ||| [AddBuildBoardPropertyIfMissing]
===Running: {0} ||| [LibrariesLoader]
===WARNING: Spurious {0} folder in '{1}' library ||| [.git AnalogTouch]
===WARNING: Spurious {0} folder in '{1}' library ||| [.git FastLED]
===WARNING: Spurious {0} folder in '{1}' library ||| [.git HID-Project]
===WARNING: Spurious {0} folder in '{1}' library ||| [.git IRLremote]
===WARNING: Spurious {0} folder in '{1}' library ||| [.git MSGEQ7]
===WARNING: Spurious {0} folder in '{1}' library ||| [.git Nintendo]
===WARNING: Spurious {0} folder in '{1}' library ||| [.git PinChangeInterrupt]
===WARNING: Spurious {0} folder in '{1}' library ||| [.git RCLSwitch]
===Running: {0} ||| [SketchLoader]
===Running: {0} ||| [SetupBuildProperties]
===Running: {0} ||| [LoadVIDPIDSpecificProperties]
===Running: {0} ||| [SetCustomBuildProperties]
===Running: {0} ||| [DumpBuildProperties]
===Progress {0} ||| [75.00]
_id=leonardo
bootloader.extended_fuses=0xcb
bootloader.file=caterina/Caterina-Leonardo.hex
bootloader.high_fuses=0xd8
bootloader.lock_bits=0x2F
bootloader.low_fuses=0xff
bootloader.tool=avrdude
bootloader.unlock_bits=0x3F
build.arch=AVR
build.board=AVR_LEONARDO
build.core=arduino
build.core.path=/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino
build.extra_flags={build.usb_flags}
build.f_cpu=16000000L
build.mcu=atmega32u4
build.path=/tmp/build69e5c36745d922c4f42be303e81a4359.tmp
build.pid=0x8036
build.project_name=sketch_sep26a.ino
build.source.path=/tmp/untitled1835095756.tmp/sketch_sep26a/sketch_sep26a.ino
build.system.path=/Arduino/build/linux/work/hardware/arduino/avr/system
build.usb_flags=-DUSB_VID={build.vid} -DUSB_PID={build.pid} '-DUSB_MANUFACTURER={build.usb_manufacturer}' '-DUSB_PRODUCT={build.usb_product}'
build.usb_manufacturer="Unknown"
build.usb_product="Arduino Leonardo"
build.variant=leonardo
build.variant.path=/Arduino/build/linux/work/hardware/arduino/avr/variants/leonardo
build.vid=0x2341
build.warn_data_percentage=75
compiler.S.extra_flags=
compiler.S.flags=-c -g -x assembler-with-cpp
compiler.ar.cmd=avr-ar
compiler.ar.extra_flags=
compiler.ar.flags=rcs
compiler.c.cmd=avr-gcc
compiler.c.elf.cmd=avr-gcc
compiler.c.elf.extra_flags=
compiler.c.elf.flags={compiler.warning_flags} -Os -Wl,--gc-sections
compiler.c.extra_flags=
compiler.c.flags=-c -g -Os {compiler.warning_flags} -std=gnu11 -ffunction-sections -fdata-sections -MMD
[... 913 lines cut ...]
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-gcc" -w -Os -Wl,--gc-sections -mmcu=atmega32u4  -o "/tmp/build69e5c36745d922c4f42be303e81a4359.tmp/sketch_sep26a.ino.elf" "/tmp/build69e5c36745d922c4f42be303e81a4359.tmp/sketch/sketch_sep26a.ino.cpp.o" "/tmp/build69e5c36745d922c4f42be303e81a4359.tmp/core/core.a" "-L/tmp/build69e5c36745d922c4f42be303e81a4359.tmp" -lm
===Running: {0} ||| [RecipeByPrefixSuffixRunner]
===Progress {0} ||| [78.57]
===Looking for recipes like {0}*{1} ||| [recipe.hooks.linking.postlink .pattern]
===Running: {0} ||| [RecipeByPrefixSuffixRunner]
===Progress {0} ||| [82.14]
===Looking for recipes like {0}*{1} ||| [recipe.hooks.objcopy.preobjcopy .pattern]
===Running: {0} ||| [RecipeByPrefixSuffixRunner]
===Progress {0} ||| [85.71]
===Looking for recipes like {0}*{1} ||| [recipe.objcopy. .pattern]
===Running recipe: {0} ||| [recipe.objcopy.eep.pattern]
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-objcopy" -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0  "/tmp/build69e5c36745d922c4f42be303e81a4359.tmp/sketch_sep26a.ino.elf" "/tmp/build69e5c36745d922c4f42be303e81a4359.tmp/sketch_sep26a.ino.eep"
===Running recipe: {0} ||| [recipe.objcopy.hex.pattern]
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-objcopy" -O ihex -R .eeprom  "/tmp/build69e5c36745d922c4f42be303e81a4359.tmp/sketch_sep26a.ino.elf" "/tmp/build69e5c36745d922c4f42be303e81a4359.tmp/sketch_sep26a.ino.hex"
===Running: {0} ||| [RecipeByPrefixSuffixRunner]
===Progress {0} ||| [89.29]
===Looking for recipes like {0}*{1} ||| [recipe.hooks.objcopy.postobjcopy .pattern]
===Running: {0} ||| [MergeSketchWithBootloader]
===Progress {0} ||| [92.86]
===Running: {0} ||| [RecipeByPrefixSuffixRunner]
===Progress {0} ||| [96.43]
===Looking for recipes like {0}*{1} ||| [recipe.hooks.postbuild .pattern]
===Progress {0} ||| [100.00]

```

ffissore, 2015-09-28:
Those files reside in `hardware` folder of IDE repo, which is not the one used by the IDE when compile. You must issue a `ant clean build` to copy the modified files to the actual hardware folder used by the IDE. Can you double check?

Also, please @NicoHood, I'm super happy about your issues, they are helping me solving a number of issues, but please post your dumps on http://gist.github.com/. My mouse scroll wheel is burning!

NicoHood, 2015-09-28:
Whoops, yes I forgot about gist.github.

I edited the files inside linux/work which ARE used bz the IDE. Its not a rebuilding issue. I want to modify the core at runtime for debugging (as always). Or do you want to say I *need to* rebuild the ide for every core modification? That sounds weird.

If I had edited the wrong files no error would occur because the wrong files dont cause any changes. I dont know if you missunderstood, but we are not talking about recompiling the IDE, its just opening the ide and modifying the core at runtime. This also happens with mc HID-Project library. Not sure if this is caused by the .a linkage, but could be. I dont have a testcase for the HID Project library but for the core as linked above.

ffissore, 2015-09-28:
I already got that this is not about rebuilding the IDE: I wanted you to double check you were modifying the right files. Can you please provide a smaller test case? Something that doesn't rely on your repo which is under development?

NicoHood, 2015-09-28:
The repo is not under development, i stopped because of that. I also linked the commit just for this testcase.

I just edited the core files as usual.

Other testcase:
Take the HID-Project 2.4 and comment out a member of the HID class:
https://github.com/NicoHood/HID/blob/dev_2_4/src/PluggableHID/HID.h#L84

This causes the same issue. Tested with and without .a linkage. The .h file change is not recognized as update and the depending .cpp files are not recompiled.

ffissore, 2015-09-29:
I think I found the root cause, which is a poor implementation of ["already compiled"](https://github.com/arduino/arduino-builder/blob/afdee0d94892349dcecfd0aaa900b3bc7b1f3951/src/arduino.cc/builder/builder_utils/utils.go#L166-L166) check. I'm working on it

closed 2015-09-29 by ffissore

NicoHood, 2015-09-29:
Seems to be fixed now. thanks!

commit 3a614d72 (2015-09-29): A better implementation of ObjFileIsUpToDate, ported from OldCompiler
Fixes #19