arduino/arduino-builder #20: Wrong Bootloader not present warning

ffissore, 2015-09-28:
_From @NicoHood on September 26, 2015 8:16_

If I compile a sketch for my custom HoodLoader2 core I now get a warning that the .hex file is missing. However a ISP programming still works. The IDE looks at the internal path, not the hardwarefolder where the HoodLoader2 is installed.
https://github.com/NicoHood/HoodLoader2/tree/dev_205

Also I suggest to somehow access the default bootloaders. For example I added a copy of the Uno/Mega board to add the Serial 1200 baud touch feature. But I still want to use the optiboot bootloader. At the moment I have to put a copy of this hexfile in my repository which I do not prefer since this can be avoided with `arduino:optiboothexfilepath`.

IDE Warning (when hitting compile):
```
/Arduino/build/linux/work/arduino-builder -dump-prefs -logger=machine -hardware "/Arduino/build/linux/work/hardware" -hardware "/home/linuxuser/.arduino15/packages" -hardware "/sketchbook/hardware" -tools "/Arduino/build/linux/work/tools-builder" -tools "/Arduino/build/linux/work/hardware/tools/avr" -tools "/home/linuxuser/.arduino15/packages" -libraries "/sketchbook/libraries" -libraries "/Arduino/build/linux/work/libraries" -fqbn=HoodLoader2:avr:HoodLoader2atmega16u2:board=uno -ide-version=10606 -build-path "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/tmp/untitled1542786869.tmp/sketch_sep26a/sketch_sep26a.ino"
/Arduino/build/linux/work/arduino-builder -compile -logger=machine -hardware "/Arduino/build/linux/work/hardware" -hardware "/home/linuxuser/.arduino15/packages" -hardware "/sketchbook/hardware" -tools "/Arduino/build/linux/work/tools-builder" -tools "/Arduino/build/linux/work/hardware/tools/avr" -tools "/home/linuxuser/.arduino15/packages" -libraries "/sketchbook/libraries" -libraries "/Arduino/build/linux/work/libraries" -fqbn=HoodLoader2:avr:HoodLoader2atmega16u2:board=uno -ide-version=10606 -build-path "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/tmp/untitled1542786869.tmp/sketch_sep26a/sketch_sep26a.ino"
WARNING: Spurious .git folder in 'AnalogTouch' library
WARNING: Spurious .git folder in 'HID-Project' library
WARNING: Spurious .git folder in 'IRLremote' library
WARNING: Spurious .git folder in 'MSGEQ7' library
WARNING: Spurious .git folder in 'Nintendo' library
WARNING: Spurious .git folder in 'PinChangeInterrupt' library
WARNING: Spurious .git folder in 'RCLSwitch' library
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++" -w -x c++ -M -MG -MP -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16  "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp/sketch/sketch_sep26a.ino.cpp"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++" -w -x c++ -M -MG -MP -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-I/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/sketchbook/hardware/HoodLoader2/avr/variants/HoodLoader2" "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp/sketch/sketch_sep26a.ino.cpp"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++" -w -x c++ -M -MG -MP -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-I/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/sketchbook/hardware/HoodLoader2/avr/variants/HoodLoader2" "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp/sketch/sketch_sep26a.ino.cpp"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++" -w -x c++ -M -MG -MP -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-I/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/sketchbook/hardware/HoodLoader2/avr/variants/HoodLoader2" "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp/sketch/sketch_sep26a.ino.cpp"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++"  -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16       "-I/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/sketchbook/hardware/HoodLoader2/avr/variants/HoodLoader2" "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp/sketch/sketch_sep26a.ino.cpp"
"/Arduino/build/linux/work/tools-builder/ctags/5.8-patched/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp/preproc/ctags_target.cpp"
"/Arduino/build/linux/work/tools-builder/ctags/5.8-patched/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp/preproc/ctags_target.cpp"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega16u2 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-I/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/sketchbook/hardware/HoodLoader2/avr/variants/HoodLoader2" "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp/sketch/sketch_sep26a.ino.cpp" -o "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp/sketch/sketch_sep26a.ino.cpp.o"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-gcc" -c -g -x assembler-with-cpp -mmcu=atmega16u2 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-I/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/sketchbook/hardware/HoodLoader2/avr/variants/HoodLoader2" "/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino/wiring_pulse.S" -o "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp/core/wiring_pulse.S.o"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega16u2 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-I/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/sketchbook/hardware/HoodLoader2/avr/variants/HoodLoader2" "/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino/WInterrupts.c" -o "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp/core/WInterrupts.c.o"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega16u2 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-I/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/sketchbook/hardware/HoodLoader2/avr/variants/HoodLoader2" "/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino/hooks.c" -o "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp/core/hooks.c.o"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega16u2 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-I/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/sketchbook/hardware/HoodLoader2/avr/variants/HoodLoader2" "/Arduino/build/linux/work/hardware/arduino [... cut ...]
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.06s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x0F:
avrdude: load data lock data from input file 0x0F:
avrdude: input file 0x0F contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...
avrdude: 1 bytes of lock verified

avrdude done.  Thank you.
```

_Copied from original issue: arduino/Arduino#3869_

ffissore, 2015-09-28:
_From @per1234 on September 26, 2015 11:49_

> I now get a warning that the .hex file is missing.

See https://github.com/arduino/Arduino/issues/3831
> I suggest to somehow access the default bootloaders.

https://github.com/arduino/Arduino/issues/2656 (and I agree, this is a very necessary feature)

closed 2015-09-28 by ffissore

per1234, 2015-09-28:
Should this be fixed in the hourly build or does that not include commits to this repository?

NicoHood, 2015-09-28:
First a release need to be published again:
https://github.com/arduino/arduino-builder/releases

Then the IDE needs this patch and then the builder needs to be triggered again.
I guess he will do this later this day.

ffissore, 2015-09-28:
Yes indeed. Please wait for beta16 release. I'm still working on #18 which has opened a pandora's box

commit d661e83a (2015-09-28): RUNTIME_PLATFORM_PATH and RUNTIME_HARDWARE_PATH point to target platform path,
not actual. Fixes #20