arduino/arduino-builder #21: Multiple libraries with the same name cause "Rel: can't make... relative to..." error

per1234, 2015-09-29:
Arduino IDE 1.6.6 2015/09/28 06:42 Windows 7 64bit
I'm guessing that this issue belongs in this repository because the issue started around the time arduino-builder was added.

If there are libraries in C:\Program Files (x86)\arduino-nightly\libraries and sketchbook\libraries with the same name then compiling a sketch that includes that library fails. For example if I copy the Servo library to my sketchbook\libraries folder and compile Knob.ino I get:
```
C:\Program Files (x86)\arduino-nightly\arduino-builder -dump-prefs -logger=machine -hardware "C:\Program Files (x86)\arduino-nightly\hardware" -hardware "C:\Users\per\AppData\Local\Arduino15\packages" -hardware "E:\Stuff\misc\electronics\arduino\hardware" -tools "C:\Program Files (x86)\arduino-nightly\tools-builder" -tools "C:\Program Files (x86)\arduino-nightly\hardware\tools\avr" -tools "C:\Users\per\AppData\Local\Arduino15\packages" -libraries "E:\Stuff\misc\electronics\arduino\libraries" -libraries "C:\Program Files (x86)\arduino-nightly\libraries" -fqbn=arduino:avr:uno -ide-version=10606 -build-path "C:\Users\per\AppData\Local\Temp\buildbba0b171f7b3ad0f8360c2d9fa887ccb.tmp" -warnings=all -prefs=build.warn_data_percentage=75 -verbose "E:\Stuff\misc\electronics\arduino\libraries\Servo\examples\Knob\Knob.ino"
C:\Program Files (x86)\arduino-nightly\arduino-builder -compile -logger=machine -hardware "C:\Program Files (x86)\arduino-nightly\hardware" -hardware "C:\Users\per\AppData\Local\Arduino15\packages" -hardware "E:\Stuff\misc\electronics\arduino\hardware" -tools "C:\Program Files (x86)\arduino-nightly\tools-builder" -tools "C:\Program Files (x86)\arduino-nightly\hardware\tools\avr" -tools "C:\Users\per\AppData\Local\Arduino15\packages" -libraries "E:\Stuff\misc\electronics\arduino\libraries" -libraries "C:\Program Files (x86)\arduino-nightly\libraries" -fqbn=arduino:avr:uno -ide-version=10606 -build-path "C:\Users\per\AppData\Local\Temp\buildbba0b171f7b3ad0f8360c2d9fa887ccb.tmp" -warnings=all -prefs=build.warn_data_percentage=75 -verbose "E:\Stuff\misc\electronics\arduino\libraries\Servo\examples\Knob\Knob.ino"
Warning: platform.txt from core 'Arduino ARM (32-bits) Boards' contains deprecated recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mcpu={build.mcu} "-T{build.variant.path}/{build.ldscript}" "-Wl,-Map,{build.path}/{build.project_name}.map" {compiler.c.elf.extra_flags} -o "{build.path}/{build.project_name}.elf" "-L{build.path}" -mthumb -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--warn-unresolved-symbols -Wl,--start-group "{build.path}/syscalls_sam3.c.o" {object_files} "{build.variant.path}/{build.variant_system_lib}" "{build.path}/{archive_file}" -Wl,--end-group -lm -gcc, automatically converted to recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mcpu={build.mcu} "-T{build.variant.path}/{build.ldscript}" "-Wl,-Map,{build.path}/{build.project_name}.map" {compiler.c.elf.extra_flags} -o "{build.path}/{build.project_name}.elf" "-L{build.path}" -mthumb -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--warn-unresolved-symbols -Wl,--start-group "{build.path}/core/syscalls_sam3.c.o" {object_files} "{build.variant.path}/{build.variant_system_lib}" "{build.path}/{archive_file}" -Wl,--end-group -lm -gcc. Consider upgrading this core.
Warning: platform.txt from core 'Arduino ARM (32-bits) Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
Warning: platform.txt from core 'Arduino SAMD (32-bits ARM Cortex-M0+) Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
"C:\Program Files (x86)\arduino-nightly\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR    "C:\Users\per\AppData\Local\Temp\buildbba0b171f7b3ad0f8360c2d9fa887ccb.tmp\sketch\Knob.ino.cpp"
Rel: can't make \Stuff\misc\electronics\arduino\libraries\Servo\src relative to \Program Files (x86)\arduino-nightly\hardware\arduino\avr
Error compiling.
```
If I delete either library it compiles without errors. Using Arduino IDE 1.6.5r5 it compiles without errors(I just get the `Multiple libraries were found for "Servo.h"` notification).

Full output with duplicate libraries: https://gist.github.com/per1234/3b579b92234f25c1e76e

Full output without duplicate libraries(deleted C:\Program Files (x86)\arduino-nightly\libraries\Servo): https://gist.github.com/per1234/d746e420135c04fbe87d

ffissore, 2015-09-29:
Investingating

closed 2015-09-29 by ffissore

ffissore, 2015-09-29:
FYI fix available since beta17 available with hourly builds http://www.arduino.cc/en/Main/Software#hourly

sej7278, 2015-11-07:
did this fix not make it into the 1.6.6 ide, as i'm seeing it now.

calling the arduino-builder command output from the ide fails (see below), as does using the json file from -dumpprefs, but the ide builds fine.

```
~/arduino-1.6.6/arduino-builder -compile -logger=machine \
-hardware "~/arduino-1.6.6/hardware" -hardware "~/arduino16/hardware" \
-tools "~/arduino-1.6.6/tools-builder" -tools "~/arduino-1.6.6/hardware/tools/avr" \
-built-in-libraries "~/arduino-1.6.6/libraries" -libraries "~/arduino16/libraries" \
-fqbn=esp8266com:esp8266:nodemcuv2:CpuFrequency=80,UploadTool=esptool,UploadSpeed=115200,FlashSize=4M3M \
-ide-version=10606 -build-path "build" -warnings=none -prefs=build.warn_data_percentage=75 \
-verbose "~/arduino16/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/examples/WiFiWebServer/WiFiWebServer.ino"
...
Rel: can't make ~/build/core/core.a relative to build
```

commit 1f368a88 (2015-09-29): Avoid using filepath.Rel when unnecessary. Fixes #21