arduino/arduino-builder #22: library-to-library dependencies

descampsa, 2015-09-29:
See https://github.com/arduino/Arduino/issues/3877 for previous discussion.
The problem is that a library is not recompiled when the include list of the main sketch (or of another library) has changed, despite the compilation option (include paths) are modified.

Note that when library-to-library dependencies will be completely supported (https://github.com/arduino/arduino-builder/issues/12), i think it will be harder to find a reasonnable case where it is really noticeable. The only one i can think of is the one suggested by matthijskooijman, with the conflicting secondary header files. 

I still think however that the proposed solution (each library generates its own dependency list) would make more sense and avoid bad surprises in some corner cases.

ffissore, 2015-09-30:
@descampsa can you provide an example sketch + libraries, even fake libraries?

descampsa, 2015-09-30:
Sure, here you have a (very) minimal example:
https://app.box.com/s/jkk2bk01l2y3omfdcakdbyhnk8kfj72w

1) compile with #include<SoftwareSerial.h> commented in the sketch -> fail to find SoftwareSerial.h in Foo.cpp
2) uncomment #include<SoftwareSerial.h> -> compile fine
3) comment again -> still compile fine, because the library is not recompiled

If you want a real lib where this happened to me, see https://github.com/descampsa/ardyno. 
You will have to replace the hacky #include<../../libraries/SoftwareSerial/SoftwareSerial.h> in DynamixelInterface.cpp to reproduce the same issue.

ffissore, 2015-10-01:
Uhm I see. I'm looking for a clean solution. In the meanwhile, move `#include <SoftwareSerial.h>` to Foo.h as that will leverage current library to library dependency discovery

ffissore, 2015-10-29:
Note to self: while this issue is still valid, it's not reproducible anymore with the given sketch, because arduino-builder is now able to discover SoftwareSerial dependency specified in Foo.cpp

ffissore, 2015-12-01:
@descampsa while I couldn't replicate this any more thanks to the smarter dependency discovery algorithm, I've also added a step that DELETES previously compiled libraries files IF those libraries are no longer used. Fix will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly

closed 2015-12-01 by ffissore