arduino/arduino-builder #23: Regression when using default parameters

matthijskooijman, 2015-09-30:
Consider this sketch:

    void test(int x = 1) {
    }

    void setup() {
    }

    void loop() {
    }

When using 1.6.5 this compiles, but it breaks when using arduino-builder. I remember reading that supporting optional arguments with arduino-builder/ctags is too tricky, so it was intentionally left out. However, since the old regex-based parser code simply did not recognize functions using default arguments at all (so did not generate a prototype for them), this means that functions using default arguments compiled properly before, but not anymore.

Perhaps it is possible to detect that default arguments are present in a function prototype and if so, refrain from generating a prototype? Ideally, a prototype is generated with the default arguments removed, but that is probably tricky to get right (without involving things like libclang).

ffissore, 2015-09-30:
I think I can just look a the signature of the function and if it contains an `=` sign I can just skip prototype generation. Thank you, I'll work on that asap

matthijskooijman, 2015-09-30:
Yeah, I can't think of any prototype that contains an = without using a default argument, so that should work. Perhaps something fancy using C++11 `exprtype`, but even then preventing a prototype wouldn't really hurt I guess.

closed 2015-09-30 by ffissore

matthijskooijman, 2015-09-30:
Seems to work, thanks!

commit 57243135 (2015-09-30): Skipping prototype generation for function with default args. Fixes #23