arduino/arduino-builder #24: No sketch folder property in build properties

stevemarple, 2015-09-30:
`build.source.path` contains the absolute name of the sketch .ino file. No property exists which defines the sketch folder. This is required for recipes which need to reference files within the original sketch folder.

The closest available is `build.path` (/tmp/...) but when referencing files in there the timestamp changes which forces compilation to be run even when not required.

ffissore, 2015-10-01:
Are you sure? `build.source.path` is part of build properties since first release. See https://github.com/arduino/arduino-builder/blob/master/src/arduino.cc/builder/setup_build_properties.go#L98-L104 
This is visible even when running `-dump-prefs` (the first command the IDE runs when you compile)
Can you add some more details and possibly share your platform.txt?

stevemarple, 2015-10-01:
`build.source.path` was what I expected to use, dumping prefs I get the .ino file, not its folder.

    build.source.path=/home/mango/marple/work/arduino15/sketches/sketch_sep24a/sketch_sep24a.ino

I'm using the standard AVR platform.txt file from an unmodifed version of arduino/Arduino (commit 435fc323e0ab3b03bc5835e5309b5022bf57388a)

ffissore, 2015-10-01:
Ahh you're right. I mistakenly set it to the sketch file instead of its folder. Fixing it right now

closed 2015-10-01 by ffissore

ffissore, 2015-10-01:
Ok, fix will be available with next IDE hourly, which is being built http://www.arduino.cc/en/Main/Software#hourly

commit 02968993 (2015-10-01): BUILD_PROPERTIES_SOURCE_PATH was set to main .ino file location, while it must
be set to its container folder. Fixes #24