arduino/arduino-builder #25: Make the "Using library" output messages a bit more visible

cmaglie, 2015-10-02:
I'm talking about the:

```
Using library Keyboard at version 1.0.0 in folder: ....
Using library Mouse at version 1.0.0 in folder: ....
Using library HID at version 1.0 in folder: ....
```

displayed when the verbose output is enabled.

Currently those messages appears in the middle of a bunch of gcc command lines, making them barely visible.

[image]

If it's not possible to put them at the beginning, maybe it's worth printing them at the end with a line of separation from the "bulk" of the build?

PaulStoffregen, 2015-10-02:
These used to be at the very beginning, and the duplicate messages were at the very end, right before the size info.

closed 2015-10-02 by ffissore

ffissore, 2015-10-02:
Both messages push to the bottom

commit 5e2ac096 (2015-10-02): Storing library resolution results for later print them. Fixes #25