arduino/arduino-builder #26: Arduino builder builds the same sketch with different size

NicoHood, 2015-10-03:
Very weird bug: Hitting compile several times results in different flash usage, even with no change at all.

Used this IDE for linux64bit (okay I made some changes to the usb code, but this should not matter here).
https://github.com/arduino/Arduino/pull/3896#issuecomment-144994262

**I compiled for Arduino Mega**

Sketch size is between 12356 and 12348. Very useful for debugging and getting the last byte out of the code ;) (not)

The sketch (requires FastLED3.1 release, no  dev version or sth like this)
```
// FastLED
#include "FastLED.h"

#define LED_PINS    MOSI, SCK // DATA_PIN, or DATA_PIN, CLOCK_PIN
#define COLOR_ORDER RGB
#define CHIPSET     WS2801 // WS2811 LPD8806
#define NUM_LEDS    25

#define BRIGHTNESS  255  // reduce power consumption
#define DITHER_LEDS 255  // try 0 to disable flickering
#define CORRECTION  TypicalLEDStrip

CRGB leds[NUM_LEDS]; // Define the array of leds

// Leds
const uint8_t pinLed = LED_BUILTIN;

#define DEBUG_STREAM Serial1

void setup() {
  // Prepare debug Led
  pinMode(pinLed, OUTPUT);

  // FastLED setup
  FastLED.addLeds<CHIPSET, LED_PINS, COLOR_ORDER>(leds, NUM_LEDS).setCorrection(CORRECTION);
  FastLED.setBrightness(BRIGHTNESS);
  FastLED.setDither(DITHER_LEDS);

  // Startup animation
  fill_solid(leds, NUM_LEDS, CRGB::Red);
  FastLED.show();
  delay(500);
  fill_solid(leds, NUM_LEDS, CRGB::Green);
  FastLED.show();
  delay(500);
  fill_solid(leds, NUM_LEDS, CRGB::Blue);
  FastLED.show();
  delay(500);
  FastLED.clear();
  FastLED.show();

  // Serial we get Adalight data from has to be baud 115200
  // Start Serial after startup animation to avoid data corruption
  Serial.begin(115200);
  DEBUG_STREAM.begin(115200);
  DEBUG_STREAM.println("Start2");
}

void loop() {
  // Visualize errors via led
  static uint32_t lasterr = 0;
  if (millis() - lasterr > 1000) {
    digitalWrite(pinLed, LOW);

    // Updates leds on idle (to avoid flickering when inactive)
    FastLED.show();

    DEBUG_STREAM.println(FastLED.getFPS());
  }

  //static uint32_t lastUpdate = 0;
#define ADA_TIMEOUT_ERR 1000

  // Visualize leds via Adalight
  int8_t newData = adalight(Serial, leds, NUM_LEDS);

  // Check if any reading error occured
  if (newData < 0) {
    // Print error count
    static uint8_t e = 0;
    DEBUG_STREAM.print(++e);
    DEBUG_STREAM.println(newData);

    lasterr = millis();
    digitalWrite(pinLed, HIGH);

    // Turn off leds
    FastLED.clear();
    FastLED.show();
  }
}

// Magic keyword that the host sends via Serial for Adalight
const uint8_t magic[] PROGMEM = {'A', 'd', 'a'};

int8_t adalight(Stream & s, CRGB l[], const uint16_t numLeds) {
  // Static variables for reading incoming data
  // Reading from multiple streams is possible here.
  // Switching the stream might cause a few errors first.
  static uint8_t magicPos = 0;
  static uint8_t magicLo;
  static uint8_t magicHi;
  static uint16_t numPixel = 0;

  // Differenciate between errors
  enum AdaErrorLevel : int8_t
  {
    ADA_UPDATE = 1,
    ADA_NO_ERROR = 0,
    ADA_MAGIC_WORD = -1,
    ADA_CHECKSUM = -2,
    ADA_LENGTH = -3,
    ADA_CORRUPTION = -4,
  };

  // Empty the current data in buffer but do not block forever
  auto bytesAvailable = s.available();
  while (bytesAvailable--)
  {
    // Get next byte from the stream
    uint8_t input = s.read();

    // Simulate data corruption TODO
    if (random16(1024) == 0) {
      input = random8();
    }

    // Check the magic word
    if (magicPos < (sizeof(magic)))
    {
      /*
            // Compare incoming bytes with the magic word
            if (input != pgm_read_byte(magic + magicPos))
            {
              // Current Magic Series is wrong, start from the beginning
              magicPos = 0;

              // Maybe the last led data bytes were accidently a magic letter.
              // Try to check the first letter (instead of the 2nd or 3rd).
              if (input != pgm_read_byte(magic))
              {
                // We will continuously check for a new magic word, even while updating the led data
                // This is not an error unless we are really waiting for a new signal.
                if (!numPixel) {
                  return ADA_MAGIC_WORD;
                }
              }
              else {
                magicPos ++;
              }
            }
            else
              magicPos++;*/

      // 12356

      // Compare incoming bytes with the magic word
      if (input == pgm_read_byte(magic + magicPos))
      {
        magicPos++;
      }

      // Maybe the last led data bytes were accidently a magic letter.
      // Try to check the first letter (instead of the 2nd or 3rd).
      else if (input == pgm_read_byte(magic))        {
        magicPos = 1;
      }

      // We will continuously check for a new magic word, even while updating the led data
      // This is not an error unless we are really waiting for a new signal.
      else {
        magicPos = 0;
        if (!numPixel) {
          return ADA_MAGIC_WORD;
        }

      }

    }

    // Check the high byte
    else if (magicPos == sizeof(magic)) {
      magicHi = input;
      magicPos++;
    }

    // Check the low byte
    else if (magicPos == (sizeof(magic) + 1)) {
      magicLo = input;
      magicPos++;
    }

    // Check the checksum
    else
    {
      // In any case reset the counter to look for new magic words
      magicPos = 0;

      // Checksum is correct
      if (input == (magicHi ^ magicLo ^ 0x55))
      {
        // Save new number of leds
        uint16_t newLeds = ((magicHi << 8) | magicLo) + 1;

        // Check if we have enough leds in our strip
        if (newLeds == numLeds)
        {
          // Cache if we were still reading new data
          auto oldPixel = numPixel;

          // Each led has 3 pixel
          numPixel = newLeds * 3;

          // Last reading did not finish, this is a critical error
          if (oldPixel) {
            return ADA_CORRUPTION;
          }

          // Do not update the leds now but in the next loop
          continue;
        }
        else {
          // Something in the PC side configuration must be set wrong.
          // It is not very likely that this is normal led data,
          // but pretending it is, we flush the last pixels, if any.
          if (!numPixel) {
            return ADA_LENGTH;
          }
        }
      }
      // Wrong checksum
      else {
        // This might happen if we accidently get "Ada" as led data.
        // This is not an error unless we are really waiting for a new signal.
        if (!numPixel) {
          return ADA_CHECKSUM;
        }
      }
    }

    // Proceed Led signal
    if (numPixel)
    {
      // Proceed next pixel
      numPixel--;
      l[(numLeds - 1) - (numPixel / 3)].raw[2 - numPixel % 3] = input;

      // Update Leds if this is the last pixel
      if (!numPixel) {
        return ADA_UPDATE;
      }
    }
  }

  // Flag if the leds got a new frame that should be updated
  return ADA_NO_ERROR;
}
```

Arduino builder commands:
```
/Arduino/build/linux/work/arduino-builder -dump-prefs -logger=machine -hardware "/Arduino/build/linux/work/hardware" -hardware ".arduino15/packages" -hardware "/sketchbook/hardware" -tools "/Arduino/build/linux/work/tools-builder" -tools "/Arduino/build/linux/work/hardware/tools/avr" -tools ".arduino15/packages" -libraries "/sketchbook/libraries" -libraries "/Arduino/build/linux/work/libraries" -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10606 -build-path "/tmp/buildc0fbe65fc3436bea157b420c01dcd579.tmp" -warnings=all -prefs=build.warn_data_percentage=75 -verbose "/tmp/arduino_c0fbe65fc3436bea157b420c01dcd579/sketch_oct03b.ino"
/Arduino/build/linux/work/arduino-builder -compile -logger=machine -hardware "/Arduino/build/linux/work/hardware" -hardware ".arduino15/packages" -hardware "/sketchbook/hardware" -tools "/Arduino/build/linux/work/tools-builder" -tools "/Arduino/build/linux/work/hardware/tools/avr" -tools ".arduino15/packages" -libraries "/sketchbook/libraries" -libraries "/Arduino/build/linux/work/libraries" -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10606 -build-path "/tmp/buildc0fbe65fc3436bea157b420c01dcd579.tmp" -warnings=all -prefs=build.warn_data_percentage=75 -verbose "/tmp/arduino_c0fbe65fc3436bea157b420c01dcd579/sketch_oct03b.ino"
```

NicoHood, 2015-10-03:
It also compiles the SD Card library for some reason. You can see the warnings here:
https://gist.github.com/NicoHood/8472a52d2bfbe62cc958

\cc @kriegsman and @focalintent

PaulStoffregen, 2015-10-03:
Is the builder parsing *all* files for includes, or only those which are actually used?

One of the many examples in FastLED includes the SD library.  Maybe arduino-builder is mistakenly seeing that?

NicoHood, 2015-10-03:
I now got a weird error:
It said a function is not declared, marked it red in the IDE, skipped and compiled fine. The error was there but compiling finished still fine.

I could not reproduce this error once more.

focalintent, 2015-10-03:
Anything that references the ```__DATE__``` or ```__TIME__``` macros are likely to cause changes in the final output size, ditto any kind of textual time stamping that gcc adds to files.  As for the warnings, i'll deal with them later, but they're mostly just noise (and also aren't showing up in 1.6.5).

focalintent, 2015-10-03:
and https://github.com/FastLED/FastLED/commit/8021b0775217b83944e63078097bda2de4ceea79 should shut up the FastLED warnings.

As for why it's including the SD library, I think @PaulStoffregen is right - arduino-builder is being a bit too liberal in which files it's including in its search for linked libraries.  It should be careful about this, however - in the main library, FastLED does conditional inclusion of other library header files - but just because the library has #include<SmartMatrix.h> doesn't mean that said include is actually going to happen (because in that particular example/case, the include only happens if SmartMatrix_h has been defined - which means the end user would have to have included SmartMatrix.h in their .ino file first)

ffissore, 2015-10-05:
in beta21 arduino-builder also collected example sketches. this is fixed in 45b744f355c24d7ef64de483359bcd40e4578ed5 and available with beta22 and latest IDE nightly. As for not parsing all the files, and only those actually used, I'm afraid I can do nothing about it. Both old and new compiling procedures used to compile every single cpp file, delegating to the linker the task to filter unused code.
You may want to add `dot_a_linkage=true` to your library.properties and see if sketch size improves.

So @NicoHood and @PaulStoffregen please ensure you're using the very latest hourly build
@NicoHood, about the warning and the successful compilation: may this be due to some gcc params like in https://github.com/arduino/Arduino/commit/7d0355c8f15692e4b7d1ac58c96af0059c3325bb ?

NicoHood, 2015-10-05:
I will try again today.

The skipped error never showed up again. However the commit you linked doesnt matter for me since I am using avr and not sam.

ffissore, 2015-10-05:
Let me know how your tests go

NicoHood, 2015-10-05:
Original Problem (Mega compiling with different size) is gone.
SD Card Warnings also gone.

However the warnings themselves should be fixed. But thats not related to this ;)

closed 2015-10-05 by NicoHood

ffissore, 2015-10-05:
I can't wait for a new issue @NicoHood . File it asap :)