arduino/arduino-builder #28: Another Preprocessor error

NicoHood, 2015-10-03:
```cpp
void setup() {}

void loop() {
  // Visualize leds via Adalight
  int8_t newData = adalight();

}

//#define ADALIGHT_USE_TEMPLATE

#ifdef ADALIGHT_USE_TEMPLATE
#error
int16_t adalight_template()
#else
int8_t adalight()
#endif
{
  // Flag if the leds got a new frame that should be updated
  return 0;
}

/*
 * sketch_oct03d:5: error: 'adalight' was not declared in this scope
   int8_t newData = adalight();
 */
```
Error in code above, should be self explaining.

NicoHood, 2015-10-03:
Another example this the same error. Might be two separat issues, I dont know whats behind the code. So I post it here, pretending its the same reason which causes this:

```cpp
void setup() {}
void loop() {
  int8_t newData = adalight();
}

//#define ADALIGHT_USE_TEMPLATE

#ifdef ADALIGHT_USE_TEMPLATE
#error
#else
// commenting this header makes the sketch compile
int8_t adalight();
#endif
int8_t adalight()
{
  return 0;
}
```

ffissore, 2015-10-05:
In your second sketch, `int8_t adalight();` is a prototype and you must specify it at the beginning of the sketch

closed 2015-10-05 by ffissore

ffissore, 2015-10-05:
Your first sketch still doesn't compile, unless you rename `int16_t adalight_template()` to `int16_t adalight()`. CTags gets fooled by define and it's not even seeing `int8_t adalight()`

ffissore, 2015-10-05:
More info at http://linux.die.net/man/1/ctags, scroll down to "Operational Details"

NicoHood, 2015-10-05:
So this is intended?

ffissore, 2015-10-05:
No, it's a known limitation of ctags. We have to either fix it or get used to it. Maybe [this one](https://github.com/arduino/ctags/issues/2) will help us

commit a25b7e8c (2015-10-05): Introducing types.Prototype: useful for carrying more data about original
ctags output, including function name
Function name is used in compare_prototypes_from_source_and_preproc_source
to understand if a function is part of preprocessed output.
Also, prototype from preprocessed output is used instead of that coming from
ctags on plain sketch, as ctags may be fooled by #define
Fixes #28