arduino/arduino-builder #32: HID-Project.a: No such file or directory

NicoHood, 2015-10-09:
Ha! Found another (minor) bug.
If you just have a single header file within a library that is linked via .a file you get this error.

Normally this wont happen and this error is more or less expected if you provide such a nonsense library. Just want to report it anyways.

For sure I can catch better bugs this weekend ;) Looking carefully now ;D

matthijskooijman, 2015-10-12:
Sounds like a real bug, though, a header-only library isn't entirely unheard of or nonsensical.

cmaglie, 2015-10-13:
AFAIU this is happening only when `dot_a_linkage` is enabled in `library.properties` right?
If this is the case this issue can be low-priority, since the workaround is easy (just don't set `dot_a_linkage` in `library.properties`).

matthijskooijman, 2015-10-13:
Ah, good point, I believe so :-)

NicoHood, 2015-10-13:
For this case yes. A single .h file needs no .a linkage at all. But it should be fixed some day.

ffissore, 2015-10-27:
can a `.a` file contain a compiled `.h` file? I guess not. If so, making such a library and declaring it supports dot_a_linkage is wrong

NicoHood, 2015-10-27:
.h files cannot be compiled, they are just headers. As said, it does not make sense to create such a library, I accidentally run into that error, but normally this wont happen.

closed 2015-10-27 by ffissore