arduino/arduino-builder #34: undefined reference with multiple names .ccp file in different folders

NicoHood, 2015-10-11:
There we go! Found a "better" bug finally :)

If you have a library structure like this:
/src/folderA/file.cpp
/src/folderB/file.cpp

You will get compile errors, even if the .cpp file (with the same name) just exists (empty):
```
Arduino/sketchbook/libraries/HID-Project/src/MultiReport/../HID-APIs/GamepadAPI.hpp:38: undefined reference to `Gamepad'
[more of them...}
```

Commit can be found here:
https://github.com/NicoHood/HID/commit/6d6c7d3068afffb13050b3047b12fb28a55b4624

This PR is required to build the Gamepad example without (other) compile errors:
https://github.com/arduino/Arduino/pull/3948

So it looks like the builder does not differentiate between folders.

matthijskooijman, 2015-10-13:
Weird, I thought we fixed this by keeping the library directory structure intact, but perhaps this wasn't copied over to the arduino-builder code?

Could you post a full verbose compilation output?

I suspect that this is a limitation of `gcc-ar` / .a files. If I read the ar manpage right, files inside an archive are by default identified by just their filenames, no path included. It seems there is a non-standard extension to use the full pathname, but gnu ar only supports reading such files, not writing them. Perhaps using `compiler.ar.flags=qc` (using quick append, rather then replace) might help, though I'm not sure how the linker then treats this archive.

ffissore, 2015-10-27:
@NicoHood can you pls provide full debug output?

NicoHood, 2015-10-30:
I could not reproduce this anymore. The commit above does not work with the new Arduino API.
I do not intent to hunt this bug now. Maybe if I feel like renaming my files again I can check if it exists...

ffissore, 2015-11-02:
Leaving it open then. If if happens to find a way to reproduce it, let us know

facchinm, 2016-05-10:
AFAIK it was never reproduced, reopen if needed

closed 2016-05-10 by facchinm