arduino/arduino-builder #35: integrate board manager into arduino builder

pgollor, 2015-10-22:
Is there any plan to integrates the boards manager into the command line builder? This could be very useful for continuous integration with 3rd party hardware libraries like [arduino-esp](https://github.com/esp8266/Arduino).

I wrote a python and bash script [here](https://github.com/pgollor/esp8266-arduino-autobuilder) which reads the json file and download all dependencies. I would be very nice if the arduino autobuilder could be handle it by itself.

ffissore, 2015-10-27:
arduino-builder will not integrate boards manager. Use the arduino CLI from the Arduino IDE. See https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc

closed 2015-10-27 by ffissore

kubark42, 2017-11-02:
> arduino-builder will not integrate boards manager. 

Could I ask why? Being able to call the boards manager from the CLI is very useful for doing continuous integration.