arduino/arduino-builder #36: Compile error on weird circumstances

NicoHood, 2015-10-24:
Originally I had a huge sketch which failed to compile. The issue could be solved by removing a warning (removed a typedef before an enum). This was weird enough.

I simplified my example and the original "workaround" with the enum was no required anymore. Now including another library caused the error. In the original sketch even more libraries were included, so I guess its a library detection problem.

The sketch:
```cpp
#include "HID-Project.h"

#include "PinChangeInterrupt.h"

void setup() {
  Keyboard.begin();
}

void loop() {}
```

Dependencies:
https://github.com/NicoHood/PinChangeInterrupt/tree/7b01fe1ec760c419daacf9348f8930d85c52f45c
https://github.com/NicoHood/HID/tree/60a30c1cdba9186a88e1953d542f2df58b7c7083
https://github.com/arduino/Arduino/tree/9552cc6ce255850dccb26d634f6373a3ffd32643

The verbose output (nah I will not do the arduino builder output here, sorry):
```
/Arduino/build/linux/work/arduino-builder -dump-prefs -logger=machine -hardware "/Arduino/build/linux/work/hardware" -hardware "/home/linuxuser/.arduino15/packages" -hardware "/sketchbook/hardware" -tools "/Arduino/build/linux/work/tools-builder" -tools "/Arduino/build/linux/work/hardware/tools/avr" -tools "/home/linuxuser/.arduino15/packages" -libraries "/sketchbook/libraries" -libraries "/Arduino/build/linux/work/libraries" -fqbn=arduino:avr:leonardo -vid-pid=0X2341_0X8036 -ide-version=10606 -build-path "/tmp/builda905e339abe3b59fd5c5e8847955095e.tmp" -warnings=all -prefs=build.warn_data_percentage=75 -verbose "/tmp/arduino_a905e339abe3b59fd5c5e8847955095e/Remote_USB_MCU_2_test.ino"
/Arduino/build/linux/work/arduino-builder -compile -logger=machine -hardware "/Arduino/build/linux/work/hardware" -hardware "/home/linuxuser/.arduino15/packages" -hardware "/sketchbook/hardware" -tools "/Arduino/build/linux/work/tools-builder" -tools "/Arduino/build/linux/work/hardware/tools/avr" -tools "/home/linuxuser/.arduino15/packages" -libraries "/sketchbook/libraries" -libraries "/Arduino/build/linux/work/libraries" -fqbn=arduino:avr:leonardo -vid-pid=0X2341_0X8036 -ide-version=10606 -build-path "/tmp/builda905e339abe3b59fd5c5e8847955095e.tmp" -warnings=all -prefs=build.warn_data_percentage=75 -verbose "/tmp/arduino_a905e339abe3b59fd5c5e8847955095e/Remote_USB_MCU_2_test.ino"
WARNING: Spurious .git folder in 'AnalogTouch' library
WARNING: Spurious .git folder in 'FastLED' library
WARNING: Spurious .dev folder in 'HID-Project' library
WARNING: Spurious .git folder in 'HID-Project' library
WARNING: Spurious .git folder in 'IRLremote' library
WARNING: Spurious .git folder in 'MSGEQ7' library
WARNING: Spurious .git folder in 'Nintendo' library
WARNING: Spurious .git folder in 'PinChangeInterrupt' library
WARNING: Spurious .git folder in 'RCLSwitch' library
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"'  "/tmp/builda905e339abe3b59fd5c5e8847955095e.tmp/sketch/Remote_USB_MCU_2_test.ino.cpp"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/sketchbook/hardware/arduino/avr/cores/arduino" "-I/sketchbook/hardware/arduino/avr/variants/leonardo" "-I/sketchbook/libraries/HID-Project/src" "-I/sketchbook/libraries/PinChangeInterrupt/src" "/tmp/builda905e339abe3b59fd5c5e8847955095e.tmp/sketch/Remote_USB_MCU_2_test.ino.cpp"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/sketchbook/hardware/arduino/avr/cores/arduino" "-I/sketchbook/hardware/arduino/avr/variants/leonardo" "-I/sketchbook/libraries/PinChangeInterrupt/src" "-I/sketchbook/libraries/HID-Project/src" "-I/sketchbook/hardware/arduino/avr/libraries/HID" "/tmp/builda905e339abe3b59fd5c5e8847955095e.tmp/sketch/Remote_USB_MCU_2_test.ino.cpp"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/sketchbook/hardware/arduino/avr/cores/arduino" "-I/sketchbook/hardware/arduino/avr/variants/leonardo" "-I/sketchbook/libraries/HID-Project/src" "-I/sketchbook/hardware/arduino/avr/libraries/HID" "-I/sketchbook/libraries/PinChangeInterrupt/src" "/tmp/builda905e339abe3b59fd5c5e8847955095e.tmp/sketch/Remote_USB_MCU_2_test.ino.cpp"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/sketchbook/hardware/arduino/avr/cores/arduino" "-I/sketchbook/hardware/arduino/avr/variants/leonardo" "-I/sketchbook/libraries/PinChangeInterrupt/src" "-I/sketchbook/hardware/arduino/avr/libraries/HID" "-I/sketchbook/libraries/HID-Project/src" "/sketchbook/libraries/HID-Project/src/MultiReport/AbsoluteMouse.cpp"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/sketchbook/hardware/arduino/avr/cores/arduino" "-I/sketchbook/hardware/arduino/avr/variants/leonardo" "-I/sketchbook/libraries/PinChangeInterrupt/src" "-I/sketchbook/hardware/arduino/avr/libraries/HID" "-I/sketchbook/libraries/HID-Project/src" "/sketchbook/libraries/HID-Project/src/MultiReport/Consumer.cpp"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/sketchbook/hardware/arduino/avr/cores/arduino" "-I/sketchbook/hardware/arduino/avr/variants/leonardo" "-I/sketchbook/libraries/HID-Project/src" "-I/sketchbook/hardware/arduino/avr/libraries/HID" "-I/sketchbook/libraries/PinChangeInterrupt/src" "/sketchbook/libraries/HID-Project/src/MultiReport/Gamepad.cpp"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/sketchbook/hardware/arduino/avr/cores/arduino" "-I/sketchbook/hardware/arduino/avr/variants/leonardo" "-I/sketchbook/libraries/HID-Project/src" "-I/sketchbook/libraries/PinChangeInterrupt/src" "-I/sketchbook/hardware/arduino/avr/libraries/HID" "/sketchbook/libraries/HID-Project/src/MultiReport/ImprovedKeyboard.cpp"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/sketchbook/hardware/arduino/avr/core [... cut ...]
/sketchbook/libraries/HID-Project/src/SingleReport/../HID-APIs/KeyboardAPI.hpp:312: undefined reference to `Keyboard'
/sketchbook/libraries/HID-Project/src/SingleReport/../HID-APIs/KeyboardAPI.hpp:312: undefined reference to `Keyboard'
/tmp/builda905e339abe3b59fd5c5e8847955095e.tmp/sketch/Remote_USB_MCU_2_test.ino.cpp.o:/sketchbook/libraries/HID-Project/src/SingleReport/../HID-APIs/KeyboardAPI.hpp:312: more undefined references to `Keyboard' follow
collect2: error: ld returned 1 exit status
exit status 1
Error compiling.
```

NicoHood, 2015-10-24:
Now I tried to fix this (original sketch) by removing the enum typedef but ran into the next error on adding an if statement. This needs to be fixed.

I also want to note that compiling takes a long time.

I cannot go back in time and use an old arduino builder, since I need the PUSB changes to test this bug. Otherwise my libraries wont compile.

BTW:
PinChangeInterrupt and HID-Project both use .a linkage!

matthijskooijman, 2015-10-27:
The problem is here: https://github.com/arduino/arduino-builder/blob/master/src/arduino.cc/builder/phases/libraries_builder.go#L101

`compileLibrary()` gets passed a slice of object files and should add new object files to it, but when DotALinkage is set, it instead takes _all_ the previously compiled object files (including any from other libraries), puts them all into a .a file and then replaces the entire list of object files with a new slice containing just the single .a file.

The correct fix here would be to, if DotALinkage is set, call `builder_utils.CompileFilesRecursive()` with an empty slice as the first argument and store the result in a new variable (not the existing `objectFiles` parameter). Then pass that new variable to `builder_utils.ArchiveCompiledFiles()` and add the resulting .a file to the `objectFiles` slice.

@ffissore, I presume you will take care of this one?

NicoHood, 2015-10-27:
Wow. I am really impressed. :O

Will this code ever get any comments or is it intended to stay like that?

ffissore, 2015-10-27:
Comments are scent for smelling code. Plus, they are not executed, hence they can lie (and they usually do)
@matthijskooijman thanks, I'll look into it

closed 2015-10-27 by ffissore

ffissore, 2015-10-27:
@NicoHood I can't include your code in the test suite as it relies on an unreleased version of the avr core. So I blindly fixed this. Find it in next Arduino IDE hourly build http://www.arduino.cc/en/Main/Software#hourly

matthijskooijman, 2015-10-27:
Fix looks good to me (slightly different from what I had in mind, but it should work).

NicoHood, 2015-10-30:
Seems to work now.

commit 08f193ff (2015-10-27): When compiling libraries, don't use a shared accumulator for all libraries
Fixes #36