arduino/arduino-builder #38: Duplicate library in sketchbook no longer has precedence over install folder library

per1234, 2015-10-29:
Using Arduino IDE 1.6.6 2015/10/28 05:06 with Windows 7 64 bit

The duplicate library located in the Arduino IDE install folder is now used instead of the one in the sketchbook/libraries folder.
- copy arduino-nightly/libraries/Servo to sketchbook/libraries
- Open IDE
- File > Examples > Servo > Knob
- Verify - the multiple libraries warning is:
```
Multiple libraries were found for "Servo.h"
 Used: C:\Program Files (x86)\arduino-nightly\libraries\Servo
 Not used: E:\Stuff\arduino\libraries\Servo
```
If I repeat this process with Arduino IDE 1.6.6 2015/10/26 06:42 the warning is:
```
Multiple libraries were found for "Servo.h"
 Used: E:\Stuff\arduino\libraries\Servo
 Not used: C:\Program Files (x86)\arduino-nightly\libraries\Servo
```

ffissore, 2015-10-29:
Thanks for the heads up, I'm working on it

closed 2015-10-29 by ffissore

ffissore, 2015-10-29:
Fixed in latest hourly build http://www.arduino.cc/en/Main/Software#hourly. Thank you very much!

per1234, 2015-10-30:
@ffissore I'm still having the same issue using Arduino IDE 1.6.6 2015/10/29 11:41

ffissore, 2015-10-30:
That's strange. However, I've just pushed another fix. Try downloading the newest hourly again

per1234, 2015-10-30:
That fixed it. One difference I noticed, and I'm not saying it's an issue, just mentioning in case it's not your desired result. With the Servo library in both the Arduino install folder and in sketchbook/libraries in IDE versions before Arduino IDE 1.6.6 2015/10/30 09:58 the Servo examples appeared in the **Examples from Custom Libraries** section of **File > Examples** but with Arduino IDE 1.6.6 2015/10/30 09:58 they appear in the **Example from Libraries** section instead.

ffissore, 2015-10-30:
That's because the hourly was probably built before the library listing was updated, so that the IDE didn't recognize that new version of the Servo lib. So it's expected and not an issue. Thanks for the heads up :)

commit 3577d1bb (2015-10-29): Libraries in platforms were being chosen instead of those outside, keeping
users from customizing them. Fixes #38