arduino/arduino-builder #39: Function with default parameter doesn't get it's prototype in *.h header file.

ffissore, 2015-10-30:
_From @obiwan-bartek on October 30, 2015 0:20_

When you write a function with at least one parameter with default value
eg.
```c
void foo(int a, int b = 0)
{
}
```
it does not gets generated it's prototype in header file, which results in an unreferenced calls in project to that function. When you remove the default parameter
```c
 = 0
```
the prototype shows in header file as expected.

_Copied from original issue: arduino/Arduino#4048_

ffissore, 2015-10-30:
_From @NicoHood on October 30, 2015 7:24_

You need to do:
```cpp
// prototype
void foo(int a, int b = 0);

// Implementation, better place inside a .cpp or marke the prototype above as static inline
void foo(int a, int b)
{

}
```

ffissore, 2015-10-30:
@obiwan-bartek as @NicoHood said, you have to write the prototype on your own. Current implementation is unable to handle default parameter (see #23). Correct prototype generation includes moving the default value from the function implementation to the prototype: this is something we can't do at the moment

closed 2015-10-30 by ffissore

NicoHood, 2015-10-30:
I answered this because he was talking about .h files. In previous version of the IDE it was possible in .ino files. Since I dont know much about the .ino file generation I dont want to answer to this. I also dont like to code inside .ino files with more complex stuff. Normally I use libraries for this, so I dont care about such features.

matthijskooijman, 2015-10-30:
Yeah, the old regex-based implementation tried harder to get correct prototypes, but also regularly  messed up. The new ctags-based approach only tries for the easy functions (no attributes, no inline, no function pointers, no default arguments) so it will at least generate working code. If you use more complex stuff, you'll have to take care of the prototypes (or proper ordering) yourself.

NicoHood, 2015-10-30:
I think thats okay if it then enables stuff like templates which made a lot of problems before.