arduino/arduino-builder #41: Recursion applied to old format & legacy libraries

PaulStoffregen, 2015-10-31:
Recursion seems to be applied to old & legacy format libraries.  According to the library spec:

https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification

    For backward compatibility with Arduino 1.0.x, the library author may opt to not place
    source code into a folder called src. In this case the 1.0 library format is applied and the
    source code is searched from the library root folder and the utility folder

Recursion is being used, compiling files in directories other than the main folder and utility.  It might also be finding .c and .cpp files within the examples.

I will create a test case....

PaulStoffregen, 2015-10-31:
Here is the test case.  This compiles in 1.6.5, but fails on the latest hourly build.

https://github.com/PaulStoffregen/LegacyNorecurseTest

ffissore, 2015-11-02:
Good catch! Working on it

closed 2015-11-02 by ffissore

ffissore, 2015-11-02:
Fixed. Fix is available with latest hourly build http://www.arduino.cc/en/Main/Software#hourly

commit d89b51e0 (2015-11-02): Source files were recursively collected despite target library was layout was
FLAT. Fixes #41