arduino/arduino-builder #45: include Arduino.h not added in some cases on Mac

PaulStoffregen, 2015-11-04:
I'm seeing a strange problem where the preprocessed sketch isn't getting #include "Arduino.h" added, and then of course fails to compile.  So far, it only seems to happen with my platform.txt and only on Mac.  Same code, same platform.txt works on Linux and Windows.

Will post a test case later today.

PaulStoffregen, 2015-11-06:
I still don't know why, but arduino-builder is creating this:

    #line 1 "/var/folders/6g/mqjhk3b153gbkhzjhn__xswh0000gn/T/untitled1098453877.tmp/sketch_nov06a/sketch_nov06a.ino"
    void setup() {
      // put your setup code here, to run once:

    }

    void loop() {
      // put your main code here, to run repeatedly:

    }

The first couple lines, which should include Arduino.h, are missing!

PaulStoffregen, 2015-11-06:
On Linux, same setup produces this:

    #include <Arduino.h>
    #line 1
    #line 1 "/tmp/untitled293481106.tmp/sketch_nov06a/sketch_nov06a.ino"
    void setup();
    void loop();
    #line 1
    void setup() {
      // put your setup code here, to run once:

    }

    void loop() {
      // put your main code here, to run repeatedly:

    }

ffissore, 2015-11-06:
What osx is it? can you paste the actual command line used to invoke arduino-builder?

PaulStoffregen, 2015-11-06:
It's 10.7.5.

PaulStoffregen, 2015-11-06:
The command line is:

/Users/paul/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware "/Users/paul/Arduino.app/Contents/Java/hardware" -tools "/Users/paul/Arduino.app/Contents/Java/tools-builder" -tools "/Users/paul/Arduino.app/Contents/Java/hardware/tools/avr" -built-in-libraries "/Users/paul/Arduino.app/Contents/Java/libraries" -libraries "/Users/paul/Documents/Arduino/libraries" -fqbn=teensy:avr:teensy31:usb=serial,speed=96opt,keys=en-us -ide-version=10606 -build-path "/var/folders/6g/mqjhk3b153gbkhzjhn__xswh0000gn/T/build01b68ddc75d190a5b880e56249c97a03.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/var/folders/6g/mqjhk3b153gbkhzjhn__xswh0000gn/T/untitled1098453877.tmp/sketch_nov06a/sketch_nov06a.ino"

PaulStoffregen, 2015-11-06:
Here's a big zip file with platform.txt, boards.txt and all the other stuff under Arduino.app/Contents/Java/hardware/teensy
http://www.pjrc.com/tmp/teensy_core_osx_issue.zip

This doesn't include the toolchain, but the problem is happening before even the first run of gcc -M.  The sketch/sketch_nov06a.ino.cpp file is created without the #include "Arduino.h", and from there everything fails.

PaulStoffregen, 2015-11-06:
This problem is really strange.  If I try copying the command from the IDE console and run it from the command line in Terminal, it works!  But when the IDE runs it, failure.

PaulStoffregen, 2015-11-06:
Another strange side effect I'm noticing, perhaps unrelated, is the sketch_nov06a.ino.d is sometimes being created in my home directory, not inside the temporary directory.

PaulStoffregen, 2015-11-06:
Is the builder program perhaps sensitive to whatever working directory or environment variables it gets?

ffissore, 2015-11-06:
No, every path it uses is specified from the command line. I've haven't yet started investigating this bug, but it sure looks weird. I checked the code that adds the include to the top and there is no conditional: that include is just added

PaulStoffregen, 2015-11-06:
I'm definitely getting a .ino.cpp file generated which does have this line added:

    #line 1 "/Users/paul/Documents/Arduino/sketch_nov06a/sketch_nov06a.ino"

but the other stuff.  It must be possible somehow.....

PaulStoffregen, 2015-11-06:
Looking at the builder source, it seems the one #line is coming from SketchSourceMerger.

It seems PrototypesAdder is having no effect.

Is there any way to make the ctags command visible?  I've got verbose turned on, but the ctags still seems to be invisible.  Something must be going wrong very early in the build.  Can the builder give me more visibility of this very early stuff?

PaulStoffregen, 2015-11-06:
This does indeed seem to be a dependency on the current working directory.  If I add this in Compiler.java, it magically starts working on OSX

         executor.setWorkingDirectory(new File("/tmp"));

PaulStoffregen, 2015-11-07:
I'm going to close this issue for now, since I have a workaround and there's no indication it impacts anyone else.  Instead, I'll start a new issue just about the ctags visibility.  Maybe later I'll dig into this more and reopen if I get more info.

closed 2015-11-07 by PaulStoffregen