arduino/arduino-builder #46: Wrong include path on multiple cores for one board

awatterott, 2015-11-04:
I have a boards.txt with the option to switch between the standard Arduino AVR core and an own one for ATtiny controller. The board package worked till Arduino v1.6.5 but with v1.6.6 it is not working, because the Arduino Builder doesn't set the correct include path.

Package:
https://github.com/watterott/wattuino/raw/master/software/Arduino/package_watterott_index.json

Parameter for Arduino Builder from the IDE, if *Standard Arduino* is selected:
`-fqbn=watterott:avr:attiny841:core=arduino`
Generated path for core:
`...Arduino15\packages\watterott\hardware\avr\1.0.1\cores\arduino`
Correct path for core:
`<ARDUINO_PATH>\hardware\arduino\avr\cores\arduino`

Parameter for Arduino Builder from the IDE, if *ATtiny841 (by Spence Konde)* is selected:
`-fqbn=watterott:avr:attiny841:core=spencekonde`
Generated path for core:
`...\Arduino15\packages\watterott\hardware\avr\1.0.1\cores\arduino`
Correct path for core:
`...\Arduino15\packages\watterott\hardware\avr\1.0.1\cores\tiny841`

My boards.txt (truncated):
```
menu.core=Core

attiny841.name=ATtiny841 (8 MHz)
# use Standard Arduino Core
attiny841.menu.core.arduino=Standard Arduino
attiny841.menu.core.arduino.build.core=arduino:arduino
attiny841.menu.core.arduino.build.variant=tiny14
# use Spence Konde Core
attiny841.menu.core.spencekonde=ATtiny841 (by Spence Konde)
attiny841.menu.core.spencekonde.build.core=tiny841
attiny841.menu.core.spencekonde.build.variant=tiny14
attiny841.build.mcu=attiny841
attiny841.build.f_cpu=8000000L
attiny841.build.board=AVR_ATTINY841
```

ffissore, 2015-11-06:
Thank you @awatterott, looking into it

closed 2015-11-06 by ffissore

ffissore, 2015-11-06:
@awatterott this issue is fixed and a newer arduino builder is available with the hourly builds http://www.arduino.cc/en/Main/Software#hourly

awatterott, 2015-11-06:
@ffissore I have tested it and it is working now. Thanks.

commit fb446067 (2015-11-06): TargetBoardResolver was handling 4th fqbn value too late and setting 'core'
to a wrong value. Fixes #46