arduino/arduino-builder #47: OSX 10.10.1 using Arduino 1.6.6: build failure message is wrong?

ffissore, 2015-11-06:
_From @i-make-robots on November 4, 2015 20:8_

```    Makelangelo-firmware/firmware_ams/firmware_ams.ino: In function 'void adjustSpoolDiameter(float)':
    firmware_ams:226: error: 'void adjustSpoolDiameter(float)' was declared 'extern' and later 'static' [-fpermissive]
     static void adjustSpoolDiameter(float diameter1) {
                                                    ^
    firmware_ams:226: error: previous declaration of 'void adjustSpoolDiameter(float)' [-fpermissive]
     static void adjustSpoolDiameter(float diameter1) {
          ^
    exit status 1
    'void adjustSpoolDiameter(float)' was declared 'extern' and later 'static' [-fpermissive]```

The keyword 'extern' does not appear in my code anywhere.  Instead I removed the keyword 'static' and Arduino was able to finish compiling.  This code was compiling fine on 1.6.4.

_Copied from original issue: arduino/Arduino#4079_

ffissore, 2015-11-06:
I've just cloned your repo https://github.com/MarginallyClever/Makelangelo-firmware and this other repo https://github.com/i-make-robots/Stewart-Platform-Crane. Then moved  `Stewart-Platform-Crane/libraries/AFMotorDrawbot/` into my sketchbook `libraries` folder. Selected Arduino/Genuino Uno and the sketch successfully compiled.
Can you add more details?

ffissore, 2015-11-06:
_From @i-make-robots on November 5, 2015 18:2_

Makelangelo firmware was already fixed when I reported the error. Add the keyword "static" to any method in the firmware and try to recompile.

ffissore, 2015-11-06:
Ok, arduino-builder omits `static` when generating prototypes. Moving the issue. Anyway, please note that your sketch won't compile anymore with latest hourly because you put libraries folders inside sketch folder. Released 1.6.6 was supposed to do recursive compilation (see https://github.com/arduino/Arduino/issues/1004#issuecomment-153891462 and subsequent comments). Hourly (and 1.6.7) will attempt to compile those libraries code and will fail when compiling examples, because the library is not where it's supposed to be and the IDE can't find it

closed 2015-11-06 by ffissore

matthijskooijman, 2015-11-06:
@ffissore, I previously reported this issue as well at https://github.com/arduino/arduino-builder/issues/30#issuecomment-152457677 and fixed it in the functionpointer branch. My fix was to *not* generate a prototype (by being more strict in the "prototype matches code" check), but I think both fixes can actually co-exist: if your code generates a correct prototype, the functionpointer branch code will accept the prototype and leave it in place. The testcases do need some manual merging in the functionpointer branch, though.

commit 539cb7d6 (2015-11-06): Preserving 'static' keyword when generating function prototypes. Fixes #47