arduino/arduino-builder #48: printing list of used libraries after compilation-errors inconvenient

FrankBoesing, 2015-11-06:
Leads to more scrolling than neccessary and it is not very useful to see the list every time.
Is it possible to move this list to the top ?
And... would it be possible to move the compilation-output to an extra window ?

PaulStoffregen, 2015-11-07:
1.6.5 printed the full library path list at the top, only in verbose mode.

It also *always* printed info about conflicts as the very last thing (before the size info) in both modes, but only when conflicting libraries matched the sketch's #include lines.

ffissore, 2015-11-09:
@FrankBoesing as @PaulStoffregen printing it at the bottom is what we did in 1.6.5 and previous versions
@PaulStoffregen atm we are always printing used and not used libraries. Should we change that to 
```
IF verbose OR conflicts
  print
```
?

FrankBoesing, 2015-11-12:
yes maybe is was the same before. that does not change anything ?
"we do it this way, we have always done it that way" is not the best path for progress..

ffissore, 2015-11-16:
As you can read from #25, we are far from that approach. Indeed, by mistake, that message was moved up but ppl expected to find it where it's printed now, at the bottom

closed 2015-11-16 by ffissore