arduino/arduino-builder #52: Run ctags only once and drop ComparePrototypesFromSourceAndPreprocSource step

matthijskooijman, 2015-11-10:
Because ctags now runs with --line-directives, the file and line information in the ctags output correctly reflects the original location of a definition. By running ctags only once on the preprocessed source, and using this information to only generate prototypes for functions defined in the sketch's .ino files, it should no longer be needed to run ctags on the unprocessed sketch source as well.

I previously discussed this with @ffissore, this ticket serves as a reminder for him to implement this later.

ffissore, 2015-11-12:
Working on it

closed 2015-11-16 by ffissore

ffissore, 2015-11-16:
Fixed. Fix will be available in next hourly build http://www.arduino.cc/en/Main/Software#hourly

matthijskooijman, 2015-11-27:
Just (finally) had a look at the commit, it looks good and a lot more clean than the old code :-D

commit 8d4dce34 (2015-11-16): CTags is now run only once, using the file it reports to filter prototypes
which belong to the sketch. Fixes #52