arduino/arduino-builder #56: "Library can't use both 'src' and 'utility' folders." error doesn't specify which library + possible bug in error

per1234, 2015-11-11:
This is actually 2 issues that compound each other. I can't replicate the second issue and it's hard to diagnose because of the first issue.

Issue 1(Arduino IDE 1.6.7 2011/11/10 12:55):
`Library can't use both 'src' and 'utility' folders.` error doesn't specify which library is causing the problem. This error breaks all compilation so the library causing it must be fixed before the IDE can be used.

Issue 2(Arduino IDE 1.6.6):
I can't reproduce this but It seems like there might be a bug with this error because users are reporting that removing correctly formatted libraries fixes the error:

http://forum.arduino.cc/index.php?topic=357312.msg2464456#msg2464456
User fixed the error by renaming C:\Program Files\Arduino\libraries\Firmata\utility  to C:\Program Files\Arduino\libraries\Firmata\xutility even though Firmata doesn't have a src folder

http://forum.arduino.cc/index.php?topic=357966.msg2468910#msg2468910
User fixed the error by deleting Ethernet, Robot_Control, SD, TFT, and WiFi libraries even though those all have correct folder structure.

per1234, 2015-11-14:
http://forum.arduino.cc/index.php?topic=357966.msg2478929#msg2478929
User fixed the error by deleting the SD library included with Arduino IDE 1.6.6. They were able to install the library with Library manager without causing the error to occur again.

closed 2015-11-16 by ffissore

ffissore, 2015-11-16:
Fixed. Fix will be available in next hourly build http://www.arduino.cc/en/Main/Software#hourly

commit 5ee45857 (2015-11-16): Added library path to error message about both src and utility folders
used by library. Fixes #56