arduino/arduino-builder #57: Please only show platform.txt warnings relevant to the selected board

ffissore, 2015-11-11:
_From @PaulStoffregen on November 7, 2015 13:39_

Extraneous warning messages are tremendously distracting for novices, who struggle to understand why their code doesn't compile, or worse yet, their code does compile but doesn't work as intended.

For example, consider a novice using an AVR board who uses digitalWrite() but neglects pinMode().  The pin will change voltage, because digitalWrite() controls the pullup resistor when the pin is in input mode, but a LED will turn on only very dim.  This situation is hard enough.... but the IDE shows this in its console window:

    Warning: platform.txt from core 'Arduino SAMD (32-bits ARM Cortex-M0+) Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
    
    Sketch uses 716 bytes (2%) of program storage space. Maximum is 32,256 bytes.
    Global variables use 9 bytes (0%) of dynamic memory, leaving 2,039 bytes for local variables. Maximum is 2,048 bytes.

This warning is irrelevant info about the SAMD platform.  The problem is **novice Arduino users do not know this message is irrelevant**.  All they know is their code uploaded and sort-of works, but the LED doesn't turn on properly, *and* the Arduino IDE is trying to warn them about something they can't understand.

Any reasonably intelligent but inexperienced person would waste time trying to figure out what this message means.  Afterall, it's printed in red... so it must be important, right?  Ultimately, it's a huge distraction from actually discovering the need to use pinMode(), or whatever is actually wrong.  Arduino should try as much as possible to help novices, and avoid distracting them.

_Copied from original issue: arduino/Arduino#4104_

per1234, 2015-11-15:
I recommend only showing these warnings when Boards Manager is opened. There's no reason to show it every time anything is compiled. Any 3rd party hardware files author who cares about their users won't update the recipe because that will [break backwards compatibility](https://github.com/arduino/Arduino/issues/4114). Showing the warning when Library Manager is opened is completely pointless.

matthijskooijman, 2015-11-16:
I suspect the warnings are shown whenever the boards are (re)loaded, which apparently happens before building. Agreed that it is pointless to show warnings unrelated to the current build / board, though.

commit c3f43f86 (2015-11-23): Platform rewrite warnings are now emitted only for the actual selected library.
Fixes #57
closed 2015-11-23 by ffissore