arduino/arduino-builder #59: DirectIO does not compile under Arduino 1.6.6 AVR 1.6.x

arkhipenko, 2015-11-11:
Hi,
DirectIO no longer compiles under the new build process with Arduino 1.6.6.  I tried with several board definitions (v1.6.9, 1.6.8 and 1.6.7) 
This works with IDE 1.6.5 avr 1.6.7 and 1.6.8

https://github.com/mmarchetti/DirectIO.git

Below is the output form compiling the first DirectIO example:

	`Arduino: 1.6.6 (Windows 8.1), Board: "Arduino/Genuino Uno"

	In file included from C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/DirectIO.h:23:0,

					 from C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO\examples\direct_io\direct_io.ino:1:

	C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:56:22: error: field initializer is not constant

	 _define_port(PORT_B, PINB, PORTB, DDRB);

						  ^

	C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:46:31: note: in definition of macro '_define_port'

			 static const u16 in = IN_REG; \

								   ^

	C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:56:28: error: field initializer is not constant

	 _define_port(PORT_B, PINB, PORTB, DDRB);

								^

	C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:47:32: note: in definition of macro '_define_port'

			 static const u16 out = OUT_REG; \

									^

	C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:56:35: error: field initializer is not constant

	 _define_port(PORT_B, PINB, PORTB, DDRB);

									   ^

	C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:48:32: note: in definition of macro '_define_port'

			 static const u16 dir = DIR_REG; \

									^

	C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:60:22: error: field initializer is not constant

	 _define_port(PORT_C, PINC, PORTC, DDRC);

						  ^

	C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:46:31: note: in definition of macro '_define_port'

			 static const u16 in = IN_REG; \

								   ^

	C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:60:28: error: field initializer is not constant

	 _define_port(PORT_C, PINC, PORTC, DDRC);

								^

	C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:47:32: note: in definition of macro '_define_port'

			 static const u16 out = OUT_REG; \

									^

	C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:60:35: error: field initializer is not constant

	 _define_port(PORT_C, PINC, PORTC, DDRC);

									   ^

	C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:48:32: note: in definition of macro '_define_port'

			 static const u16 dir = DIR_REG; \

									^

	C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:64:22: error: field initializer is not constant

	 _define_port(PORT_D, PIND, PORTD, DDRD);

						  ^

	C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:46:31: note: in definition of macro '_define_port'

			 static const u16 in = IN_REG; \

								   ^

	C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:64:28: error: field initializer is not constant

	 _define_port(PORT_D, PIND, PORTD, DDRD);

								^

	C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:47:32: note: in definition of macro '_define_port'

			 static const u16 out = OUT_REG; \

									^

	C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:64:35: error: field initializer is not constant

	 _define_port(PORT_D, PIND, PORTD, DDRD);

									   ^

	C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:48:32: note: in definition of macro '_define_port'

			 static const u16 dir = DIR_REG; \

									^

	exit status 1
	Error compiling.`

ffissore, 2015-11-16:
@cmaglie @facchinm any hints?

arkhipenko, 2015-11-16:
Discovered a second ago:

Under IDE 1.6.5 avr 1.6.8 the first DirectIO example will compile OK

    #include <DirectIO.h>
    Output<2> pin;
    void setup() {} 
    void loop() {
      while(1) {
        pin = HIGH;
        pin = LOW;
      }
    }

But as soon as I place <Arduino.h> before <DirectIO.h> it does not (which what it is now with v1.6.6):

    #include <Arduino.h>
    #include <DirectIO.h>
    Output<2> pin;
    void setup() {} 
    void loop() {
      while(1) {
        pin = HIGH;
        pin = LOW;
      }
    }

same error as reported before.
It seems to me the changes in where header files are placed in 1.6.6 might have something to do with it.

facchinm, 2015-11-16:
Yes, the inclusion on the first line of "Arduino.h" breaks the compilation. I believe there is a reason for putting it there but I must ask @ffissore first before restoring the old behaviour.

facchinm, 2015-11-16:
The issue can be worked around by patching `ports.h`

```patch
diff --git a/DirectIO/ports.h b/DirectIO/ports.h
index 42aec20..6666f5e 100644
--- a/DirectIO/ports.h
+++ b/DirectIO/ports.h
@@ -22,10 +22,13 @@
 
 #undef _AVR_COMMON_H
 #undef _AVR_IO_H_
+#undef _AVR_IOXXX_H_
 #undef _AVR_SFR_DEFS_H_
+#undef _SFR_ASM_COMPAT
 
 // _SFR_ASM_COMPAT enables the port numbers (e.g. PORTA) to be plain integers so they can be used as template parameters.
 #define _SFR_ASM_COMPAT 1
+#include <avr/io.h>
 
 #ifndef _AVR_EEPROM_H_
 // avr/eeprom.h isn't compatible with _SFR_ASM_COMPAT, so prevent its inclusion
```

Only tested with a UNO but it should work with every board. 
The new behaviour solves a lot of existing issues (notably arduino/Arduino#1518, arduino/Arduino#1310, arduino/Arduino#1328, arduino/Arduino#1468 and many more) so we are not willing to revert the previous behaviour.

cmaglie, 2015-11-16:
@mmarchetti, since you're the author of the library, what do you think?

is  this an acceptable patch for DirectIO?

arkhipenko, 2015-11-16:
Confirmed. Patch works for Uno and Nano. I just recompiled and retested a bunch of sketches and all is well. Thank you for your help!

closed 2015-11-16 by arkhipenko