arduino/arduino-builder #60: Incorrect "Library can't use both 'src' and 'utility' folders." errors

per1234, 2015-11-14:
I previously reported this as a possible issue with https://github.com/arduino/arduino-builder/issues/56 but as I continue to see more and better reports of this problem I think it now deserves a dedicated issue report.

Users are reporting that after installing Arduino IDE 1.6.6 they are unable to compile because of the `Library can't use both 'src' and 'utility' folders.` error. They eventually track it down(made very difficult because of https://github.com/arduino/arduino-builder/issues/56) to a library that has a correct folder structure and when that library is removed the error goes away. I am unable to reproduce this issue:

Here are the reports of this issue I have found:

http://forum.arduino.cc/index.php?topic=357312.msg2464456#msg2464456
User fixed the error by renaming C:\Program Files\Arduino\libraries\Firmata\utility to C:\Program Files\Arduino\libraries\Firmata\xutility even though Firmata doesn't have a src folder

http://forum.arduino.cc/index.php?topic=357966.msg2468910#msg2468910
User fixed the error by deleting Ethernet, Robot_Control, SD, TFT, and WiFi libraries even though those all have correct folder structure.
Update: http://forum.arduino.cc/index.php?topic=357966.msg2474958#msg2474958
User reports that all those libraries were included with the Windows 32 bit 1.6.6 installer package and had the correct folder structure.

http://forum.arduino.cc/index.php?topic=357966.msg2478929#msg2478929
User fixed the error by deleting the SD library included with Arduino IDE 1.6.6(Windows 32 bit). They were able to install the library with Library manager without causing the error to occur again. After moving the SD library from s

https://forum.arduino.cc/index.php?topic=354299.msg2442472#msg2442472
Error was caused by C:\Program Files\Arduino\libraries\Firmata after deleting that library the error went away.

https://forum.arduino.cc/index.php?topic=354299.msg2463515#msg2463515
Possibly also caused by Firmata.

https://forum.arduino.cc/index.php?topic=358683.msg2472874#msg2472874
User didn't find which library was causing the error.

All of the reports that stated operating system were Windows 32 bit.

ffissore, 2015-11-16:
Thank you @per1234 for your amazing work. I've added library path to the error message. I've double checked the code, and it's correct. My only guess is that users mixed libraries of different versions. Later today, hourly builds will ship the new error message. Please ping us if you can come up with a set of reproducible steps

soundanalogous, 2015-12-12:
I wish people would post Firmata issues to the [Firmata repo](https://github.com/firmata/arduino). Anyway... what's strange is that Firmata doesn't have a `src` directory, it only has a `utility` directly. I did it this way so I wouldn't have to maintain separate builds for Arduino 1.0.x and 1.5.x+. Is it required as of Arduino 1.6.6 that all code must be in a `src` directory and that `utility` (if used) must be a subdirectory of `src`?

per1234, 2015-12-12:
I don't think it's a Firmata issue, some of the Arduino Built-In libraries also were reported to cause the error. This error is only supposed to occur when there is a `src` and a `utility` folder in the library root folder which none of the libraries reported to cause the error have. I was never able to reproduce this error on my system and there have been no reports of it for a couple weeks now(I've been editing the issue to add new reports as they are posted on the forum). The Firmata library(both the version included with 1.6.6 and the most recent version) as it is has never caused the error on my system and I'm sure there are thousands of other 1.6.6 users with Firmata installed who are not having this issue.

soundanalogous, 2015-12-12:
Thanks for the clarification. I was responding more to the number of Firmata issues referenced in the Arduino forum that never made it to the repo. I guess I need to scrape the forum occasionally to pick up those issues.

per1234, 2016-01-20:
I haven't seen any reports of this issue for a few weeks now and have never been able to reproduce it or get any of the reporting users to work with me to track down the source of the problem so I'm going to go ahead and close this.

closed 2016-01-20 by per1234