arduino/arduino-builder #65: 1.6.7 hourly build directories issue

ffissore, 2015-11-16:
_From @pierrotm777 on November 8, 2015 14:18_

If a directory exist into the same folder as the main .ino file, all files .ino found into the directory are read by the compiler and return an error .
This issue existed into the 1.6.6 hourly build and solved with the 1.6.6 final version .

_Copied from original issue: arduino/Arduino#4111_

ffissore, 2015-11-16:
_From @per1234 on November 9, 2015 2:36_

@pierrotm777 are you referring to issue https://github.com/arduino/Arduino/issues/3909?

ffissore, 2015-11-16:
_From @pierrotm777 on November 10, 2015 9:13_

Yes , this is the same issue !

ffissore, 2015-11-16:
`.ino` files should belong to main folder only, not subfolders

pierrotm777, 2015-11-16:
In fact , this is the case but before that wasn't a problem !
I like to keep some older file into subfolders .

ffissore, 2015-11-16:
Yes, @pierrotm777. What I meant to say is that (maybe) arduino-builder is collecting even .ino file in subfolders, while it really shouldn't. Still to confirm though

pierrotm777, 2015-11-16:
Yes , this exact !

ffissore, 2015-11-17:
@pierrotm777 I've verified that arduino-builder does NOT load .ino files stored in subfolders of the main .ino file. This is the code that makes it happen https://github.com/arduino/arduino-builder/blob/master/src/arduino.cc/builder/sketch_loader.go#L117-L119
Can you share the sketch you are using?

pierrotm777, 2015-11-17:
[SynchTwinRcEngine_WithMacros_V3.zip.txt](https://github.com/arduino/arduino-builder/files/36634/SynchTwinRcEngine_WithMacros_V3.zip.txt)
I send you my sources . 
The error is:
C:\Users\pierre\AppData\Local\Temp\build11326b2d920aae776cf88a2171973031.tmp\sketch\Telemetrie en cours\M-Link\main.c:50:17: fatal error: htc.h: No such file or directory

 #include <htc.h>

                 ^

compilation terminated.

ffissore, 2015-11-17:
Ahh so it's not about .ino files in subfolders, it's about .c/.cpp files... Ok, then I'm sorry but this was a feature added some months ago as a subfeature of https://github.com/arduino/Arduino/issues/3080

pierrotm777, 2015-11-17:
Ok, but the 1.6.6 hasn't this issue !
Do i delete this subfolder or you think to do something ? :-)

sej7278, 2016-01-03:
i'm finding the same problem with 1.6.7 but not 1.6.6 - if you output to build-path, then re-run arduino-builder, it seems to create a subdirectory of build-path with the previous build in it or something odd, you end up with a directory structure like this:

```
$ tree -L 5 build-esp8266/
build-esp8266/
├── arduino.ar
├── build.options.json
├── core
│   ├── abi.cpp.d
│   ├── abi.cpp.o
│   ├── blah....
│   ├── WString.cpp.d
│   └── WString.cpp.o
├── gizwits_test.ino.bin
├── libraries
├── preproc
│   └── ctags_target_for_gcc_minus_e.cpp
└── sketch
    ├── build-esp8266
    │   ├── preproc
    │   │   ├── ctags_target_for_gcc_minus_e.cpp
    │   │   ├── ctags_target_for_gcc_minus_e.cpp.d
    │   │   └── ctags_target_for_gcc_minus_e.cpp.o
    │   └── sketch
    │       ├── gizwits_test.ino.cpp
    │       ├── gizwits_test.ino.cpp.d
    │       └── gizwits_test.ino.cpp.o
    ├── gizwits_test.ino.cpp
    ├── gizwits_test.ino.cpp.d
    └── gizwits_test.ino.cpp.o
```

and these sort of errors:

```
~/arduino16/gizwits_test/build-esp8266/sketch/build-esp8266/preproc/ctags_target_for_gcc_minus_e.cpp.o: In function `HardwareSerial::begin(unsigned long)':

~/arduino16/hardware/esp8266com/esp8266/cores/esp8266/HardwareSerial.h:77: multiple definition of `setup'
```

matthijskooijman, 2016-01-04:
@sej7278, again, I think your issue is different from this issue to, so it's best reported separately.

As for the original issue here, it seems that now any source files (.c / .cpp) in subdirectories of the sketch folder are compiled. @pierrotm777 has some documentation or example files, which include a .c file, which he does not intend to compile, but they are compiled (which didn't happen before). I'm not sure if this is an intentional change, though. To fix #3080, all files in the sketch directory were copied in to the build directory, so the preprocessed .ino files could refer to these files (e.g. #include a file in a subdirectory). However, it seems that in addition to copying the files, any source files in subdirectories are now also compiled, which could be viewed as a new feature, but it also potentially breaks sketches (and no longer provides any place to "hide" source files in a sketch). I haven't looked at the source code in detail yet, but perhaps this should be fixed for the next release, before people start relying on this behaviour?

sej7278, 2016-01-04:
moved my comments to #89

matthijskooijman, 2016-05-18:
This was fixed by #148, which should be in the upcoming 1.6.10 released.

closed 2016-05-18 by matthijskooijman