arduino/arduino-builder #67: "open ctags_target_for_gcc_minus_e.cpp: no such file or directory"

mplewis, 2015-11-19:
Hello! I'm trying to compile a Hello World sketch against my hardware definition. I'm running into an error and I'm not sure what to do next.

```
$ ./arduino-builder -verbose -preprocess -compile -hardware=/Users/mplewis/tmp/hw -tools=/Users/mplewis/tmp/tools -fqbn=punchthrough:avr:lightblue-bean -build-path=/Users/mplewis/tmp/out /Users/mplewis/tmp/sketch/good_sketch.ino
Warning: platform.txt from core 'Arduino AVR Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
WARNING: Category '' in library EEPROM is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library SPI is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library SoftwareSerial is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library Wire is not valid. Setting to 'Uncategorized'
"/Users/mplewis/tmp/tools/avr/bin/avr-g++"  -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -DF_CPU=8000000L -DARDUINO=10600 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR         "-I/Users/mplewis/tmp/hw/punchthrough/avr/cores/bean" "-I/Users/mplewis/tmp/hw/punchthrough/avr/variants/bean" "/Users/mplewis/tmp/out/sketch/good_sketch.ino.cpp"
"/Users/mplewis/tmp/tools/avr/bin/avr-g++"  -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -DF_CPU=8000000L -DARDUINO=10600 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR         "-I/Users/mplewis/tmp/hw/punchthrough/avr/cores/bean" "-I/Users/mplewis/tmp/hw/punchthrough/avr/variants/bean" "/Users/mplewis/tmp/out/sketch/good_sketch.ino.cpp"
"/Users/mplewis/tmp/tools/avr/bin/avr-g++"  -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -DF_CPU=8000000L -DARDUINO=10600 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR         "-I/Users/mplewis/tmp/hw/punchthrough/avr/cores/bean" "-I/Users/mplewis/tmp/hw/punchthrough/avr/variants/bean" "/Users/mplewis/tmp/out/sketch/good_sketch.ino.cpp"
open /Users/mplewis/tmp/out/preproc/ctags_target_for_gcc_minus_e.cpp: no such file or directory
```

Build stops at this point. `out/preproc/ctags_target_for_gcc_minus_e.cpp` does not exist.

It seems like Builder is looking for a file that doesn't exist, but I'm not sure why. Can anyone shed some light on the issue I am facing?

ffissore, 2015-11-19:
What core are you using?

mplewis, 2015-11-19:
I'm using the core `punchthrough:avr:lightblue-bean`, a custom core we use for @punchthrough's [LightBlue Bean](https://punchthrough.com/bean). But I see the same error when I try to compile against `arduino:avr:uno`:

```
./arduino-builder -verbose -preprocess -compile -hardware=/Users/mplewis/tmp/hw -tools=/Users/mplewis/tmp/tools -fqbn=arduino:avr:uno -build-path=/Users/mplewis/tmp/out /Users/mplewis/tmp/sketch/good_sketch.ino
Warning: platform.txt from core 'Arduino AVR Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
Build options changed, rebuilding all
"/Users/mplewis/tmp/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10600 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Users/mplewis/tmp/hw/arduino/avr/cores/arduino" "-I/Users/mplewis/tmp/hw/arduino/avr/variants/standard" "/Users/mplewis/tmp/out/sketch/good_sketch.ino.cpp"
"/Users/mplewis/tmp/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10600 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Users/mplewis/tmp/hw/arduino/avr/cores/arduino" "-I/Users/mplewis/tmp/hw/arduino/avr/variants/standard" "/Users/mplewis/tmp/out/sketch/good_sketch.ino.cpp"
"/Users/mplewis/tmp/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10600 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Users/mplewis/tmp/hw/arduino/avr/cores/arduino" "-I/Users/mplewis/tmp/hw/arduino/avr/variants/standard" "/Users/mplewis/tmp/out/sketch/good_sketch.ino.cpp"
open /Users/mplewis/tmp/out/preproc/ctags_target_for_gcc_minus_e.cpp: no such file or directory
```
Here's [an archive of my `hw`, `src`, `tools`, and `out` folders](https://punchthrough.com/files/open-ctags-target-build-failure.zip) if you want to try and reproduce this on your end.

ffissore, 2015-11-23:
Have you already tried latest hourly build? http://www.arduino.cc/en/Main/Software#hourly File platform.txt in `hw` misses a param

ffissore, 2015-11-25:
Sorry I've just noticed you are calling builder with -preprocess and providing hw folders on your own. Which version of arduino-builder are you using? I see platform.txt files in `hw` are quite old.

Moreover, I see you have are defining a platform.local.txt which is almost a copy of the original avr one. Instead, you should specify a build.core like `arduino:bean` in your boards.txt, that will help you inherit almost everything from avr platform, without having you to copy/paste/maintain them

mplewis, 2015-11-25:
I would like to try that. We have had problems with outdated Arduino core code in the past.

Are there any guides/docs on how to use a core that inherits from the AVR platform?

ffissore, 2015-11-25:
Here: https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification#referencing-another-core-variant-or-tool

mplewis, 2015-11-25:
That's great, thanks for the info! I'll close this for now since it is not clear that this is an issue in arduino-builder.

closed 2015-11-25 by mplewis

lovettchris, 2018-07-05:
I see the same problem in VS Code when it invokes Arduino-builder, and the relative path trick didn't solve it fully.  If I change library code the build fails with:
```
Build options changed, rebuilding all
open preproc\ctags_target_for_gcc_minus_e.cpp: The system cannot find the path specified.
```
so I have to delete my local 'build' folder and try again, then it will work.  So incremental build is broken...

gajop, 2018-07-20:
Same as above. It seems relative build doesn't work well. This issue should probably be reopened.

cmaglie, 2018-07-26:
Please open a new issue with a step-by-step procedure to reproduce the problem.

gajop, 2018-07-26:
I cannot easily reproduce it. Seems to be occurring frequently but with no apparent cause. Some bugs are just not easily reproducible :)