arduino/arduino-builder #68: preprocessor fails on #include "ESP8266wifi.h"  (for external ESP8266 addon)

ladyada, 2015-11-21:
there's something about this header file that crashes the preprocessor, probably indicates some more general purpose bug. reproduce by:
   * add `http://arduino.esp8266.com/stable/package_esp8266com_index.json` to board manager prefs
   * use board manager to install ESP8266 support
   * try to compile this sketch
```
#include "ESP8266WiFi.h"
void setup() {}
void loop() { hello(); }
void hello(void) {}
```
  * will fail due to not having auto-generated prototype for hello()
  * now put `#include "ESP8266WiFi.h"` on last line, recompile
  * works now because while it does crash on the #include, its after the prototypes are made (?)

Also mentioned @ https://github.com/esp8266/Arduino/issues/1066

igrr, 2015-11-21:
The `#include` which triggers the issue is `#include <memory>` in WiFiClient.h.
I have created a library with a single .h file (Dummy.h):
```c++
#include <memory>
```
Including this library in the sketch seems to be enough to trigger the issue.

igrr, 2015-11-21:
I pulled the latest git version of Arduino and rebuilt it. Also renamed `test` to `myfunc` to make it easier to grep.

Here's arduino-builder output: http://hastebin.com/mupuhalehe.cs
ctags_target_for_gcc_minus_e.cpp: https://gist.github.com/igrr/3095758a66bc28f7db23 (too large for hastebin:)
ctags output: https://gist.github.com/igrr/805dc5ef96cf9c0a0892
Note that there's really no `myfunc` in ctags output.

I had a thought that ctags [gets confused](http://stackoverflow.com/questions/31329387/does-exuberant-ctags-support-c11) by some of the c++11 constructs. Tried [universal-ctags](https://github.com/universal-ctags/ctags), and output looks much better (i.e. `noexcept` is no longer tagged as a function), but still it doesn't have `myfunc`.

tigoe, 2015-11-22:
Haven't got a solution but I can reproduce all of the errors down to the rebuild reliably, and it definitely seems to be what caused #4137.

ffissore, 2015-11-23:
At a first look, it looks like ctags is interrupting. It's not getting beyond that `tuple_element`. I'm investigating why

igrr, 2015-11-23:
For the record, here's what i get from universal-ctags: https://gist.github.com/igrr/b82b708af81e7baa02fc
Unlike ctags shipped with Arduino, it is able to parse newer C++ keywords like `noexcept`. Still doesn't return full output though.

ffissore, 2015-11-23:
Indeed, even universal-ctags misses `myfunc`. Rebasing our ctags fork on top of `universal-ctags` is a nice-to-have (and tracked at https://github.com/arduino/ctags/issues/2) but currently not in my todo list

ffissore, 2015-11-23:
I'm now using this 
```arduino
#include "ESP8266WiFi.h"
void setup() {
}
void loop() {
  myfunc();
}
void myfunc() {
}
```

When running ctags with `-V` (verbose) at the top of its output it prints
`/tmp/arduino_12f75e84123cf24fa7ce3d06ac6342b3/sketch_nov23a.ino: failed to find match for '(' at line 36991`
36991 is the last line of ctags_target_for_gcc_minus_e.cpp. A similar error message is displayed even after manually adding the prototype at the top of the sketch or removing all functions and leaving the sketch with just the first `#include`.

ffissore, 2015-11-23:
Same does universal-ctags

matthijskooijman, 2016-01-20:
#88 also ran into a problem with ESP8266wifi.h. I dug into that, without being aware of this issue and found that static_assert was the culprit. I wrote:

> I've found the cause of the problem: The static_assert keyword, which is
used by the ESP / libstdc++ include files, is confusing ctags and
prevents it from emitting information about subsequent functions. I've
filed a report here: https://sourceforge.net/p/ctags/bugs/370/ (though
exuberant ctags doesn't seem to be really maintained anymore, so this
report serves more as documentation than that I really expect it to be
fixed).

> The proper fix would be to add support for static_assert in the Arduino
ctags version, but I'm afraid that might not be so easy. I won't have
time for looking at that on the short term.

Looking through the comments here, I'm not sure if this is exactly the same issue, though that might be caused by changes in the ESP core since this issue was created. I haven't tried universal-ctags to see what that gives.

igrr, 2016-01-20:
Oh wow, thanks for info Matthijs. I was going through the logs multiple times but missed this `static_assert` somehow.
I am pretty sure the issue was not related to the ESP core itself. It was easy to reproduce by including `<memory>` in the sketch, which is a libstdc++ header.
I'll give a shot at writing a patch for ctags (should be a matter of adding this missing keyword).

matthijskooijman, 2016-01-20:
@igrr, yeah I think in my testing, the offending `static_assert` was in `<array>`. Having a patch would be great, thanks for looking at that!

igrr, 2016-01-20:
@matthijskooijman here you go: https://github.com/arduino/ctags/pull/4

matthijskooijman, 2016-01-20:
@igrr, awesome!

I just tested that PR with the example from my sourceforge bug report and the sketch from #88, which both work now. @cmaglie, can you take care of building binaries for ctags and doing an arduino-builder release?

closed 2016-01-21 by cmaglie

cmaglie, 2016-01-22:
@igrr
The new ctags is in the current hourly build of Arduino IDE, would you like to test it?

igrr, 2016-01-22:
Thanks, I can test it on weekends. Will report here.

stefangordon, 2016-01-23:
@cmaglie I can confirm that this is fixed in Arduino 1.6.8 Hourly Build 2016/01/21 04:46

igrr, 2016-01-24:
Thanks, works for me with the latest hourly (21 January 2016 18:15:49 GMT).

matthijskooijman, 2016-01-24:
Thanks for confirming!

mkeyno, 2016-02-03:
Dear All (@igrr, @ matthijskooijman   ,@ stefangordon, @......... ) , so what we should do , should I include the dummy header file or just waiting for 1.6.8? also I find many new changes in esp core file in last stable version ! I also update my library to this last version , I really need stable version and stable IDE for my project , I hope you guys help me out soon

matthijskooijman, 2016-02-03:
@mkeyno, I'm not sure if there's a workaround for this issue really. What dummy file are you referring to?

igrr, 2016-02-03:
@mkeyno I think you can get an hourly build of Arduino, it has this issue
fixed.

https://www.arduino.cc/en/Main/Software (see Hourly Builds section).

stefangordon, 2016-02-03:
Yeah we have been pairing esp8266/arduino Staging with Arduino Hourly for a pretty good experience so far.  Eager to see Arduino push 1.6.8 too, but I suspect that isn't too far away.

mkeyno, 2016-02-03:
hi guys , glad to have your quick responds , I've read the gommon tip [http://www.gammon.com.au/forum/?id=12625](url) and I'm referring to Mr Igrr note on Arduino  pre possessing issue 

> The #include which triggers the issue is #include <memory> in WiFiClient.h.
> I have created a library with a single .h file (Dummy.h):

however has anyone check this issue on the Arduino Hourly version?
also I have another question about reliability of new staging library , cause it seems for sometime I have to refresh the browser and re download the page form SPIFF whereas for previous library , the requested page totally downloaded on the first request?

mkeyno, 2016-02-04:
@igrr 
I downloaded the hourly version but not fixed , I've got same compiler error, I don't know what should I suppose to do? 

captive advance portal example
*********
exit status 1
'handleRoot' was not declared in this scope

stefangordon, 2016-02-04:
If the problem is related to this issue you can work around it by providing your own function prototypes as you would typically do in a C application.  You just need to define the functions at the top of your .ino for example, or potentially in a separate header file.

igrr, 2016-02-04:
@matthijskooijman Looks like we have another issue due to use of `enum class` feature in DNSServer library: https://github.com/esp8266/Arduino/blob/master/libraries/DNSServer/src/DNSServer.h#L9
ctags is not aware of this new C++11 construct, which breaks parsing. I'll attempt a pull request to arduino-ctags.

igrr, 2016-02-04:
I have cherry-picked two more changes from geany. This fixes an issue with DNSServer library:
https://github.com/arduino/ctags/pull/5
@cmaglie Could you please update ctags again? Thanks.

julianh2o, 2016-02-23:
How does one generate the useful debugging output for this bug? This issue is closed, but it seems like including the FastLED library is still causing issues:

    #include "FastLED.h"
    void setup() { myctagstestfunc(); }
    void myctagstestfunc() { }
    void loop() {}
 
Fails with: error: 'myctagstestfunc' was not declared in this scope

On the 1.6.8 hourly build 2016/02/22 10:12 (downloaded just now)

According to FastLED/FastLED#258, this is still an arduino-builder issue, is that the case?

Should this be reopened?

focalintent, 2016-02-24:
@julianh2o I just opened up #120 for the specific issue you're seeing there (since I think this is being caused by slightly different things that were causing the problems here).

focalintent, 2016-02-24:
Pull request for a fix to #120 opened here - https://github.com/arduino/ctags/pull/7

vmanan, 2016-03-06:
Folks,

I tried hourly build 2016/03/04 and still seeing the error. Isnt the original error "memory" fixed?

In file included from C:\Users\vmanan\Documents\Arduino\libraries\src/ESP8266WiFi.h:32:0,
                 from myrobot.ino:3:
C:\Users\vmanan\Documents\Arduino\libraries\src/WiFiClient.h:28:18: fatal error: memory: No such file or directory
 #include <memory>

joeuser846, 2016-04-05:
This still happens with 1.6.8.  What was/is the resolution?

facchinm, 2016-04-06:
@joeuser846 , it is fixed in 1.6.8+ (just tested with the initial sketch and ESP core 2.1.0)
Can you post the complete error?

netmindz, 2016-04-15:
I'm still seeing issues with 1.6.8

igrr, 2016-04-15:
@netmindz Possibly you are seeing #1898, caused by https://github.com/arduino/ctags/issues/10.
Try removing ESP8266WiFiMulti if you are using it.

stephanemorin, 2016-04-18:
I confirm I'm still seeing issues with 1.6.8 and ESP8266Wifi.
No improvement with the latest hourly build.
Not able to use the ESP8266wifi library anymore.

igrr, 2016-04-18:
@stephanemorin please provide the sketch which causes issues. Thanks.

stephanemorin, 2016-04-22:
Hi igrr,

The following files from the ESPWifi library examples are all causing the same problem:
WifiClient.ino
WifiClientBasic.ino
WifiWebServer.ino

The error message is:

ESP8266WiFi\src/WiFiClient.h:24:18: fatal error: memory: No such file or directory
 #include <memory>
                  ^
compilation terminated.
exit status 1
Error compiling for board Arduino/Genuino Uno.

19eighties, 2016-10-10:
I am seeing a similar issue with 1.6.9 and ESP8266WiFiMulti.h.