arduino/arduino-builder #70: WARNING: Spurious .git folder in library

ffissore, 2015-11-23:
_From @hryamzik on November 23, 2015 1:18_

Due to #3237 there shouldn't be such a warning but I do get in in Arduino IDE v1.6.6.

_Copied from original issue: arduino/Arduino#4204_

closed 2015-11-23 by ffissore

NicoHood, 2015-11-23:
I still get this warning for other folders like this:
`WARNING: Spurious .dev folder in 'HID-Project' library`

It would be cool to remove this as well.

ffissore, 2015-11-24:
The silenced warnings were about SCCS sofware. It's expected that warnings about actual spurious directories are notified

ladyada, 2016-06-20:
heya @ffissore - could .github also be ignored? its used to store issue templates and (likely in the future) other github configurations and setups, and will get included in the library zip. 

thank you :)

ffissore, 2016-06-20:
Hi @ladyada sorry you missed the news but I don't work for arduino any more. See [this message](https://groups.google.com/a/arduino.cc/forum/#!msg/developers/YJLX6AZHem4/kpsLef7XBAAJ). My former colleagues @facchinm @cmaglie and @sandeepmistry will take care of your issue

ladyada, 2016-06-20:
@ffissore apologies for the mixup - and good luck with all your endeavours! :)

matthijskooijman, 2016-06-21:
@ladyada, I created a PR for that: https://github.com/arduino/arduino-builder/pull/162

Thanks for the suggestion :-)

ladyada, 2016-06-21:
love it, thanks!

commit f5edd23c (2015-11-23): Avoid warning about SCCS folders in libraries, just ignore them. Fixes #70