arduino/arduino-builder #76: ctags pattern is missing

spiderkeys, 2015-12-01:
I am trying to do something fairly unconventional, which is trying to use arduino-builder on a Beaglebone Black (armhf single board computer) to build an Arduino sketch with the latest 1.6.6 Arduino files. arduino-builder is required, since the distribution itself is CLI only, no display (and no one has built an armhf port of 1.6.6). Here are the steps I have taken (all on a beaglebone):

- Cloned the latest Arduino master branch on Git to get the latest /hardware/ folder
- Installed the latest avr-gcc toolchain (4.8.1) to /usr/bin/
- Successfully built arduino-builder
- Created a test sketch which is an empty init() and a loop(){ Delay(1000); }
- Added the line "mega.build.mcu=atmega2560" to Arduino/hardware/arduino/avr/boards.txt, since it was missing for some reason and caused arduino-builder to error with an empty -mmcu flag
- Executed the following command:

```
./arduino-builder -hardware="/root/workspace/Arduino/hardware/" -tools="/usr/bin/" -fqbn=arduino:avr:mega -verbose test.ino 
``` 

Following is the resulting output. The apparent error is "ctags pattern is missing"

```
"/usr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10600 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/root/workspace/Arduino/hardware/arduino/avr/cores/arduino" "-I/root/workspace/Arduino/hardware/arduino/avr/variants/mega" "/tmp/arduino-sketch-F65E51C7CC67F8F1D777CCD86FAE2D65/sketch/test.ino.cpp" -o "/tmp/arduino-sketch-F65E51C7CC67F8F1D777CCD86FAE2D65/preproc/test.ino.cpp_134020434_preprocessed.cpp"

"/usr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10600 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/root/workspace/Arduino/hardware/arduino/avr/cores/arduino" "-I/root/workspace/Arduino/hardware/arduino/avr/variants/mega" "/tmp/arduino-sketch-F65E51C7CC67F8F1D777CCD86FAE2D65/sketch/test.ino.cpp" -o "/tmp/arduino-sketch-F65E51C7CC67F8F1D777CCD86FAE2D65/preproc/test.ino.cpp_1597969999_preprocessed.cpp"

"/usr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10600 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/root/workspace/Arduino/hardware/arduino/avr/cores/arduino" "-I/root/workspace/Arduino/hardware/arduino/avr/variants/mega" "/tmp/arduino-sketch-F65E51C7CC67F8F1D777CCD86FAE2D65/sketch/test.ino.cpp" -o "/tmp/arduino-sketch-F65E51C7CC67F8F1D777CCD86FAE2D65/preproc/ctags_target_for_gcc_minus_e.cpp"
ctags pattern is missing
```

I tried grepping through the arduino-builder folder, but couldn't find that specific error string anywhere. I then realized that the machine itself didn't have ctags installed, so I ran apt-get install exuberant-ctags, which added ctags to /usr/bin/ as well. No change observed.

Any ideas on how I might fix this problem? 

Also, I notice that there is a arduino-builder/src/arduino.cc/builder/hardware/platform.txt file which sets the variables:

```
tools.ctags.path={runtime.tools.ctags.path}
tools.ctags.cmd.path={path}/ctags
tools.ctags.pattern="{cmd.path}" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "{source_file}"
```

Where are these being used? They aren't present in my Arduino/hardware/..../platform.txt file, and as far as I can tell, the arduino-builder executable doesn't know anything about the platform.txt file in  its build folder (unless those options were built into it somehow when I ran the Go build?)

spiderkeys, 2015-12-01:
After reading around a bit more, I added the lines from that partial platform.txt in this repo to my Arduino/.../platform.txt, with
```
tools.ctags.path={runtime.tools.ctags.path}
```
modified to
```
tools.ctags.path=/usr/bin/
```

This let me get a bit further and now the resulting output is:

```
"/usr/bin/avr-g++"  -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -DF_CPU=16000000L -DARDUINO=10600 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR         "-I/root/workspace/Arduino/hardware/arduino/avr/cores/arduino" "-I/root/workspace/Arduino/hardware/arduino/avr/variants/mega" "/tmp/arduino-sketch-F65E51C7CC67F8F1D777CCD86FAE2D65/sketch/test.ino.cpp" -o "/tmp/arduino-sketch-F65E51C7CC67F8F1D777CCD86FAE2D65/preproc/test.ino.cpp_134020434_preprocessed.cpp"
"/usr/bin/avr-g++"  -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -DF_CPU=16000000L -DARDUINO=10600 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR         "-I/root/workspace/Arduino/hardware/arduino/avr/cores/arduino" "-I/root/workspace/Arduino/hardware/arduino/avr/variants/mega" "/tmp/arduino-sketch-F65E51C7CC67F8F1D777CCD86FAE2D65/sketch/test.ino.cpp" -o "/tmp/arduino-sketch-F65E51C7CC67F8F1D777CCD86FAE2D65/preproc/test.ino.cpp_1597969999_preprocessed.cpp"
"/usr/bin/avr-g++"  -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -DF_CPU=16000000L -DARDUINO=10600 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR         "-I/root/workspace/Arduino/hardware/arduino/avr/cores/arduino" "-I/root/workspace/Arduino/hardware/arduino/avr/variants/mega" "/tmp/arduino-sketch-F65E51C7CC67F8F1D777CCD86FAE2D65/sketch/test.ino.cpp" -o "/tmp/arduino-sketch-F65E51C7CC67F8F1D777CCD86FAE2D65/preproc/ctags_target_for_gcc_minus_e.cpp"
"/usr/bin//ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/tmp/arduino-sketch-F65E51C7CC67F8F1D777CCD86FAE2D65/preproc/ctags_target_for_gcc_minus_e.cpp"
"/usr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10600 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/root/workspace/Arduino/hardware/arduino/avr/cores/arduino" "-I/root/workspace/Arduino/hardware/arduino/avr/variants/mega" "/tmp/arduino-sketch-F65E51C7CC67F8F1D777CCD86FAE2D65/sketch/test.ino.cpp" -o "/tmp/arduino-sketch-F65E51C7CC67F8F1D777CCD86FAE2D65/sketch/test.ino.cpp.o"
/root/workspace/test/test.ino:1:9: error: expected constructor, destructor, or type conversion before ';' token
 void setup()
         ^
/root/workspace/test/test.ino:6:8: error: expected constructor, destructor, or type conversion before ';' token
 void loop()
        ^
exit status 1

```

ffissore, 2015-12-02:
I dag well into the code @spiderkeys.
The recipes you miss are stored in separate folders of the IDE: `tools-builder` and `hardware`.
Since you hacking, you may want to start with the [latest release](https://github.com/arduino/arduino-builder/releases) which bundles the minimum tools and platform*.txt stuff

That said, stock ctags will not work, because it does not output functions return type. We compile and bundle a modified version, which you can [find here](https://github.com/arduino/ctags). You'll have to clone and compile it on your own, since we don't provide ARM binaries

facchinm, 2015-12-02:
Hi @spiderkeys 

One topic at a time :smile: 
* No need to add `mega.build.mcu=atmega2560` to boards.txt, simply call the builder with ` -fqbn=arduino:avr:mega:cpu=atmega2560`
* as you noticed, `ctags` is needed. There is no need to modify the files since the builder CLI is powerful enough to resolve a lot of combinations. To have a completely functional environment on the BBB you can use this script and adapt to your environment
```bash
cd $HOME
wget http://downloads.arduino.cc/arduino-builder-arm/arduino-builder
chmod +x  arduino-builder
mkdir -p hardware/tools/
wget http://downloads.arduino.cc/arduino-builder-arm/ctags
chmod +x ctags
mv ctags hardware/tools/
wget http://downloads.arduino.cc/cores/avr-1.6.9.tar.bz2
mkdir -p hardware/arduino/
tar xvf avr-1.6.9.tar.bz2
mv avr hardware/arduino/
wget https://raw.githubusercontent.com/arduino/arduino-builder/master/src/arduino.cc/builder/hardware/platform.keys.rewrite.txt
mv platform.keys.rewrite.txt hardware/
wget https://raw.githubusercontent.com/arduino/arduino-builder/master/src/arduino.cc/builder/hardware/platform.txt
mv platform.txt hardware/
```
and compile with 
```
arduino-builder -verbose -debug-level=10 -compile -hardware /root/hardware/ -tools /usr/bin/ -fqbn arduino:avr:mega:cpu=atmega2560 $sketch_name.ino
```

You need to have `avr-gcc` installed system-wide using your distro package manager
Let me know if it works!

facchinm, 2015-12-02:
@ffissore I had arm binaries for `arduino-builder` and `ctags` wandering in my box and I shared them :wink: 
@spiderkeys still no official release though, but feel free to grab it, report your results and share the modified procedure for BBB!

ffissore, 2015-12-02:
@facchinm good to know!

ShorTie8, 2015-12-02:
Thankz @facchinm  for the binaries to play with .. :)~
Any chance to get the arduino-builder-arm source or diff ??

facchinm, 2015-12-02:
The sources are exactly the master branches of http://github.com/arduino/arduino-builder and http://github.com/arduino/ctags compiled for armv7 :wink:

spiderkeys, 2015-12-02:
Thanks for the help everyone. @facchinm following your guide, I was able to almost get everything working in one shot. Unfortunately, the tools didn't automatically pick up the directories for the tools so here are the changes I had to make:

- In hardware/platform.txt, I had to specifically set the path of the ctags executable, as I got the error: *os.PathError fork/exec {runtime.tools.ctags.path}/ctags: no such file or directory

```
tools.ctags.path=/root/workspace/armbuilder/hardware/tools/
```

- In hardware/arduino/avr/platform.txt, I had to explicitly set the path for avr-gcc to /usr/bin/, as it was trying to find avr-g++ in /bin/ (I believe the {runtime.tools.avr-gcc.path} variable was evaluating to nothing).

```
compiler.path=/usr/bin/
```

Then for my build command:

```
./arduino-builder -verbose -debug-level=10 -compile -build-path /root/workspace/build/ -hardware /root/workspace/hardware/ -tools /usr/bin/ -libraries /root/workspace/Arduino/libraries/ -fqbn arduino:avr:mega:cpu=atmega2560 ./source/test.ino
```

Once I did that, I was able to build my simple test.ino. Any ideas why I still need to provide absolute paths to tools in my platform.txt files? It's not a big deal, since our use environment is pretty contained, just curious.

I decided to push my luck a bit and tried building our entire firmware stack for our robot: https://github.com/OpenROV/openrov-software-arduino/tree/30.0.3/OpenROV. Everything seemed to work, and I got my hex file as expected.

Now, to get to the real aim of this endeavour. We are developing a new controller board for our robot using the J18A variant of the Atmel SAMD21 32bit arm as our MCU. The core supplied by Arduino is for the SAMD21G, so that doesn't work right out of the box. Luckily the fine people at Sodaq made a custom core for the J18A which can be found here: https://github.com/SodaqMoja/SodaqCore-samd

Do you have any recommendations for what steps I will need to take to get arduino-builder working using a sam core package? I imagine if I could even get the Arduino provided sam package working, the same steps would carry over to Sodaq's core.

I've already taken the steps of cloning their package to hardware/arduino/sam/ and installing the arm-none-eabi-gcc toolchain. I notice in the sam platform.txt file that there is a {runtime.tools.CMSIS.path} referenced, so it's worth asking if there is any Arduino specific version of it required, much like ctags.

Again, thanks for all of the help.

facchinm, 2015-12-03:
Hi @spiderkeys ,
sorry for the need to change the files, in my env it works out of the box but path resolution is always a nightmare if the package is not completely self contained. 
About the extra tools, you can grab the correct ones from [this](http://downloads.arduino.cc/packages/package_index.json) json file, just search for samd dependencies and you'll get the links
One last thing, the builder accepts multiple `-hardware`, `-tools` and `-libraries` flags, so you can extract the CMSIS package where you prefer an then add a `-tools` with its path (yes, I really hate changing the config files :smile: )

spiderkeys, 2015-12-03:
Progress! I was able to build the same test.ino using both the SAMD21G and Sodaq SAMD21J18A cores. Still some more manual path specification pain points, but luckily only a few. Here are the steps I took beyond the last steps I provided getting things built for the ATMega2560:

- Installed gcc-arm-none-eabi with apt-get (current version in Debian repository was 4.8.4)

- Downloaded CMSIS folder from arduino at: http://downloads.arduino.cc/CMSIS-4.0.0.tar.bz2
- Extracted to ./hardware/tools/

- Downloaded SAMD 1.6.2 core (SAMD21G) from: http://downloads.arduino.cc/cores/samd-1.6.2.tar.bz2
- Extracted to ./hardware/arduino/

- In /hardware/arduino/samd/platform.txt, had to make a few manual path specifications:
	- compiler.path=/usr/bin/
	- compiler.arm.cmsis.path="-I/root/workspace/hardware/tools/CMSIS/CMSIS/Include/" "-I/root/workspace/hardware/tools/CMSIS/Device/ATMEL/"

For building with the Sodaq Core, I simply had to clone their core package to hardware/arduino/sodaq/ and make the same changes as above to the platform.txt file, with the added step of undoing a change that they had made for some reason:

https://github.com/SodaqMoja/SodaqCore-samd/commit/2639679b9b66f61421fd6c6ceada654657825878

- Re-added "{build.path}/" to {archive_file} portion of archiver pattern

My build command:

```
./arduino-builder -verbose -debug-level=10 -compile -build-path /root/workspace/build/ -hardware /root/workspace/hardware/ -tools /usr/bin/ -libraries /root/workspace/Arduino/libraries/ -fqbn arduino:samd:arduino_zero_native ./source/test.ino
```

One thing to note is that I originally tried to use the option "-tools /root/workspace/hardware/tools/CMSIS/" and "-tools /root/workspace/hardware/tools/CMSIS/CMSIS/" to get arduino-builder to correctly find the path variable value for it, but neither worked, resulting in the hardcoded path in platforms.txt.

@facchinm I'm curious, could you describe your environment that you tested this in a bit more in terms of where and how you installed your packages, directory structure with relation to the arduino-builder executable, your hardware/ folder, etc? I'm wondering if my issue with needing to manually modify the platform.txt files is a result of things not being perfectly named or placed as expected by arduino-builder. On that note, is there any insight into how the tool is actually searching for and populating those environment variables, particularly with respect to when you supply the -hardware and -tools options? It seems like it is ultimately finding the executables themselves from the options I provide, but is unable to set the runtime.path.x.path variables.

Once I've cleaned up the process a bit and written some a script to automate most of this, I'll link it here. Hopefully I can figure out how to negate the need for any modification of platforms.txt to make that script a bit nicer.

Thanks and cheers

NicoHood, 2015-12-04:
Could you please explain me how you build the arduino-builder for arm?
I tried on arm and x64 linux and had no luck yet.

On x64 I get an error here:
```
Arduino/arduino-builder$ go get golang.org/x/codereview/patch
go: missing Mercurial command. See http://golang.org/s/gogetcmd
package golang.org/x/codereview/patch: exec: "hg": executable file not found in $PATH

Arduino/arduino-builder$ go get golang.org/x/tools/cmd/vet
# cd .; git clone https://go.googlesource.com/tools /home/linuxuser/Documents/Arduino/arduino-builder/src/golang.org/x/tools
Cloning into '/Arduino/arduino-builder/src/golang.org/x/tools'...
fatal: unable to access 'https://go.googlesource.com/tools/': Could not resolve host: go.googlesource.com
package golang.org/x/tools/cmd/vet: exit status 128
```

On arm I only have the apt-get 1.3.3 package and no chance to recompile the 1.5.2 source myself.

If this is solved, we can possibly enable the IDE for arm, beaglebone included:
https://github.com/arduino/Arduino/pull/3549

spiderkeys, 2016-02-04:
Another "out of the box" ARM issue: Tests fail due to "Unknown OS: linux-arm"

My test output XML:
[report.txt](https://github.com/arduino/arduino-builder/files/118150/report.txt)

cmaglie, 2016-02-05:
@spiderkeys a lot of efforts has been put into this recently, please have a look at this branch:

https://github.com/arduino/arduino-builder/compare/arm?expand=1

you can see that we use http://downloads.arduino.cc/packages/test_package_arm_index.json as the url for downloading tools, this is a small hack to run tests because we are still testing the tools for arm. Once tested the tools will be published in the main package_index.json and the hack removed.

See also:
https://github.com/arduino/Arduino/pull/4517
https://github.com/arduino/Arduino/pull/4457
https://github.com/arduino/arduino-builder/issues/105

kgranat, 2016-03-11:
Hello,
I hope this is the right place to ask this - I'm working in a Raspberry Pi 3 (ARM7), I've got Go and Arduino builder built from source. I also built ctags from source as @ffissore recommended (after I was unable to downloads @facchinm 's Arm binary. Here's my command

`sudo ./arduino-builder -hardware="/home/pi/go/arduino-builder/hardware/"  -tools="/usr/bin/" -fqbn=arduino:avr:uno B-blinkBasic.ino`

and my output.

`/tmp/arduino-sketch-6367DD41E1AFABE232F53944DD4B02C3/sketch/ctags/argproc.c:24:21: fatal error: descrip.h: No such file or directory
 #include <descrip.h>
                     ^
compilation terminated.
exit status 1`

Any ideas? I'm guessing that I did something dumb while building ctags or something of the like.

cmaglie, 2016-04-14:
There is now an experimental hourly build of the Arduino IDE for ARM:

https://www.arduino.cc/en/Main/Software

closed 2016-04-14 by cmaglie