arduino/arduino-builder #78: Needs "help" functions, such as "-list-boards"

mralexgray, 2015-12-04:
Great work on this essential, much needed tool.  So far, it is a bit unfriendly.  A smart person should theoretically be able to use this without scouring the web.  There does not seem to be a `man` page, and entering `--help` nets the same result as having entered nothing.. or nonsense, lol.

The output should stick to convention, provide help, and utility options...

    Usage: 
         arduino_builder -hardware ... -tools ... -fqbn ...  [options]
         arduino_builder -list-boards [-hardware ...] [-tools ...]  # lists available board, per the current config
         arduino_builder -help # show this text
         etc....

Please refer [Wikipedia's "Usage_message"](https://en.wikipedia.org/wiki/Usage_message), if I am at all unclear.

You'll notice I mentioned a non-existent parameter, `-list-boards`.  This type of option should be able to run, with our without "options", to *help the user when formulating their _own_ command-line*.. so they don't have to resort to searching the web.. or worse yet.. your source code, like I did. (I don't know what my board's FQBN is! And I'm sure I am not the only one.)

`--dump-prefs` is another example of a command trhat should be allowed to be run "on its own", independent of a compile, or what not.

I could go on... and will, if there is a willingness to proceed.

ffissore, 2015-12-09:
arduino-builder knows nothing about where you put your boards and libraries. Therefore, running with your proposed `--list-boards` without any additional argument will print nothing. You would have to provide at least one `--hardware` param.
If that would still make sense to you, I can easily code such `--list-boards`

mralexgray, 2015-12-20:
that would be great.. also, one complete, valid usage example would be nice in the readme/onscreen help.

vshymanskyy, 2016-11-19:
so.. no ```--list-boards``` yet? ;)

roncapat, 2017-01-04:
I'm probably going to submit a pull request for something like:
```
Usage: 
     arduino_builder -hardware ... -tools ... -fqbn ...  [options]
     arduino_builder -help # show this text
```
--list-boards, as pointed out by @ffissore, is a more complex flag to add, which depends on the context, and is not directly related to the need of a better "Usage header". That could be done in future by someone. 
Suggestions on the usage intro, as quoted above?