arduino/arduino-builder #79: ARM: fork/exec {runtime.tools.ctags.path}/ctags: no such file or directory

NicoHood, 2015-12-05:
I think I will just refer to the ARM build PR this time, because I might edit it over and over again. The error is described in the section "Fix Ctags":
https://github.com/arduino/Arduino/pull/3549

It does not happen under x64 linux:
```bash
 1321  git clone https://github.com/arduino/arduino-builder.git
 1322  cd arduino-builder
 1323  sudo apt-get install golang
 1324  export GOPATH=`pwd`
 1325  go get github.com/go-errors/errors
 1326  go get github.com/stretchr/testify
 1327  go get github.com/jstemmer/go-junit-report
 1328  go get golang.org/x/codereview/patch
 1329  sudo apt-get install mercurial
 1330  go get golang.org/x/codereview/patch
 1331  go get golang.org/x/tools/cmd/vet
 1332  go build
 1333  ls
 1334  cp ./arduino-builder ../linux/work/
 1335  ../linux/work/arduino-builder --version
```

ffissore, 2015-12-07:
ctags must be a tool in order to work with arduino builder: i.e. it must be located inside a `FOLDER/ctags/VERSION` folder structure, and `FOLDER` must be provided to arduino-builder `--tools` param

NicoHood, 2015-12-07:
```
/home/pi/Documents/Arduino/arduino-1.6.7/arduino-builder -dump-prefs -logger=machine -hardware "/home/pi/Documents/Arduino/arduino-1.6.7/hardware" -tools "/home/pi/Documents/Arduino/arduino-1.6.7/tools-builder" -tools "/home/pi/Documents/Arduino/arduino-1.6.7/hardware/tools/avr" -built-in-libraries "/home/pi/Documents/Arduino/arduino-1.6.7/libraries" -libraries "/home/pi/Arduino/libraries" -fqbn=arduino:avr:leonardo -ide-version=10607 -build-path "/tmp/buildb4c70589a36ab7f45a3478b556ec2c13.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/tmp/untitled928648742.tmp/sketch_dec07a/sketch_dec07a.ino"
/home/pi/Documents/Arduino/arduino-1.6.7/arduino-builder -compile -logger=machine -hardware "/home/pi/Documents/Arduino/arduino-1.6.7/hardware" -tools "/home/pi/Documents/Arduino/arduino-1.6.7/tools-builder" -tools "/home/pi/Documents/Arduino/arduino-1.6.7/hardware/tools/avr" -built-in-libraries "/home/pi/Documents/Arduino/arduino-1.6.7/libraries" -libraries "/home/pi/Arduino/libraries" -fqbn=arduino:avr:leonardo -ide-version=10607 -build-path "/tmp/buildb4c70589a36ab7f45a3478b556ec2c13.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/tmp/untitled928648742.tmp/sketch_dec07a/sketch_dec07a.ino"
```

```
pi@raspberrypi ~/Documents/Arduino/arduino-1.6.7 $ ls
arduino          examples    lib        revisions.txt  uninstall.sh
arduino-builder  hardware    libraries  tools
dist             install.sh  reference  tools-builder
pi@raspberrypi ~/Documents/Arduino/arduino-1.6.7 $ ls tools-builder/
5.8-arduino5
```

Is there anything incorrect?
I did not wrote those params myself, this did the recompiled IDE itself.

ffissore, 2015-12-07:
What if you run the first command and add `| grep runtime.tools.ctags` ?

NicoHood, 2015-12-07:
```
cat log1.txt | grep runtime.tools.ctags
tools.ctags.path={runtime.tools.ctags.path}
```

Those logs were attached to the post, if you missed it:
https://gist.githubusercontent.com/NicoHood/18ee8c4567927bf8857c/raw/c9024100c866d75700cacf06b2b76ef033fc3114/log1.txt

ffissore, 2015-12-07:
I don't understand. The tools are there but the builder can't find them

NicoHood, 2015-12-07:
Exactly. I've uploaded all binaries for raspi, so you can try yourself if you like to. I have no idea what is happening there. It is the very last bug that keeps us from releasing an ARM build (except that arduino.cc need to accept arm builds).

NicoHood, 2016-01-20:
This bug seems to be caused because the arduino-builder **arm** package is packed wrong.
If you move the files into the right folder, it all compiles fine and arm builds are extremely simple to add now!

```bash
wget downloads.arduino.cc/tools/arduino-builder-arm-1.3.9.tar.bz2
tar -jtvf arduino-builder-arm-1.3.9.tar.bz2

drwxr-xr-x jenkins/jenkins   0 2015-12-16 15:53 ./
drwxr-xr-x jenkins/jenkins   0 2015-12-16 15:53 ./tools/
-rwxr-xr-x jenkins/jenkins 746649 2015-12-16 15:53 ./tools/ctags
drwxr-xr-x jenkins/jenkins      0 2015-12-16 15:53 ./hardware/
-rw-r--r-- jenkins/jenkins   1481 2015-12-16 15:53 ./hardware/platform.txt
-rw-r--r-- jenkins/jenkins   7194 2015-12-16 15:53 ./hardware/platform.keys.rewrite.txt
-rwxr-xr-x jenkins/jenkins 3311192 2015-12-16 15:53 ./arduino-builder

wget downloads.arduino.cc/tools/arduino-builder-linux64-1.3.9.tar.bz2
tar -jtvf arduino-builder-linux64-1.3.9.tar.bz2

drwxr-xr-x jenkins/jenkins   0 2015-12-15 11:37 ./
drwxr-xr-x jenkins/jenkins   0 2015-12-15 11:37 ./tools/
drwxr-xr-x jenkins/jenkins   0 2015-12-15 11:37 ./tools/ctags/
drwxr-xr-x jenkins/jenkins   0 2015-12-15 11:37 ./tools/ctags/5.8-arduino5/
-rwxr-xr-x jenkins/jenkins 267026 2015-12-15 11:37 ./tools/ctags/5.8-arduino5/ctags
drwxr-xr-x jenkins/jenkins      0 2015-12-15 11:37 ./hardware/
-rw-r--r-- jenkins/jenkins   1481 2015-12-15 11:37 ./hardware/platform.txt
-rw-r--r-- jenkins/jenkins   7194 2015-12-15 11:37 ./hardware/platform.keys.rewrite.txt
-rwxr-xr-x jenkins/jenkins 4088088 2015-12-15 11:37 ./arduino-builder
```

NicoHood, 2016-01-21:
You should also include the arm builds in the releases page:
https://github.com/arduino/arduino-builder/releases/tag/1.3.10

facchinm, 2016-05-10:
Should be ok now since the builder is built natively also on ARM so the paths are exactly as in linux64/linux32 builds

closed 2016-05-10 by facchinm