arduino/arduino-builder #83: builder is unable to find libraries for 3rd party boards

robertinant, 2015-12-14:
Builder (1.3.8) in Arduino 1.6.7 is no longer able to find libraries (e.g. SPI.h) for the panStamp NRG board. http://panstamp.org/arduino/package_panstamp_index.json. This issue might also exist for other boards. This was working fine with builder 1.0.5 in Arduino 1.6.6. I have looked at all the configuration files and do not see anything obvious that should be changed to support the latest builder. Can you please look into this?

facchinm, 2015-12-14:
Hi @robertinant ,
I've tested the core you are referring to and I only found a problem with HardwareSerial (patch this in `Arduino.h`)
```patch
 #include "WCharacter.h"
 #include "WString.h"
+#include "HardwareSerial.h"
 #if defined(HAVE_HWSERIAL0) && defined(HAVE_CDCSERIAL)
 #error "Targets with both UART0 and CDC serial not supported"
 #endif
``` 

After fixing this I tested `sdCardTest` and it compiled correctly.
Can you give more hints (core version, sketch code...)
Thanks!

facchinm, 2015-12-14:
This is how the libraries get resolved on my setup
```
Using library SPI at version 1.0 in folder: /home/martino/.arduino15/packages/panstamp_avr/hardware/avr/1.5.3/libraries/SPI 
Using library SD at version 1.0.6 in folder: /home/martino/Arduino-master/build/linux/work/libraries/SD
```

robertinant, 2015-12-14:
Sorry, I should been more specific. The issue is with the panStamp NRG boards. All is fine for the panStamp AVR boards.

facchinm, 2015-12-15:
Hi @robertinant ,
you were right indeed, I was testing the wrong board. 
The issue should be fixed with bd5d3eca0132ad590e561991c8ce680c5f5af67e and become available in the next hourly build of the IDE.
Thanks for the report!

robertinant, 2015-12-15:
Awesome! That fixed it. Thanks for the quick turn-around on this!

closed 2015-12-15 by robertinant

dberenguer, 2016-04-17:
This issue was fixed for core libraries but not for 3rd-party generic libraries (installed under sketches/libraries/) being used by 3rd-party boards.

This was reported in issue #97 some months ago. IDE 1.6.8 is still showing the problem.