arduino/arduino-builder #85: Generated prototypes are added after global variable declarations

per1234, 2015-12-29:
Using Arduino IDE 1.6.8 2015/12/28 12:34 with Windows 7 64 bit
Generated prototypes are added after global variable declarations which causes error when the global variable initialization calls a function.
```arduino
byte x = one();

byte one() {
  return 1;
}

void setup() {}
void loop() {}
```
Compiler error: `'one' was not declared in this scope`

Generated code:
```arduino
#include <Arduino.h>
#line 1
#line 1 "C:\\Users\\per\\AppData\\Local\\Temp\\arduino_ada7ab83d38e9f542377ce1f8da0ce54\\sketch_dec29a.ino"
byte x = one();

#line 3 "C:\\Users\\per\\AppData\\Local\\Temp\\arduino_ada7ab83d38e9f542377ce1f8da0ce54\\sketch_dec29a.ino"
byte one();
#line 7 "C:\\Users\\per\\AppData\\Local\\Temp\\arduino_ada7ab83d38e9f542377ce1f8da0ce54\\sketch_dec29a.ino"
void setup();
#line 8 "C:\\Users\\per\\AppData\\Local\\Temp\\arduino_ada7ab83d38e9f542377ce1f8da0ce54\\sketch_dec29a.ino"
void loop();
#line 3
byte one() {
  return 1;
}

void setup() {}
void loop() {}
```
This compiles fine in Arduino IDE 1.6.5r5 and works in 1.6.6 and up if the function prototype is manually added.

Originally reported at http://forum.arduino.cc/index.php?topic=366573.msg2540168#msg2540168

matthijskooijman, 2015-12-29:
The prototypes are generated just before the first function definition. I believe the idea is that this helps to put the prototypes *after* any type definitions, but in your example it also puts them after the first reference, breaking them. For now, the fix is to add a prototype manually. To really properly fix this, arduino-builder should actually parse the source and construct a dependency tree and use that to figure out where to put each prototype, but the current ctags-based approach doesn't allow that. I believe @cmaglie was looking into using clang instead of ctags, but even then the dependency generation and resolution is still a non-trivial problem...

per1234, 2015-12-29:
Understood, thanks. I wasn't sure if this was intentional or unavoidable but thought it was worth at least to document the change from 1.6.5. I have already suggested manually adding the prototype as a workaround to the reporting user

facchinm, 2017-03-17:
Will be solved by merging https://github.com/arduino/arduino-builder/pull/191