arduino/arduino-builder #88: Arduino 1.6.7 fails to correctly extract function prototypes from .ino, requires forward declarations

matthijskooijman, 2015-12-29:
_From @ratkins on December 28, 2015 20:17_

My project at [this commit](https://bitbucket.org/ratkins/technocolourdreamcoat2016controllertest/commits/7ba31482d36ee26b7c02cfeffcf6a823da6d270a) successfully compiled and built under Arduino 1.6.5-betasomething, with Teensyduino 1.26. It fails to build under 1.6.7/1.27 without forward declaring functions that are defined in the .ino (below `loop()`, from which they're called.)

[This commit](https://bitbucket.org/ratkins/technocolourdreamcoat2016controllertest/commits/90ffcf10689d2199fc25e94f18a78022a25fb71a) contains changes necessary for it to compile.

_Copied from original issue: arduino/Arduino#4359_

matthijskooijman, 2015-12-29:
_From @sebastius on December 29, 2015 9:23_

Ah so that's what it is. Been messing for hours...

I'm downgrading to 1.6.5. Such a shame.

matthijskooijman, 2015-12-29:
I just tried your sketch on a recent git version, and (after installing the right libraries) it compiles just fine. Perhaps this problem has been fixed in git already. Could you try the latest hourly build from arduino.cc and see if that also works for you?

GimpMaster, 2015-12-31:
I see the same problem but it is ONLY when you have teensy selected as your board. If you try a different board like the UNO you don't have the issue. This is 1.6.7.

julianh2o, 2016-01-06:
This seems to still be an issue in the "hourly" build as downloaded a few minutes ago.

I'm on OS X and compiling for the "Generic ESP8266 Module". I've seen the same problem in 1.6.6, 1.6.7, and the hourly build.

odilonafonso, 2016-01-18:
was this issue resolved ? I cant compile my sketch with arduino 1.6.7

matthijskooijman, 2016-01-18:
Can someone paste the full error output, and preferably also the preprocessed source? To find the latter, enable verbose compilation and look for a line similar to this one:

> "/home/matthijs/docs/Electronics/Arduino/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/matthijs/docs/Electronics/Arduino/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/home/matthijs/docs/Electronics/Arduino/Arduino/build/linux/work/hardware/arduino/avr/variants/standard" "**/tmp/arduino_build_718470/sketch/sketch_jan18a.ino.cpp**" -o "/tmp/arduino_build_718470/sketch/sketch_jan18a.ino.cpp.o"

The (highlighted) .ino.cpp file is the file I mean.

Even better would be to reduce the sketch to a small single file sketch, not depending on any libraries, that also causes this error, but that might not be easy or at all possible.

Also, there is a fair chance that multiple commenters are having a different problem with similar symptoms, so we shouldn't assume this is just one problem.

@PaulStoffregen, you might be interested in this, since it seems to affect Teensy (though also ESP, so it's probably not Teensy-specific).

odilonafonso, 2016-01-18:
Hi,

Attached ino file and error log.

The error is when include the ESP8266WiFi.h file:

#include <ESP8266WiFi.h>

If not, the compiler goes ok...

I reproduce this error from my application sketch that run "real"
functions, not fake like the attached ones..

Apparently the error happens when you included the header file "
ESP8266WiFi.h" but why is that? Where is the mistake?

*Odilon Cenamo*
*Vivo (11)95636-1956*

*Skype (11)3042-1956Nextel (11)7720-4140 Id 30*69343*

matthijskooijman, 2016-01-19:
@odilonafonso This was the file I was looking for: /tmp/build4e19292b391286a706dcd650275fe0b9.tmp/sketch/testFun.ino.cpp (if you restart the IDE, the build path might have changed, though).

The sketch itself looks good to me, no weirdness in there.

It looks like for the ESP core, arduino-builder does not run ctags to collect function prototypes. I'm not sure why, looking at the arduino-builder source, I can't find any condition that would cause ctags to be skipped without an error message.

matthijskooijman, 2016-01-19:
I just installed the ESP core and was able to reproduce this, so I should be able to figure this out from here :-)

matthijskooijman, 2016-01-19:
> It looks like for the ESP core, arduino-builder does not run ctags to collect function prototypes.

W00ps, I wasn't looking properly, it *does* run ctags, but somehow does not generate prototypes. I suspect there is something in the ESP8266WiFi.h that confuses ctags, looking closer now.

odilonafonso, 2016-01-19:
Hi matthijskooijman,
you still need the cpp file? I can generate one. 
But i think the problem is with the included file. I don't know why...
I put this include file in the sample sketch for causing the error.
But my application needs to include this file. I had to put a prototype of all the functions that I use in my sketch
Thanks for help

matthijskooijman, 2016-01-19:
I've found the cause of the problem: The static_assert keyword, which is
used by the ESP / libstdc++ include files, is confusing ctags and
prevents it from emitting information about subsequent functions. I've
filed a report here: https://sourceforge.net/p/ctags/bugs/370/ (though
exuberant ctags doesn't seem to be really maintained anymore, so this
report serves more as documentation than that I really expect it to be
fixed).

The proper fix would be to add support for static_assert in the Arduino
ctags version, but I'm afraid that might not be so easy. I won't have
time for looking at that on the short term.

A workaround that you can use now is to re-order your functions (so they
are defined before they are used), or add function prototypes manually
in the .ino file).

@odilonafonso, I don't need anything else from you, thanks!
@ratkins, @GimpMaster, could you get the `/tmp/buildnnn/preproc/ctags_target_for_gcc_minus_e.cpp` file from a failed build and attach it here? I want to confirm that you are seeing the same issue, or perhaps something else with similar symptoms. Note that the `/tmp/buildnnn` path is different per platform and per build, so look around the verbose build output to find that part of the path.

odilonafonso, 2016-01-19:
Cool Matthijs !!!
Maybe it's not hard to fix, in release 1.6.5 this problem does not happen.
Finding the problem is a big step!
Grateful for the help.

matthijskooijman, 2016-01-19:
@odilonafonso IIRC 1.6.5 used a different way for detecting prototypes (not using arduino-builder and ctags), which had a lot of other problems, so it's not some change we can revert back to the 1.6.5 version, unfortunately.

per1234, 2016-01-20:
I think the ESP8266WiFi.h issue is a duplicate of https://github.com/arduino/arduino-builder/issues/68

matthijskooijman, 2016-01-20:
@per1234 good call, seems I totally missed that issue somehow. I've forwarded relevant information there, so I'm now closing this one as a duplicate.

closed 2016-01-20 by matthijskooijman