arduino/arduino-builder #90: [WIN] arduino-builder doesn't walk a path with directory junctions

facchinm, 2016-01-08:
from thewknd  - https://github.com/arduino/Arduino/issues/4402

> With Arduino 1.6.7 on Windows 7 I've been unable to verify/upload when the folder for the custom libraries (*Documents\Arduino\libraries*) contains a directory junction. (I use directory junctions to switch between different library versions)

> Error message:

```
C:\arduino167\arduino-builder -dump-prefs -logger=machine -hardware "C:\arduino167\hardware" -tools "C:\arduino167\tools-builder" -tools "C:\arduino167\hardware\tools\avr" -built-in-libraries "C:\arduino167\libraries" -libraries "C:\Users\thewknd\Documents\Arduino\libraries" -fqbn=arduino:avr:uno -ide-version=10607 -build-path "C:\Users\thewknd\AppData\Local\Temp\buildf861fa3b4440ac4767ab26e18e654c6c.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\thewknd\AppData\Local\Temp\untitled175997220.tmp\sketch_jan08a\sketch_jan08a.ino"
    readlink C:\Users\thewknd\Documents\Arduino\libraries\test: The system cannot find the file specified.

   Error compiling.
```
> test is the directory junction, created with

```
MKLINK /J "C:\Users\thewknd\Documents\Arduino\libraries\test" "C:\Users\thewknd\Documents\test" 
```
 
> With Arduino 1.6.5r5 everything was fine (didn't use 1.6.6)

facchinm, 2016-01-08:
The problem arises because the `Filepath.Walk` golang function does not traverse links. This has been resolved for UNIX systems but not for Win

facchinm, 2016-01-19:
Duplicate of arduino/arduino-builder#90

facchinm, 2016-02-05:
This issue can be solved by patching go sources with https://github.com/golang/go/commit/571738569e9b6bb30ebceaf2eb5936f6ee254e34
Tested on Win7 VM, works correctly if the junction has the same name of the origin
@cmaglie

facchinm, 2016-03-11:
Glad it worked. Closing as the main problem is solved, and the side problem will be solved by a golang update hopefully :smile:

rcarmo, 2017-10-22:
I'm having this issue with IDE 1.8.5 on Windows 10 Fall Creators Update. I've tried both the standalone installer and the Windows Store app, and I can't compile anything that lives inside OneDrive. Moving the sketch folder to the root of the C: drive magically fixes the problem, and it took me quite a while to figure out that junctions were involved.

matthijskooijman, 2017-11-16:
I wanted to reopen this issue, but decided to open a new issue (#254) instead.

closed 2017-11-16 by matthijskooijman