arduino/arduino-builder #97: Builder is unable to find generic libraries for 3rd party boards

dberenguer, 2016-01-18:
Issue https://github.com/arduino/arduino-builder/issues/83 is not totally fixed. Using platform-generic libraries (under SKETCHES/libraries/) from 3rd-party boards does not work. Libraries cannot be found by the builder.

I tested the SWAP library:
https://github.com/panStamp/swap

with this example:
https://github.com/panStamp/swap/tree/master/examples/binouts

for panStamp NRG2 board:
http://panstamp.org/arduino/package_panstamp_index.json

dberenguer, 2016-04-17:
Any on-going work regarding this issue? 1.6.8 is still showing the problem.

We're using IDE 1.6.5 in the meantime.

matthijskooijman, 2016-05-18:
Turns out the fix that was applied only worked for includes directly from the main .ino file. When scanning the compiler error output for missing includes, it only looked at the first line. However, when the .ino file includes a .h file in the sketch, which again includes a .h file from a library, the output looks like:
```
In file included from /tmp/arduino_modified_sketch_815412/binouts.ino:52:0:
/tmp/arduino_build_static/sketch/regtable.h:31:22: fatal error: register.h: No such file or directory
```
So the error in the first line is not found. The fix is of course simple: Scan all lines instead of just the first. I'll prepare a PR in a minute.

dberenguer, 2016-05-19:
Thanks @matthijskooijman!

Looking forward to testing the fix on the next release.

PR #152 (2016-05-23): Improve include resolution fix for older compilers
The primary way to detect includes is to scan the compiler error output
for "#include" lines. However, older gcc versions did not print the
error line itself, only a message about the error, which would not be
detected. In bd5d3ec (Fix include resolution for older compilers), a
fallback was added to scan for "fatal error" followed by the missing
include filename, but that fallback only looked at the first line of the
error output.

When the missing #include line is not in the source file being compiled,
but in an included file, the error messa [... cut]
commit 5a79106e (2016-05-23): Improve include resolution fix for older compilers

The primary way to detect includes is to scan the compiler error output
for "#include" lines. However, older gcc versions did not print the
error line itself, only a message about the error, which would not be
detected. In bd5d3ec (Fix include resolution for older compilers), a
fallback was added to scan for "fatal error" followed by the missing
include filename, but that fallback only looked at the first line of the
error output.

When the missing #include line is not in the source file being compiled,
but in an included file, the error mess [... cut]
closed 2016-05-23 by cmaglie