arduino/arduino-builder #106: `-mmcu=` not passed while generating dependencies

matthijskooijman, 2016-02-02:
I haven't thoroughly verified this, but https://github.com/arduino/Arduino/issues/4512#issuecomment-178780444 seems to indicate that, while running the preprocessor to collect dependencies, the `-mmcu=` option is not passed to gcc. You would say that this is not needed for preprocessing, but this also prevents defines like `__AVR_ATmega328p__` from being present, which could be used to determine what libraries to use (as happens in [this library](https://github.com/digistump/DigistumpArduino/blob/master/digistump-avr/libraries/DigisparkLCD/LiquidCrystal_I2C.h)).

matthijskooijman, 2016-02-02:
@cmaglie, I think this is something to fix for the next release.

cmaglie, 2016-02-03:
There is a smaller sketch that we can use to reproduce the problem? (sorry I didn't read the full thread yet...)

matthijskooijman, 2016-02-03:
I had a closer look, and it seems the problem is that [digistup does not define](https://github.com/digistump/DigistumpArduino/blob/master/digistump-avr/platform.txt) the [recipe.preproc.macros](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/platform.txt#L88-L89) value. arduino-builder has [a default value for it](https://github.com/arduino/arduino-builder/blob/master/src/arduino.cc/builder/hardware/platform.txt#L16), but that does not seem to include -mmcu (at least not when combined with the digistump platform.txt.

Looking through the history, the arduino-builder version never had the -mmcu option, and the Arduino version always had it, so I'm not sure what happened there.

I was going to suggest to remove the recipe.preproc.macros entry from Arduino's platform.txt and use the below sketch, but that doesn't seem to break it somehow (`-mmcu` is still passed). Not sure if that's a problem on my setup (possibly it's using a different platform.txt than I think it is), but no more time to test now. So, here's the sketch:

```
#if defined(__AVR__)
#include<Wire.h>
#endif

void setup() {
    Wire.begin();
}

void loop() {
}
```

per1234, 2016-02-28:
I think that test sketch doesn't cause the issue because `__AVR__` is defined by `-DARDUINO_ARCH_{build.arch}` which is in the arduino-builder default [`recipe.preproc.macros`](https://github.com/arduino/arduino-builder/blob/master/src/arduino.cc/builder/hardware/platform.txt#L16).

- Comment [`recipe.preproc.macros`](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/platform.txt#L89) in the Arduino AVR Boards platform.txt.
- Tools > Board > Arduino/Genuino Uno
- Compile:
```arduino
#ifndef __AVR_ATmega328P__
#error
#endif
void setup() {}
void loop() {}
```
Arduino IDE 1.6.8 2016/02/24 06:33:
```
sketch_feb28a:2: error: #error 

 #error

  ^

exit status 1
#error 
```
Note that the arduino-builder default [`recipe.preproc.includes`](https://github.com/arduino/arduino-builder/blob/master/src/arduino.cc/builder/hardware/platform.txt#L12) is also missing `-mmcu={build.mcu}` which is in Arduino AVR Boards 1.6.9 [platform.txt](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/platform.txt#L86) but I haven't found a case where that causes a problem.

matthijskooijman, 2016-04-26:
I finally managed to find why I could not reproduce this issue before: I was changing the platform.txt in my git checkout, but there was still a "cached" version in build/linux/work/hardware. I had set up symlinks from my sketchbook into my git checkout to make sure any changes to source files would be picked up without having to update the build folder copy, but it turns out both versions of the platform.txt are removed, so uncommenting the recipe didn't work.

In any case, now that I can reliably reproduce the problem, I can confirm it no longer occurs with the latest git master of arduino-builder, so I'm closing this issue.

closed 2016-04-26 by matthijskooijman

ShyPixie, 2016-08-30:
I have the same problem today, but only when I set the fqbn to arduino:avr:nano.

It's working fine with arduino:avr:uno.

matthijskooijman, 2016-08-30:
Could you post the full verbose log of an attempt with both the nano and the uno? Are you using the IDE, or calling builder directly?

ShyPixie, 2016-08-30:
Uno Log: http://termbin.com/jm64
Nano Log: http://termbin.com/p4sx

I'm using cmake to build, but the CMakeLists (Here: http://termbin.com/yaxd) just run the arduino-builder command.

cmaglie, 2016-08-30:
You must use `arduino:avr:nano:cpu=atmega328` as FQBN, that determines the value of -mcpu (that can be also atmega168 depending on the version of the nano).

An easy way to debug this kind of issues is to enable the "verbose compile" from the preferences of the IDE and see how arduino-builder is called when you compile.