arduino/arduino-builder #109: HELP on installing manually  staging and stable library

mkeyno, 2016-02-04:
hi guys
I'm so tired to repeatedly download the  over 150 MB library by Arduino broad manager feature , whereas  it seems the extensa tool chain  has no changed and only core file has been updated, I've tried couple of efforts to change the performance.text  and json file , but failed and no success 
any help is really appreciated

mkeyno, 2016-02-04:
any help is really appreciated

cmaglie, 2016-02-08:
You don't need to "repeatedly" download the toolchain, the IDE downloads the toolchains only once and updates only the cores. The package index has been done exactly for that.

mkeyno, 2016-02-16:
thanks @cmaglie  but it is wrong , whenever add new links to board manger url , it remove what every exist in localdata/arduino 15 folder and  download new files , however I want use the old form of installation whenever we manually add the folder in arduino installation folder(_not the c:\Users\Private\AppData\local_), any help really appreciated

cmaglie, 2016-02-16:
> thanks @cmaglie but it is wrong

I'm sorry but for me it works as expected, and I don't know the current status of your installation.
Unless you provide more details on how to reproduce the problem, I fear that I can't help you.

> however I want use the old form of installation whenever we manually add the folder in arduino installation folder(not the c:\Users\Private\AppData\local),

the "old" method of adding the "hardware" folder in the sketchbook **still works**.
You can use the "Boards manager" or the old installation method, it's your choice.

per1234, 2022-06-12:
Closing due to lack of response.

@mkeyno if are able to reproduce the issue with [the latest nightly build of Arduino CLI](https://arduino.github.io/arduino-cli/latest/installation/#nightly-builds) (the replacement for the deprecated arduino-builder) and will provide the requested information by commenting here, I'll be happy to re-open this issue.

closed 2022-06-12 by per1234