arduino/arduino-builder #110: Not enough time for downloads

ShorTie8, 2016-02-05:
trying to run
```
go test -v ./src/arduino.cc/builder/test/...
```
and get
```
Downloading http://downloads.arduino.cc/gcc-arm-none-eabi-4.8.3-2014q1-arm.tar.bz2
SIGQUIT: quit
PC=0x115e50

*** Test killed with quit: ran too long (10m0s).
FAIL    arduino.cc/builder/test 600.095s
```

Any way to ignore download time ??

ShorTie
ps: Dail-up  rulez

cmaglie, 2016-02-05:
add `-timeout 60m`:

`go test -timeout 60m -v ./src/arduino.cc/builder/test/...`

PS: can you try the pre-build version here: https://github.com/arduino/Arduino/pull/4517#issuecomment-180287368 ?
I need to know if the tools works also on other boards (I tested only on RPi 1 & 2) so I can move them to the main index.json and remove the hack to make it build!

ShorTie8, 2016-02-05:
Thankz,
Yes I will, in a few

Have A Great Day

ShorTie8, 2016-02-05:
is it really downloading twice ??  Downloading http://downloads.arduino.cc/gcc-arm-none-eabi-4.8.3-2014q1-arm.tar.bz2
```
Downloading http://downloads.arduino.cc/cores/avr-1.6.9.tar.bz2
Downloading http://downloads.arduino.cc/cores/sam-1.6.5.tar.bz2
Downloading http://downloads.arduino.cc/cores/samd-1.6.2.tar.bz2
Downloading http://downloads.arduino.cc/tools/avrdude-6.0.1-arduino5-armhf-pc-linux-gnu.tar.bz2
Downloading http://downloads.arduino.cc/tools/avr-gcc-4.8.1-arduino5-armhf-pc-linux-gnu.tar.bz2
Downloading http://downloads.arduino.cc/gcc-arm-none-eabi-4.8.3-2014q1-arm.tar.bz2
Downloading http://downloads.arduino.cc/tools/coan-5.2-arm-linux-gnueabihf.tar.bz2
Downloading http://downloads.arduino.cc/tools/ctags-5.8-arduino6-armv6-linux-gnueabihf.tar.bz2
Downloading http://downloads.arduino.cc/bossac-1.6.1-arduino-arm-linux-gnueabihf.tar.bz2
Downloading http://downloads.arduino.cc/bossac-1.5-arduino2-arm-linux-gnueabihf.tar.bz2
Downloading http://downloads.arduino.cc/tools/OpenOCD-0.9.0-arduino-arm-linux-gnueabihf.tar.bz2
Downloading http://downloads.arduino.cc/CMSIS-4.0.0.tar.bz2
Downloading http://downloads.arduino.cc/gcc-arm-none-eabi-4.8.3-2014q1-arm.tar.bz2

```

ShorTie8, 2016-02-05:
still fails with
```
FAIL
exit status 1
FAIL    arduino.cc/builder/test 1745.780s
OdroidC1 Arduino_IDE #
```
I see 2 things
```
/tmp/test867067295/core/core.a(main.cpp.o): In function `main':
/usr/local/src/Arduino_IDE/Arduino/build/arduino-builder/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/cores/arduino/main.cpp:43: undefined reference to `setup'
/usr/local/src/Arduino_IDE/Arduino/build/arduino-builder/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/cores/arduino/main.cpp:46: undefined reference to `loop'
collect2: error: ld returned 1 exit status

=== RUN TestSetupBuildPropertiesWithMissingPropsFromParentPlatformTxtFiles
Board my_avr_platform:avr:mymega doesn't define a 'build.board' preference. Auto-set to: AVR_MYMEGA
Board my_symlinked_avr_platform:avr:mymega doesn't define a 'build.board' preference. Auto-set to: AVR_MYMEGA
WARNING: Category '' in library SPI is not valid. Setting to 'Uncategorized'
--- PASS: TestSetupBuildPropertiesWithMissingPropsFromParentPlatformTxtFiles (0.18s)
=== RUN TestSetupHumanLogger
--- PASS: TestSetupHumanLogger (0.00s)
=== RUN TestLoadSketchWithFolder
--- PASS: TestLoadSketchWithFolder (0.00s)
=== RUN TestLoadSketchNonExistentPath
--- PASS: TestLoadSketchNonExistentPath (0.00s)
=== RUN TestLoadSketch
--- PASS: TestLoadSketch (0.00s)
=== RUN TestFailToLoadSketchFromFolder
--- PASS: TestFailToLoadSketchFromFolder (0.00s)
=== RUN TestLoadSketchFromFolder
--- PASS: TestLoadSketchFromFolder (0.00s)
=== RUN TestLoadSketchWithBackup
--- PASS: TestLoadSketchWithBackup (0.00s)
=== RUN TestLoadSketchWithMacOSXGarbage
--- PASS: TestLoadSketchWithMacOSXGarbage (0.00s)
=== RUN TestMergeSketch
--- PASS: TestMergeSketch (0.00s)
=== RUN TestStoreBuildOptionsMap
--- PASS: TestStoreBuildOptionsMap (0.00s)
=== RUN TestTargetBoardResolverUno
--- PASS: TestTargetBoardResolverUno (0.03s)
=== RUN TestTargetBoardResolverDue
--- PASS: TestTargetBoardResolverDue (0.02s)
=== RUN TestTargetBoardResolverMega1280
--- PASS: TestTargetBoardResolverMega1280 (0.03s)
=== RUN TestTargetBoardResolverMega2560
--- PASS: TestTargetBoardResolverMega2560 (0.02s)
=== RUN TestTargetBoardResolverCustomYun
--- PASS: TestTargetBoardResolverCustomYun (0.03s)
=== RUN TestTargetBoardResolverCustomCore
--- PASS: TestTargetBoardResolverCustomCore (0.03s)
=== RUN TestTime
--- FAIL: TestTime (0.01s)
```
trying yours now

cmaglie, 2016-02-05:
I see that `TestTime` fails on mine too, something I have to check.

The other errors are expected (for example the test TestSetupBuildPropertiesWithMissingPropsFromParentPlatformTxtFiles checks how the builder behaves when there are missing properties). The important thing is that the test PASS no matter the output.

commit ae801063 (2016-02-08): Added larger timeout on the command line for test (readme)

Fix #110
closed 2016-02-08 by cmaglie