arduino/arduino-builder #120: ctags breaks on template instantiations that bitshift in parameters

focalintent, 2016-02-24:
FastLED makes pretty extensive use of templates to do a lot of this work.  Some of what's done with templates involves creating bitmasks from template parameters as parameters to other templates.  Unfortunately, when the ctags parser sees "1<<X" as a template parameter, it eats itself.

This code demonstrates the problem:

```
template <int PORT_MASK> class controller { 
  
};

template <int LANES> class myClass { 
  public:
    void useController(controller<(1<<LANES)-1> & controller);    
};

void setup() { myctagstestfunc(); }
void myctagstestfunc() { }
void loop() {}
```

Because ctags stops parsing, the prototype for myctagstestfunc() is never created, and the dreaded ```'myctagstestfun' was not declared in this scope``` error (as seen in #68)

masatake, 2016-02-24:
```
[yamato@x201]~/var/ctags-github% ./ctags --sort=no -o - foo.cpp
controller	foo.cpp	/^template <int PORT_MASK> class controller { $/;"	c	file:
myClass	foo.cpp	/^template <int LANES> class myClass { $/;"	c	file:
setup	foo.cpp	/^void setup() { myctagstestfunc(); }$/;"	f
myctagstestfunc	foo.cpp	/^void myctagstestfunc() { }$/;"	f
loop	foo.cpp	/^void loop() {}$/;"	f
```
This is the expected tags output of this issue?

focalintent, 2016-02-24:
Pretty much, yes.

masatake, 2016-02-24:
https://github.com/masatake/ctags/commit/9a7ce2337c1f681ae53758768fd45f77afdda3a4

focalintent, 2016-02-24:
Unfortunately - that commit will most likely fail in the case of:

```
template <int PORT_MASK> class controller { 
};

controller< 1<<8 > aController;
```

because the code in the commit that you linked to only ignores the <> characters after an = or when inside of ()'s.  

So even with your commit - you should still do the one character lookahead to decide whether or not you've got a bit shift happening.   Of course, then there's also having conditional expressions as template parameters which would likely trip up both of our fixes.

focalintent, 2016-02-24:
Yup - just tried here with 

```
template <int P> class c {};
c< 8 > bVar;
c< 1<<8 > aVar;
```

and I get:

```
Xantico:u-ctags dgarcia$ ./ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/x.cpp
ctags: Warning: Unsupported parameter 'T' for "fields" option
bVar	/tmp/x.cpp	/^c< 8 > bVar;$/;"	kind:variable	line:2
```

aVar does'n't come out - if I add the lookahead/skip in c.c (similar to what's in my pull request in arduino/ctags#7) then I get this output:

```
template <int P> class c {};
c< 8 > bVar;
c< 1<<8 > aVar;
```

focalintent, 2016-02-24:
https://github.com/universal-ctags/ctags/pull/807 for a fix on universal-ctags

facchinm, 2016-03-04:
Fixed by merging arduino/ctags#7

closed 2016-03-04 by facchinm